epics-base/pvAccessCPP #15: DBD conformance

NickeZ, 2015-10-20:
In EPICS 3.14 this file, `pvAccessCPP/src/ioc/PVAServerRegister.dbd`, doesn't expand correctly. It works in 3.15. I don't know whether this is a bug towards the library or EPICS base.

The following command yields a broken registerRecordDeviceDriver.cpp

```
perl /opt/epics/bases/base-3.14.12.5/bin/centos7-x86_64/registerRecordDeviceDriver.pl ../../builddir/3.14.12.5/dbd/pvAccessCPP.dbd pvAccessCPP_registerRecordDeviceDriver > pvAccessCPP_registerRecordDeviceDriver.cpp.tmp
```

The fix is to remove the quotation marks:
```
registrar("startPVAServerRegister")
registrar("stopPVAServerRegister")
```

becomes
```
registrar(startPVAServerRegister)
registrar(stopPVAServerRegister)
```

anjohnson, 2016-03-03:
I acknowledge that the simplistic DBD parser in Base-3.14's registerRecordDeviceDriver.pl script does not accept double quotes inside registrar statements, although the dbStaticLib parser does. The [documented DBD file syntax](http://www.aps.anl.gov/epics/base/R3-14/12-docs/AppDevGuide/node7.html#SECTION00720000000000000000) implies that quotes should be permitted, so the Perl script in 3.14 is really where the fault lies. The script's parser was completely replaced in Base-3.15 which is why that version accepts quotes.

I am committing a fix to the script on the Base-3.14 branch so any future releases (3.14.12.6 and later) will accept quotes in DBD files in the places where they are legal.

In practice though the individual component DBD files normally get parsed by the dbExpand program when building an IOC-specific DBD file, and since dbExpand never puts double-quotes in its output the Perl script hasn't had to recognize them before. How did you come across this bug in the first place? You can't have been using the standard EPICS build system.

closed 2016-03-03 by anjohnson

NickeZ, 2016-03-04:
Yeah, I'm using a custom build system at ESS where we build epics modules more as plug-ins.

anjohnson, 2016-03-04:
Is someone from ESS planning to talk about your build system at the meeting in May? Have you developed any build system changes that might be worth feeding back into Base?

NickeZ, 2016-03-04:
Yeah, hopefully I will probably give a talk. I think there are some great ideas in our build system and some of them might be upstreamable, but there are also some added limitations which might not be universally appreciated...