epics-base/pvAccessCPP #16: pvtools segfault

NickeZ, 2015-10-26:
I accept that the issue might be in some way related to the fact that I compile pvAccessCPP using another build system. Here is the issue anyway:

If I run any of the tools (pvput, pvget, pvinfo...) without a proper argument I get a segfault:

```
[vagrant@localhost ~]$ pvput
No pv name specified. ('pvput -h' for help.)
Segmentation fault
[vagrant@localhost ~]$ pvget
No pv name(s) specified. ('pvget -h' for help.)
Segmentation fault
[vagrant@localhost ~]$ pvinfo
No pv name(s) specified. ('pvinfo -h' for help.)
Segmentation fault
[vagrant@localhost ~]$ gdb pvput
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-64.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /opt/epics/modules/pvAccessCPP/4.1.2/3.14.12.5/bin/centos7-x86_64/pvput...done.
(gdb) run
Starting program: /opt/epics/modules/pvAccessCPP/4.1.2/3.14.12.5/bin/centos7-x86_64/pvput 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff599a700 (LWP 25448)]
No pv name specified. ('pvput -h' for help.)
[Thread 0x7ffff599a700 (LWP 25448) exited]

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff757d18d in ellDelete (pList=0x7ffff77b2180 <mutexList>, pNode=0x7847e0) at ../../../src/libCom/ellLib/ellLib.c:87
87	        pNode->previous->next = pNode->next;
Missing separate debuginfos, use: debuginfo-install glibc-2.17-78.el7.x86_64 libgcc-4.8.3-9.el7.x86_64 libstdc++-4.8.3-9.el7.x86_64 ncurses-libs-5.9-13.20130511.el7.x86_64 readline-6.2-9.el7.x86_64
(gdb) bt
#0  0x00007ffff757d18d in ellDelete (pList=0x7ffff77b2180 <mutexList>, pNode=0x7847e0) at ../../../src/libCom/ellLib/ellLib.c:87
#1  0x00007ffff758cbad in epicsMutexDestroy (pmutexNode=0x7847e0) at ../../../src/libCom/osi/epicsMutex.cpp:128
#2  0x00007ffff60501ba in __cxa_finalize () from /lib64/libc.so.6
#3  0x00007ffff6c9e653 in __do_global_dtors_aux () from /opt/epics/modules/pvAccessCPP/4.1.2/3.14.12.5/lib/centos7-x86_64/libpvAccessCPP.so
#4  0x00007fffffffe070 in ?? ()
#5  0x00007ffff7deab5a in _dl_fini () from /lib64/ld-linux-x86-64.so.2
Backtrace stopped: frame did not save the PC
```

ralphlange, 2015-10-26:
What build system do you use?
What versions of g++ and standard libraries?

On a Debian squeeze system (g++ 4.9.3, glibc 2.19, libstdc++ 5.1.1) I cannot reproduce this fault with Base 3.14.3, Base 3.14 trunk or Base 3.15 trunk.

NickeZ, 2015-10-26:
We use an ["in-house" build system](https://bitbucket.org/europeanspallationsource/m-epics-environment/) based on Dirk Zimochs (PSI).

I'm using Centos 7.1 which contains the following:

* g++ 4.8.3
* libc 3.17
* libstdc++ 6.0.19 (3.4.19)
* base 3.14.12.5 and 3.15.2 (same issue with both)

I use the following versions of the V4 modules:

* /opt/epics/modules/pvDataCPP/5.0.2
* /opt/epics/modules/pvAccessCPP/4.1.2
* /opt/epics/modules/normativeTypesCPP/5.0.2
* /opt/epics/modules/pvaClientCPP/4.1.1
* /opt/epics/modules/pvaSrv/0.11.2

```
[vagrant@localhost work]$ g++ --version
g++ (GCC) 4.8.3 20140911 (Red Hat 4.8.3-9)
```

```
[vagrant@localhost work]$ /lib64/libc-2.17.so 
GNU C Library (GNU libc) stable release version 2.17, by Roland McGrath et al.
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 4.8.3 20140911 (Red Hat 4.8.3-9).
Compiled on a Linux 3.10.0 system on 2015-03-05.
Available extensions:
	The C stubs add-on version 2.1.2.
	crypt add-on version 2.1 by Michael Glad and others
	GNU Libidn by Simon Josefsson
	Native POSIX Threads Library by Ulrich Drepper et al
	BIND-8.2.3-T5B
	RT using linux kernel aio
libc ABIs: UNIQUE IFUNC
For bug reporting instructions, please see:
<http://www.gnu.org/software/libc/bugs.html>.

```

```
[vagrant@localhost work]$ strings /usr/lib64/libstdc++.so.6.0.19 | grep LIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_DEBUG_MESSAGE_LENGTH
```

ralphlange, 2015-10-26:
As the segfault seems to occur while the destructors for global objects are being called...

If I remember correctly, your in-house build system supports run-time loading of libraries (for IOCs, from the startup script). How does this provide the proper calling of constructors and destructors for global objects in libraries?

NickeZ, 2015-10-26:
Yes, when we run IOCs we load libraries dynamically. But not when we run executables like pvput.

Do I need any special compiler flags to "properly" destruct global objects? The only thing googles tells me is to avoid global objects [reference](http://stackoverflow.com/questions/21157476/why-global-or-static-object-can-lead-to-crash-when-program-exit).

msekoranja, 2015-10-26:
Just to make 100,1% sure: you've done a clean build of all the modules, right?

NickeZ, 2015-10-26:
@msekoranja I've only checked out the versions I listed above, so in that case it is a "clean" build. I do however use a custom build system which builds for all our EPICS bases and packages the resources in a uniform way. 

I'm not 100% sure that the build system I use is compatible with pvAccessCPP, but so far it works with everything I've thrown at it. I can give you the complete build log if your interested.

anjohnson, 2015-10-26:
The only thing that comes to my mind is to check that the executable programs are being linked using g++ and not ld. Since replication of the segfaults is easy for you, please run one of the programs under gdb and post the additional information it gives you; include a backtrace if possible.

anjohnson, 2015-10-26:
Sorry, I see you did send the backtrace in your original message...

NickeZ, 2015-10-26:
Here is the stacktrace from valgrind as well. Looks like a null-pointer in a global destructor. Since I'm not familiar with the code I can't really say anything. Do you have multiple global objects where the order of destruction matters?

```
==3516== Invalid write of size 8
==3516==    at 0x548218D: ellDelete (ellLib.c:87)
==3516==    by 0x5491BAC: epicsMutexDestroy (epicsMutex.cpp:128)
==3516==    by 0x68711B9: __cxa_finalize (cxa_finalize.c:55)
==3516==    by 0x5D35652: ??? (in /opt/epics/modules/pvAccessCPP/4.1.2/3.14.12.5/lib/centos7-x86_64/libpvAccessCPP.so)
==3516==    by 0x400FB59: _dl_fini (dl-fini.c:253)
==3516==    by 0x6870E48: __run_exit_handlers (exit.c:77)
==3516==    by 0x6870E94: exit (exit.c:99)
==3516==    by 0x6859AFB: (below main) (libc-start.c:308)
==3516==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
```

anjohnson, 2015-10-26:
Hmm, base/src/libCom/osi/epicsMutex.cpp has a `static int firstTime = 1;` variable which looks slightly suspicious. It should be an epicsThreadOnceId instead, provided doing that doesn't cause circular initialization issues (works for me on Linux, haven't tested other OSs yet).

Does applying this patch [epicsMutex-patch.txt](https://github.com/epics-base/pvAccessCPP/files/21005/epicsMutex-patch.txt) to your Base installation make any difference?

NickeZ, 2015-10-26:
Nope still the same issue, did you see this line in the OP?

```
0x00007ffff757d18d in ellDelete (pList=0x7ffff77b2180 <mutexList>, pNode=0x7847e0) at ../../../src/libCom/ellLib/ellLib.c:87
87          pNode->previous->next = pNode->next;
```

anjohnson, 2015-10-26:
That call to ellDelete() is this line in epicsMutexDestroy() from the epicsMutex.cpp file that I patched:
```
    ellDelete(&mutexList,&pmutexNode->node);
```
There is no code in that file which empties the `static ELLLIST mutexList` so the mutexList should be valid, and the pmutexNode should be on that list. Can you get to the above point and give gdb the commands `print *pList` and `print *pNode` and post the responses, just to confirm that it is the node that has been corrupted. The implication is that something has written zeros over the epicsMutexParm object (an epicsMutexId is just a pointer to an epicsMutexParm). You may be able to find out where that mutex was created by running `call epicsMutexShow(pNode)` but if it has been zeroed that might not work.

anjohnson, 2015-10-26:
Looking at your build log, do you realize that the compiler is being passed all the Base include directories twice? This is the complete list of `-I` commands being passed to both the native and cross-compilers:

```
-I.
-I../../builddir/3.15.2/include
-I../../builddir/3.15.2/include/os/Linux
-I../../src/ca/
-I../../src/client/
-I../../src/factory/
-I../../src/ioc/
-I../../src/mb/
-I../../src/pva/
-I../../src/remote/
-I../../src/remoteClient/
-I../../src/rpcClient/
-I../../src/rpcService/
-I../../src/server/
-I../../src/utils/
-I/opt/epics/modules/pvDataCPP/5.0.2/3.15.2/include
-I/opt/epics/modules/pvDataCPP/5.0.2/3.15.2/include/os/Linux
-I/opt/epics/modules/pvDataCPP/5.0.2/3.15.2/include/os/default
-I/opt/epics/bases/base-3.15.2/include/compiler/gcc
-I/opt/epics/bases/base-3.15.2/include/os/Linux
-I/opt/epics/bases/base-3.15.2/include
-I/opt/epics/bases/base-3.15.2/include
-I/opt/epics/bases/base-3.15.2/include/compiler/gcc
-I/opt/epics/bases/base-3.15.2/include/os/Linux
```

The last 3 entries are unnecessary duplicates of the previous set of 3, although I don't think they're doing any harm. The previous set are given in the correct order (compiler, then OS, then generic), so if you remove them make sure you delete the correct set.

anjohnson, 2015-10-26:
Ok, I see now that your build log compiles everything against both 3.14.12.5 and 3.15.2, which was confusing me a bit. For the 3.14 builds there is no include/compiler directory, but removing the last 3 include entries would still be correct.

NickeZ, 2015-10-27:
Thanks everyone I found the issue!

I was using ${LINK.cpp} to link my executables, but that macro contained all the object files that was being linked into the shared library. So my "pvput" contained a lot more code than it should.

Do you have a macro for linking or should I simply use ${CCC} ?

closed 2015-10-27 by NickeZ

ralphlange, 2015-10-27:
Look at the lines 95-100 (3.15) or 80-85 (3.14) of .../configure/os/CONFIG.Common.UnixCommon:

Inside the EPICS Build system, the line is
`$(CCC) -o <target> $(STATIC_LDFLAGS) $(PRODDIR_LDFLAGS) $(LDFLAGS) $(PROD_LDFLAGS) <your_objects> $(PROD_LDLIBS)`

Run a "normal" make build to find what the flags expand to on your system (some are set in CONFIG.Common.linuxCommon).
Same for compiling, probably.

NickeZ, 2015-10-27:
Thanks for the pointer, I wrongly assumed that the semantics for linking was similar to compiling. EPICS Base provides the compiler macro in `configure/CONFIG_COMMON` like this:

```
COMPILE.c = $(CC) -c $(CPPFLAGS) $(CFLAGS) $(call PATH_FILTER,$(INCLUDES))
COMPILE.cpp = $(CCC) -c $(CPPFLAGS) $(CXXFLAGS) $(call PATH_FILTER,$(INCLUDES))
```