epics-base/pvAccessCPP #23: If more than one provider than base-3.15 fails

mrkraimer, 2016-02-04:
If more than one provider is registered than attempts to connect to any channel not provided by the first provider fails.

This is NOT true for base-3.14 releases.

Consider two providers: pvaSrv and local.
pvget attempts to connect to a channel provided by local.
The following sequence of actions occurs:

A channel find is issued to pvaSrv, which fails,
A channel find is issued to local, which succeeds.
A channel create is issued.
For some reason this results in a call to the access security code in pvaSrv, which results on a failure returned to pvget,

msekoranja, 2016-02-08:
At connection time one plugin in chosen to handle all the security related things for entire connection.
Now, the problem is multiple providers within one connection where security plugin is provider specific.

msekoranja, 2016-03-22:
pvaSrc caSecurity plugin vs non-CA provider issue. Not related only to 3.15. Security API is under redesign now to allow this. Closing.

closed 2016-03-22 by msekoranja