epics-base/pvAccessCPP #24: queueSize is a string?

mdavidsaver, 2016-02-10:
Any special reason that "record._options.queueSize" is a PVString?  I'd suggest a change to handle this as PVScalar with cast to uint32.  This would start a transition to storing this as an integer value.

So code like

```
int queueSize = atoi(pvRequest->getSubFieldT<PVString>("record._options.queueSize")->get());
```

would become

```
uint32 queueSize = atoi(pvRequest->getSubFieldT<PVScalar>("record._options.queueSize")->getAs<uint32>());
```

Which is certainly less error prone than calling atoi().  ATM queueSize="oops" will produce some interesting results.

Similar changes would be necessary in (at least pvaSrv).

msekoranja, 2016-02-10:
So that it can be specified using command line tools.

msekoranja, 2016-03-22:
Closing. @mdavidsaver has a change done and scalar values will be additionally supported when pull request is made.

closed 2016-03-22 by msekoranja