epics-base/pvAccessCPP #30: Server-side seg fault with pvaSrv when monitor client exits

dhickin, 2016-03-03:
<p>If you start an IOC running pvaSrv and monitor a PV using pvget -m then when the client exits there is a server-side seg fault.</p>

<p>Specifically I've seen this with pvaSrv/testTop and exampleCPP/database</p>

<p>e.g. for testTop</p>

<p>Server-side:</p>
[pvaSrv]$ ../../bin/linux-x86_64/testDbPv st.cmd

<p>Client-side:</p>
any pvget -m, e.g

$ pvget -m double01

<p>then ^C</p>

<p>I've only seen this after the merge of PR  #27 (merge of major gateway support changes to master branch).</p>

<p>Rolling back to an earlier pvAccess (I used f0b88e349b0193e3621b76ba17321b9ff09a7ba3 Merge pull request #22 from mdavidsaver/pvsubdir) I don't see the problem.</p>

<p>Seen for every type of record in these examples and with every request I've tried.<br/>
Gets and puts unaffected. RPC unaffected.<br/>
Does not affect caProvider (i.e. if you monitor ca:///<pv-name>).<br/>
pvDatabase PVs unaffected<br/>
Seen with 3.14.12.3 and 3.15.2/3<br/>
Seen on both my RHEL system at DLS and on my mint PC at home.<br/>
Does not matter if monitors are on same or different machines.</p><br/>

<p>
RHEL 6 OS:<br/>
OS:<br/>
$ lsb_release -a<br/>
LSB Version:	:base-4.0-amd64:base-4.0-noarch:core-4.0-amd64:core-4.0-noarch:graphics-4.0-amd64:graphics-4.0-noarch:printing-4.0-amd64:printing-4.0-noarch<br/>
Distributor ID:	RedHatEnterpriseWorkstation
Description:	Red Hat Enterprise Linux Workstation release 6.7 (Santiago)<br/>
Release:	6.7<br/>
Codename:	Santiago<br/>
</p>

gdb/valgrind info to follow.

dhickin, 2016-03-03:
On R3.15.3 running on my RHEL 6 machine I get the following backtrace for linux-x86_64:

Program received signal SIGSEGV, Segmentation fault.
[Switching to Threads 0x7fffef568700 (LWP 31475)]
0x00007ffff6b0bcc6 in epicsThreadPrivateSet (id=<value optimized out>, 
    value=<value optimized out>)
    at ../../../src/libCom/osi/os/posix/osdThread.c:877
877	}
Missing separate debuginfos, use: debuginfo-install glibc-2.12-1.166.el6_7.7.x86_64 libgcc-4.4.7-16.el6.x86_64 libstdc++-4.4.7-16.el6.x86_64 ncurses-libs-5.7-4.20090207.el6.x86_64 nss-softokn-freebl-3.14.3-23.el6_7.x86_64 readline-6.0-4.el6.x86_64
(gdb) bt
#0  0x00007ffff6b0bcc6 in epicsThreadPrivateSet (id=<value optimized out>, 
    value=<value optimized out>)
    at ../../../src/libCom/osi/os/posix/osdThread.c:877
#1  0x01007ffff6b02647 in ?? ()
#2  0x00007fffa00097a0 in ?? ()
#3  0x0000000000000000 in ?? ()
I get this every time.

However it appears to be a bit of a Heisenbug. When I tried a linux-x86_64-debug build I can't reproduce it.

dhickin, 2016-03-03:
I've tried reproducing the issue Marty reported, namely running exampleCPP/database in one window:
<pre>
mrk> pwd
/home/epicsv4/master/exampleCPP/database/iocBoot/exampleDatabase
mrk> valgrind ../../bin/linux-x86_64/exampleDatabase st.cmd
</pre>

and running

<pre>
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffeebb2700 (LWP 16340)]
0x00007ffff6d4bcc6 in epicsThreadPrivateSet (id=<value optimized out>, 
    value=<value optimized out>)
    at ../../../src/libCom/osi/os/posix/osdThread.c:877
877	}
Missing separate debuginfos, use: debuginfo-install glibc-2.12-1.166.el6_7.7.x86_64
libgcc-4.4.7-16.el6.x86_64
libstdc++-4.4.7-16.el6.x86_64 ncurses-libs-5.7-4.20090207.el6.x86_64
nss-softokn-freebl-3.14.3-23.el6_7.x86_64 readline-6.0-4.el6.x86_64
(gdb) bt
#0  0x00007ffff6d4bcc6 in epicsThreadPrivateSet (id=<value optimized out>, 
    value=<value optimized out>)
    at ../../../src/libCom/osi/os/posix/osdThread.c:877
#1  0x01007ffff6d42647 in ?? ()
#2  0x00007fff9c00a100 in ?? ()
#3  0x0000000000000000 in ?? ()
</pre>

mdavidsaver, 2016-03-15:
> gdb/valgrind info to follow.

What were the results w/ valgrind for linux-x86 (non -debug)?

dhickin, 2016-03-15:
I couldn’t reproduce the issue didn't running valgrind with the memcheck tool.

mrkraimer, 2016-03-15:
I can make it happen with the following.

mrk> pwd
/home/epicsv4/master/exampleCPP/database/iocBoot/exampleDatabase
mrk> valgrind --log-file=crash --leak-check=full 
../../bin/linux-x86_64/exampleDatabase st.cmd

And in another window:

mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/examplePvaClientMultiDouble

I had to run the example many times but finally exampleDatabase crashes.

The file crash has the followimg:

==16277== Memcheck, a memory error detector
==16277== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==16277== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==16277== Command: ../../bin/linux-x86_64/exampleDatabase st.cmd
==16277== Parent PID: 15008
==16277==
==16277== Thread 37 TCP-rx:
==16277== Jump to the invalid address stated on the next line
==16277==    at 0x100000005ED1513: ???
==16277==    by 0x14D98DFF: ???
==16277==  Address 0x100000005ed1513 is not stack'd, malloc'd or 
(recently) free'd
==16277==
==16277==
==16277== Process terminating with default action of signal 11 (SIGSEGV)
==16277==  Bad permissions for mapped region at address 0x100000005ED1513
==16277==    at 0x100000005ED1513: ???
==16277==    by 0x14D98DFF: ???
==16277==
==16277== HEAP SUMMARY:
==16277==     in use at exit: 2,928,378 bytes in 15,927 blocks
==16277==   total heap usage: 32,146 allocs, 16,219 frees, 6,927,037 
bytes allocated
==16277==
==16277== Thread 1:
==16277== 28 bytes in 1 blocks are definitely lost in loss record 498 of 
3,278
==16277==    at 0x4C28C50: malloc (in 
/usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==16277==    by 0x5ED111E: mallocMustSucceed (cantProceed.c:44)
==16277==    by 0x5ED9C02: epicsEnvSet (osdEnv.c:43)
==16277==    by 0x5ECC80F: iocshLoad (iocsh.cpp:894)
==16277==    by 0x4042F1: main (exampleDatabaseMain.cpp:24)

Lots of similar output and then:

==16277== LEAK SUMMARY:
==16277==    definitely lost: 627 bytes in 8 blocks
==16277==    indirectly lost: 2,767 bytes in 53 blocks
==16277==      possibly lost: 29,240 bytes in 58 blocks
==16277==    still reachable: 2,895,744 bytes in 15,808 blocks
==16277==                       of which reachable via heuristic:
==16277==                         newarray           : 192 bytes in 2 blocks
==16277==         suppressed: 0 bytes in 0 blocks
==16277== Reachable blocks (those to which a pointer was found) are not 
shown.
==16277== To see them, rerun with: --leak-check=full --show-leak-kinds=all

I will guess that the most important info is:

==16277== Thread 37 TCP-rx:
==16277== Jump to the invalid address stated on the next line
==16277==    at 0x100000005ED1513: ???
==16277==    by 0x14D98DFF: ???
==16277==  Address 0x100000005ed1513 is not stack'd, malloc'd or 
(recently) free'd

mdavidsaver, 2016-03-15:
I can replicate, with slight differences.  I also don't see the crash with -debug or in valgrind.  I also don't see if so often without -debug.  When I do see it, starting a monitor appears to work as expected.  I see the crash when the monitor stops, or when starting a second (sequential) monitor.  Starting a second monitor concurrent with the first works as expected.

With PVA_DEBUG=5 the last message I see, immediately before the crash, is:

> 2016-03-15T14:10:55.514 TCP socket to 10.5.3.1:38189 is to be closed.
> 2016-03-15T14:10:55.514 Transport to 10.5.3.1:38189 still has 1 channel(s) active and closing...

This smells like a race w/ corruption on worker thread shutdown.  The shutdown sequence is one of the things I changed.

mdavidsaver, 2016-03-15:
> _channel = INVALID_SOCKET; //TODO: mutex to guard _channel

Gosh, isn't that helpful.  I wonder which fool did this...

msekoranja, 2016-03-15:
I am getting the following trance in gdb:

[New Thread 0x7ffff0f31700 (LWP 19940)]
2016-03-15T20:21:53.235 TCP socket to 10.5.1.1:60801 is to be closed.
2016-03-15T20:21:53.235 Transport to 10.5.1.1:60801 still has 1 channel(s) active and closing...
[Thread 0x7ffff0fb2700 (LWP 19939) exited]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff1033700 (LWP 19936)]
0x00007ffff5a86370 in ellDelete (pList=pList@entry=0x81b150, pNode=pNode@entry=0x81b101) at ../../../src/libCom/ellLib/ellLib.c:82
82	        pNode->next->previous = pNode->previous;
(gdb) bt
#0  0x00007ffff5a86370 in ellDelete (pList=pList@entry=0x81b150, pNode=pNode@entry=0x81b101) at ../../../src/libCom/ellLib/ellLib.c:82
#1  0x00007ffff5a92d2a in epicsExitCallAtExitsPvt (pep=<optimized out>) at ../../../src/libCom/misc/epicsExit.c:81
#2  epicsExitCallAtThreadExits () at ../../../src/libCom/misc/epicsExit.c:108
#3  0x00007ffff5a9b2dc in start_routine (arg=0x80bda0) at ../../../src/libCom/osi/os/posix/osdThread.c:387
#4  0x00007ffff4d1fb50 in start_thread (arg=<optimized out>) at pthread_create.c:304
#5  0x00007ffff50117bd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:112
#6  0x0000000000000000 in ?? ()

Easy to reproduce with: run "pvget -m counter01 &" multiple times and the call "killall -9 pvget".

dhickin, 2016-03-15:
I am getting similar on my machine at home:
<pre>
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff128d700 (LWP 26605)]
0x00007ffff6721ce0 in ellDelete (pList=pList@entry=0x7fff9c009200, 
    pNode=pNode@entry=0x7fff9c009201) at ../../../src/libCom/ellLib/ellLib.c:82
82	        pNode->next->previous = pNode->previous;
(gdb) bt

#0  0x00007ffff6721ce0 in ellDelete (pList=pList@entry=0x7fff9c009200, 
    pNode=pNode@entry=0x7fff9c009201) at ../../../src/libCom/ellLib/ellLib.c:82
#1  0x00007ffff672e4ea in epicsExitCallAtExitsPvt (pep=<optimised out>)
    at ../../../src/libCom/misc/epicsExit.c:81
#2  epicsExitCallAtThreadExits () at ../../../src/libCom/misc/epicsExit.c:108
#3  0x00007ffff6736b94 in start_routine (arg=0x7fffa4009810)
    at ../../../src/libCom/osi/os/posix/osdThread.c:387
#4  0x00007ffff53ed182 in start_thread (arg=0x7ffff128d700)
    at pthread_create.c:312
#5  0x00007ffff643247d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
</pre>

msekoranja, 2016-03-15:
Note: it does not happen against testServer. So it must be somehow pvaSrv related. Does pvaSrv/IOC install any thread hooks?

msekoranja, 2016-03-15:
The following "fix" works for me:

msekoranja@vmdebian77:~/epicsV4/pvaSrv/src$ git diff
diff --git a/src/dbPv/3.14/caContext.cpp b/src/dbPv/3.14/caContext.cpp
index e702438..1f82838 100644
--- a/src/dbPv/3.14/caContext.cpp
+++ b/src/dbPv/3.14/caContext.cpp
@@ -62,7 +62,7 @@ caContext::caContext(RequesterPtr const & requester)
         requester->message("ca_add_exception_event failed",warningMessage);
     }
     context = ca_current_context();
-    epicsAtThreadExit(threadExitFunc,this);
+    //epicsAtThreadExit(threadExitFunc,this);
 }
 
Can anyone verify if this works for him too. If this is the case, then we at least have a clue what's wrong.

dhickin, 2016-03-15:
Tried on DLS machine and my laptop and haven't been able to reproduce issue with epicsAtThreadExit(threadExitFunc,this);
commented out.

msekoranja, 2016-03-15:
OK. So, this is a pvaSrv issue. Note that using "this" as a reference in the code above (in constructor) is extremely bold.

mrkraimer, 2016-03-15:
I made the equivalent change for 3.15

mrk> git diff src/dbPv/3.15/caContext.cpp
diff --git a/src/dbPv/3.15/caContext.cpp b/src/dbPv/3.15/caContext.cpp
index e702438..f5cb7ee 100644
--- a/src/dbPv/3.15/caContext.cpp
+++ b/src/dbPv/3.15/caContext.cpp
@@ -62,7 +62,7 @@ caContext::caContext(RequesterPtr const & requester)
          requester->message("ca_add_exception_event 
failed",warningMessage);
      }
      context = ca_current_context();
-    epicsAtThreadExit(threadExitFunc,this);
+//    epicsAtThreadExit(threadExitFunc,this);
  }

And ran the exampleDatabase and examplePvaClientMultiDouble I described 
above built against base 3.15.

I ran examplePvaClientMultiDouble many many times and exampleDatabase 
never crashed.

Marty

mdavidsaver, 2016-03-16:
> The following "fix" works for me:

... and creates a resource leak by not removing the exiting thread from caContextCreate::contextMap, which will then grow without bound.

> OK. So, this is a pvaSrv issue.

Not is fast.

> Note that using "this" as a reference in the code above (in constructor) is extremely bold.

This isn't a "delete this" issue.  I can use enable_shared_from_this to ensure that destruction happens in threadExitFunc() and it still crashes.

I think the corruption occurs indirectly in ~caContext.  There is quite a bit happening here,  including a call to ~BlockingServerTCPTransportCodec (indirectly through ~BaseChannelRequester).

While I no longer believe the issue is the specific line I mentioned, I still think the issue is with server transport shutdown.

mdavidsaver, 2016-03-16:
> This isn't a "delete this" issue.

Which is not to say that I like to pattern.

msekoranja, 2016-03-16:
Sure, the above "fix" is not a real fix, just a "fix" helping to diagnose the problem.

mdavidsaver, 2016-03-16:
> Sure, the above "fix" is not a real fix, just a "fix" helping to diagnose the problem.

Sorry if I mis-interpreted.

FYI, I'm currently testing a better "fix" (aka workaround) by making caContext::requester a weak pointer.  This at least pushes the troublesome ~BlockingServerTCPTransportCodec to some earlier point.  No crashes so far.

dhickin, 2016-03-16:
> OK. So, this is a pvaSrv issue.

Is this definitely the case?

It is true that this is only seen in pvaSrv, but bear in mind this appeared after changes to pvAccess with pvaSrv remaining the same. Rolling back to an earlier pvAccess (e.g. f0b88e3 Merge pull request 22 from mdavidsaver/pvsubdir) the issue goes away.

So

1. Have the changes to pvAccess exposed an issue that has been there all along in pvaSrv?
<p>or</p>
2. Has an issue been introduced into pvAccess that is only shown up currently by pvaSrv?
<p>or</p>
3. Are there issues in both?

So if we re-write pvaSrv and the issue goes away that doesn't necessarily mean there aren't issues in pvAccess. In particular why has this suddenly appeared with the latest changes to pvAccess? Do we have a clear understanding of why the seg fault occurs?

mdavidsaver, 2016-03-16:
What pvaSrv is doing should not cause a crash, but imo. is still unnecessarily weird.  As far as I can tell, the situation where the crash occurs is that the PVA server channel is being deleted from a epicsAtThreadExit() handler on the channel's own RX thread.  This is a situation I had not envisioned.

> Do we have a clear understanding of why the seg fault occurs?

Despite several hours of investigation, no.  The change I propose in epics-base/pvaSrv#34 allows channel destruction to happen before the an exit handler, and seems to avoid a crash.  I consider this a workaround, and will continue to look into the PVA channel thread shutdown sequence.

mdavidsaver, 2016-03-16:
And the winner is... EPICS Base.  The epicsThreadCallEntryPoint() function stores a pointer to a local variable in the epicsThread class.  Calling epicsAtThreadExit::exitWait() from that thread's epicsAtThreadExit() handler writes to this local after epicsThreadCallEntryPoint() has returned.  Thus corrupting the stack.

The reason this appears now is that one of the changes I made was to use the epicsThread class instead of the C API.

mdavidsaver, 2016-03-16:
https://bugs.launchpad.net/epics-base/+bug/1558206

mdavidsaver, 2016-03-16:
@msekoranja I don't think any immediate change is necessary to PVA.

msekoranja, 2016-03-17:
@mdavidsaver Great! Good catch! 

On pvAccess destruction: basically deallocation all is based on shared_ptr. rx/tx thread hold reference to the transport. So it might happen that rx/tx is the trigger for destruction (when ref. count drops to 0).

Closing.

closed 2016-03-17 by msekoranja