epics-base/pvAccessCPP #32: Avoiding impromptu gateway?

mdavidsaver, 2016-03-18:
I'm looking at PVAServerRegister.dbd and PVAClientRegister.dbd (which I hadn't noticed before) with an eye to replacing some redundant code I have https://github.com/mdavidsaver/pva2pva/blob/master/p2pApp/server.cpp#L168

From reading through the ChannelProviderRegistry code it seems that if I build an IOC with both PVAServerRegister.dbd and PVAClientRegister.dbd I'll be accidentally creating a sort of gateway as the PVA server will by default find the PVA client provider.  I can see that setting EPICS_PVAS_PROVIDER_NAMES would prevent this.

This seems a rather accident prone default.  Am I missing something here?

mrkraimer, 2016-03-18:
> I'm looking at PVAServerRegister.dbd and PVAClientRegister.dbd (which
> I hadn't noticed before) with an eye to replacing some redundant code
> I have [...]

If this is a problem in an IOC then it is also a problem in a main 
program on both C++ and Java.
It is a problem whenever the PVRecord uses pva to connect to another 
record in the same server or another server.

For examples see  exampleCPP/exampleLink and exampleJava/exampleLink.

Matej,

Can You give some thoughts?

Marty

mdavidsaver, 2016-03-22:
In thinking about this, there needs to be some way of denoting which providers should be used by servers (visible outside the process), locally (inside the process), and perhaps other distinctions.  This could be accomplish with separate registry map<>s, or assigning a set of flags (bit mask) to each provider factory.  My preference would be for a bit mask.

msekoranja, 2016-03-22:
The EPICS_PVAS_PROVIDER_NAMES property specifies what providers server will expose over the network. Property is read at initialization time (as well as all other properties).

mdavidsaver, 2016-03-22:
@msekoranja What you say is true.  And I see that the default is "local".  So the PVA client won't be found by default.

I would suggest that allowing EPICS_PVAS_PROVIDER_NAMES="<all>" is not a great idea as this would include the PVA client, but as this isn't the default it would be to some extent self inflicted.

closed 2016-03-22 by mdavidsaver