epics-base/pvAccessCPP #35: command line tools lack version option

gregoryraymondwhite, 2016-03-19:
pvget, eget etc don't give an option to print version.

dhickin, 2016-03-21:
More to the point they don't have a version to print. The only versioning is of pvAccess itself.

However as I've previously advocated I think it would be better to move pvTools out of pvAccessCPP.  I'll open an issue to discuss this separately.

msekoranja, 2016-03-21:
@dhickin Please do not move pvTools out. This will make V4 deployment even more complex. Imagine not getting caget/caput with EPICS base.

dhickin, 2016-03-21:
OK, it was just a suggestion. I opened the issue already, but if there's not agreement it can just be closed.

gregoryraymondwhite, 2016-03-21:
The command line tools surely do have a version to print. You might argue that they have too many to make implementation easy, but you can't really argue that they have none. For instance, answering questions like which syntax of pvRequest is understood, which normative types does it understand.

mdavidsaver, 2016-03-21:
> ... The only versioning is of pvAccess itself.

This seems like an easy answer to me.  Include pvaVersion.h and print EPICS_PVA_*_VERSION.

gregoryraymondwhite, 2016-03-21:
Yes, that would be fine.

commit 43811e25 (2016-03-22): added version to pvaTools

closes #35
closed 2016-03-22 by msekoranja