epics-base/pvAccessCPP #37: testMonitor exits after throwing invalid mutex

dhickin, 2016-03-21:
Server-side start test server:
./testServer

Client-side:
./testMonitorPerformance   -r "record[velocious=true,queueSize=50]field(value)" -s 2764800  -l 1 -i 10 -w 5

Client terminates with
terminate called after throwing an instance of 'epicsMutex::invalidMutex'
  what():  epicsMutex::invalidMutex()
Aborted (core dumped)

Seen with latest (and recent revs). Not seen before major gateway merge (in particular I went back to f0b88e349b0193e3621b76ba17321b9ff09a7ba3)

dhickin, 2016-03-21:
gdb:

[cs04r-sc-serv-47 linux-x86_64]$ gdb ./testMonitorPerformance
...
(gdb) run -s 2764800  -l 10 -i 1000 -w 5
Starting program: /home/jms68266/epicsv4/base/R3.14.12.3/latest/pvAccessCPP/bin/linux-x86_64/testMonitorPerformance -s 2764800  -l 10 -i 1000 -w 5
[Thread debugging using libthread_db enabled]
[New Thread 0x7ffff6d62700 (LWP 20413)]
[New Thread 0x7ffff68d0700 (LWP 20414)]
[New Thread 0x7ffff684f700 (LWP 20415)]
[New Thread 0x7ffff67ce700 (LWP 20416)]
[New Thread 0x7ffff674d700 (LWP 20417)]
[New Thread 0x7ffff66cc700 (LWP 20418)]
[New Thread 0x7ffff664b700 (LWP 20420)]
1 channel(s) of double array size of 2764800 element(s) (0==scalar), 1000 iteration(s) per run, 10 run(s) (0==forever)
[New Thread 0x7fffef48d700 (LWP 20421)]
[New Thread 0x7fffef40c700 (LWP 20422)]
all connected
25.028540 seconds, 39.954 (x 1 = 39.954) monitors/s, data throughput 7.070 Gbits/s
...
24.962797 seconds, 40.060 (x 1 = 40.060) monitors/s, data throughput 7.088 Gbits/s
1 2764800 1000 10 40.114
terminate called after throwing an instance of 'epicsMutex::invalidMutex'
  what():  epicsMutex::invalidMutex()

Program received signal SIGABRT, Aborted.
0x0000003355032625 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.12-1.166.el6_7.7.x86_64 libgcc-4.4.7-16.el6.x86_64 libstdc++-4.4.7-16.el6.x86_64 ncurses-libs-5.7-4.20090207.el6.x86_64 nss-softokn-freebl-3.14.3-23.el6_7.x86_64 readline-6.0-4.el6.x86_64
(gdb) bt
#0  0x0000003355032625 in raise () from /lib64/libc.so.6
#1  0x0000003355033e05 in abort () from /lib64/libc.so.6
#2  0x0000003356cbea7d in __gnu_cxx::__verbose_terminate_handler() ()
   from /usr/lib64/libstdc++.so.6
#3  0x0000003356cbcbd6 in ?? () from /usr/lib64/libstdc++.so.6
#4  0x0000003356cbcc03 in std::terminate() () from /usr/lib64/libstdc++.so.6
#5  0x0000003356cbcd22 in __cxa_throw () from /usr/lib64/libstdc++.so.6
#6  0x00007ffff76a3863 in epicsMutex::lock (this=<value optimized out>)
    at ../../../src/libCom/osi/epicsMutex.cpp:238
#7  0x00007ffff7ceda1d in Lock (this=0x6169d0, __in_chrg=<value optimized out>)
    at /home/jms68266/epicsv4/base/R3.14.12.3/latest/pvDataCPP/include/pv/lock.h:55
#8  ~ChannelProviderFactoryImpl (this=0x6169d0, 
    __in_chrg=<value optimized out>) at ../../src/pva/clientFactory.cpp:29
#9  ChannelProviderFactoryImpl::~ChannelProviderFactoryImpl (this=0x6169d0, 
    __in_chrg=<value optimized out>) at ../../src/pva/clientFactory.cpp:37
#10 0x00007ffff7cf9367 in _M_release (this=0x7ffff7ffbce0, __x=0x616710)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/tr1_impl/boost_sp_counted_base.h:140
#11 ~__shared_count (this=0x7ffff7ffbce0, __x=0x616710)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/tr1/shared_ptr.h:153
#12 ~__shared_ptr (this=0x7ffff7ffbce0, __x=0x616710)
---Type <return> to continue, or q <return> to quit---
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/tr1/shared_ptr.h:358
#13 ~shared_ptr (this=0x7ffff7ffbce0, __x=0x616710)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/tr1/shared_ptr.h:834
#14 ~pair (this=0x7ffff7ffbce0, __x=0x616710)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_pair.h:68
#15 destroy (this=0x7ffff7ffbce0, __x=0x616710)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/ext/new_allocator.h:115
#16 _M_destroy_node (this=0x7ffff7ffbce0, __x=0x616710)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_tree.h:383
#17 std::_Rb_tree<std::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::tr1::shared_ptr<epics::pvAccess::ChannelProviderFactory> >, std::_Select1st<std::pair<std::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::tr1::shared_ptr<epics::pvAccess::ChannelProviderFactory> > >, std::less<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::tr1::shared_ptr<epics::pvAccess::ChannelProviderFactory> > > >::_M_erase (this=0x7ffff7ffbce0, __x=0x616710)
---Type <return> to continue, or q <return> to quit---
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_tree.h:972
#18 0x00007ffff7cf963d in ~_Rb_tree (this=<value optimized out>, 
    __in_chrg=<value optimized out>)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_tree.h:614
#19 std::map<std::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::tr1::shared_ptr<epics::pvAccess::ChannelProviderFactory>, std::less<std::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::tr1::shared_ptr<epics::pvAccess::ChannelProviderFactory> > > >::~map (this=<value optimized out>, __in_chrg=<value optimized out>)
    at /usr/lib/gcc/x86_64-redhat-linux/4.4.7/../../../../include/c++/4.4.7/bits/stl_map.h:87
#20 0x0000003355035ebd in __cxa_finalize () from /lib64/libc.so.6
#21 0x00007ffff7cec3d6 in __do_global_dtors_aux ()
   from /home/jms68266/epicsv4/base/R3.14.12.3/latest/pvAccessCPP/lib/linux-x86_64/libpvAccess.so.4.2.0
#22 0x0000000000000000 in ?? ()
(gdb)

mdavidsaver, 2016-03-21:
Looks like a C++ global dtor ordering problem, and not a new one.  I did remove a comment which I clearly shouldn't have:

> // TODO global static variable (de/initialization order not guaranteed)

mdavidsaver, 2016-03-22:
It seems like there is a fundamental problem with code like:

```c++
static Mutex cfact_mutex;
static ChannelProvider::shared_pointer cfact_shared_provider;
static ChannelProviderFactoryImpl::shared_pointer pva_factory;
```

There is no guarantee that the mutex won't be destroyed before the two shared_ptr<>s.  So a better pattern would be something like:

```c++
static
struct arbitrary_t {
  Mutex mutex;
  ChannelProvider::shared_pointer shared_provider;
  ChannelProviderFactoryImpl::shared_pointer factory;
} cf;
```

where there is only one global ```cf```, and the ordering of it's three members is specified.

msekoranja, 2016-03-23:
@mdavidsaver Isn't ordering *within one compilation* unit defined (as declared)?

msekoranja, 2016-03-23:
I assume that order destruction is OK, but "destroying" a shared_ptr can only decrease use_count.

msekoranja, 2016-03-23:
Should be fixed now.
@dhickin Reopen if you still see this. On my linux box I cannot reproduce it anymore after the fix.

closed 2016-03-23 by msekoranja

mdavidsaver, 2016-03-23:
> Isn't ordering within one compilation unit defined (as declared)?

As far as I know, not by spec.  Based on a quick check, gcc only emits one .init_array entry per compilation unit (by default).  So I would like this holds true.  However, if it did then I don't think @dhickin would see this crash.