epics-base/pvAccessCPP #38: Move pvTools out of pvAccessCPP

dhickin, 2016-03-21:
I think it would be better to move pvTools to a separate repository.

I’m not sure they really belong in pvAccessCPP. I guess the main reason for keeping them there is that they are fairly small and previously there wasn't a strong enough case for the extra overhead in having another repo.

However now moving them out would enable pvTools to use normativeTypes and pvaClient in future, whilst maintaining  the current extremely clean and logical dependencies.

The tools currently all reside in their own source directory so it should be fairly straight-forward.

msekoranja, 2016-03-21:
I would only consider moving eget out (since it has knowledge of normativeTypes).

mdavidsaver, 2016-03-21:
I'll second Matej.  Please don't move pvget/put/info.

anjohnson, 2016-03-22:
I also disagree with moving the pvtools.

Does eget have its own implementations of the normative types? pvAccess doesn't require the normativeTypes module to build IIRC.

msekoranja, 2016-03-22:
eget does does extracting data (according to specification) on it's own. eget existed long before normativeTypesCPP. 
normativeTypeCPP are "just" helpers classes around pvData structures.

dhickin, 2016-03-22:
OK, let's keep them where they are.

Only eget uses normative types (with as Matej says their own implementation). The idea was that eget would be able to use normativeTypesCPP and all the tools could use pvaClientCPP. If we were going to make such a move then now would have been a sensible time if we were considering a versioning scheme.

I do see the advantage in packaging the tools without requiring higher-level libraries, so I'm happy to close this.

closed 2016-03-22 by dhickin