epics-base/pvAccessCPP #40: Not-a-first segmented message expected from the client

mdavidsaver, 2016-03-24:
Another odd one for @msekoranja.  A PVA client connecting to an IOC with qsrv (new dbgroup prototype).  I restarted the client one time and and it got stuck in a reconnect loop with the server spewing this message:

```
2016-03-24T11:54:15.222 Not-a-first segmented message expected from the client at ../../src/remote/codec.cpp:334: 10.5.3.1:56367, disconnecting...
```

I restart both ends and things start up normally.  Haven't been able to reproduce since despite restarting in various orders.

I did do a packet capture while this was happening.  I'll try to attach, and email if this fails.

Note that I haven't yet updated to get the fix for #39.

mdavidsaver, 2016-03-24:
[pva-loop.pcapng.gz](https://github.com/epics-base/pvAccessCPP/files/188182/pva-loop.pcapng.gz) Compressed packet capture.  I've tried to filter out non-PVA packets.  I'll keep the raw capture for a little while, so let me know if I dropped anything relevant.

msekoranja, 2016-03-24:
I see. Server monitor response that sends introspection data looks wrong. Just before correct introspection data I see introspection data (of some pvRequest) squeezed in...

msekoranja, 2016-03-24:
@mdavidsaver Decoded a Structure of a monitor:

`structure
    structure record
        structure _options
            uint queueSize
            boolean atomic
    epics:nt/NTScalar:1.0 a
        double value
        alarm_t alarm
            int severity
            int status
            string message
        time_t timeStamp
            long secondsPastEpoch
            int nanoseconds
            int userTag
        display_t display
            double limitLow
            double limitHigh
            string description
            string format
            string units
        control_t control
            double limitLow
            double limitHigh
            double minStep
    epics:nt/NTScalar:1.0 b
        double value
        alarm_t alarm
            int severity
            int status
            string message
        time_t timeStamp
            long secondsPastEpoch
            int nanoseconds
            int userTag
        display_t display
            double limitLow
            double limitHigh
            string description
            string format
            string units
        control_t control
            double limitLow
            double limitHigh
            double minStep
`

Is this OK?
Data serialized does not seem to match!?

msekoranja, 2016-03-24:
NOTE: server-side implementation trusts the provider to provide the right PVStructure returned by monitr->poll(), i.e. the PVStructure that matches advertised Structure (at monitor initialization).
If it does not, then de-/serialization will fail badly.

mdavidsaver, 2016-03-24:
> Is this OK?

Yes.

> NOTE: server-side implementation trusts the provider

I should only have one structure definition per PV, but I've changed so many things today I can't guarantee that I wasn't doing something stupid at the time.  If I see it again I'll investigate this possibility.

> If it does not, then de-/serialization will fail badly.

Badly as in sending invaild PVA messages, or badly as in memory access violation?

mdavidsaver, 2016-03-24:
> I've changed so many things today

As I think about it.  For a while I had Monitors staying open longer than intended.  I had meant to destroy() it on Channel disconnect, but wasn't.  And I think I did add the "record._options.atomic" field during this time.  Is it possible that PVA reconnected the monitor w/o keeping the new structure description?  What should happen in this situation?

mdavidsaver, 2016-03-24:
Ok, definitely type change.  Just replicated with pvaPy.  Also managed to hit an assert() in BitSet::deserialize() which on reflection shouldn't be an assert.  The question is whether it should be an exception or silently fixed.

```
2016-03-24T18:52:34.038 Not-a-first segmented message expected from the client at ../../src/remote/codec.cpp:334: 10.5.3.1:60015, disconnecting...
error circle2 circle2 connection state DISCONNECTED
A call to 'assert(words.empty() || words.back()!=0)'
    by thread 'TCP-rx' failed in ../../src/misc/bitSet.cpp line 346. 
```

msekoranja, 2016-03-25:
On monitor reconnect init message (callback) is sent with updated Structure.
If you change PVStructure during monitor lifetime and return it via poll() then the client will deserialize with the wrong interface description (Structure). This will result in connection loss (as sequence of "Not-a-first segmented message expected" exception) or client will wait for more data...

Eventually, I could do a valdiation (by reference) `monitorElement->pvStructure->getStructure().get() == monitorStructure.get()` for every monitor.

msekoranja, 2016-03-25:
Another note: once monitor is initialized there is no allocation happening. This includes extending BitSet-s. BitSet-s size is initialized to structure length and never changes.

... until reconnect (reinitialized) is called.
(If reconnect is done w/ the same Structure, and it usually does, old instances are resused - again, no allocation).

mdavidsaver, 2016-03-27:
Is this going to effect other operations as well? (eg. get/put)

It seems to me that there is a fundamental problem here.  IMO if an operation is going to outlive the Channel it is created though, then it must be able to handle type change.

The options I see are 1) an operation should "expire" when the underlying Channel loses connection, or 2) give notification that the type has changed (eg. maybe call monitorConnect() again).

msekoranja, 2016-03-27:
For each request (put, get, monitor) Structure is defined at initialization time (depends on requested fields specified in pvRequest). So, it is a per request thing. After initialization everything is fixed. 
If anything changes on the server side (e.g. channel's Structure) a channel reconnect (and reinitialization) needs to be done. 

For the current implementation 1) and 2) is implemented.

mdavidsaver, 2016-03-27:
> For the current implementation 1) and 2) is implemented.

I'm confused by this.  Do you mean that it is _possible_ to achieve these effects?  It's clear that neither is the default.

msekoranja, 2016-03-27:
When Channel loses connection all its related requests are destroyed. The exception are monitors that gets recreated (monitorCreated will be called again) on the client side when the connection is reestablished. Each request gets a type defined at initialization time.

Changing channel type requires disconnecting the channel. This is something provider needs to do.

mdavidsaver, 2016-04-05:
You're correct that monitorConnect() is called after channel re-connect.  However, type change isn't handled correctly even by "pvget -m".

Run the following using pvaSrv or my new server.  Start "pvget -m" with one ```alias()``` uncommented.  Leave the client running and restart the IOC with the other ```alias()```.

```
cat << EOF > typechange.db 
record(ai, "test:ai") {
   field(VAL, "42.2")
#   alias("test:x")
}
record(longin, "test:li") {
   field(VAL, "4")
   alias("test:x")
}
EOF
```

Going from double to integer I see that monitorConnect() is called after reconnect, but monitorEvent() is not.  If the integer is then made to change (```dbpf test:li 5```), the "Not-a-first segmented message" error occurs.

Also, as the result is closing the TCP connection, and a guaranteed reconnect loop, I think "Not-a-first segmented message" should be logged as something more than a warning.

mdavidsaver, 2016-04-05:
As I think about it, handling Monitor type changes in a race free way is going to be quite difficult with the present Monitor interface.  Consider what has to happen if the type changes between a call to poll() and release().  Now release() has to handle the case where it may be given elements with the old type.

msekoranja, 2016-04-06:
A simple check in release() can solve this problem.
I am more concerned (but it solvable) about giving (via poll()) a new type before (near the same time as) monitorCreated() is called.

msekoranja, 2016-04-06:
I've committed a fix.

mdavidsaver, 2017-12-19:
Released in module version 6.0.0, bundle version 7.0.1

closed 2017-12-19 by mdavidsaver