epics-base/pvAccessCPP #41: caProvider: Channel Get or Monitor fetches wrong fields.

dhickin, 2016-03-29:
caProvider: Channel Get or Monitor fetches wrong fields.

pvget, eget or pvget -m via ca fetches the wrong fields. Issue is presumably in caProvider.

pvinfo gives the fields value,alarm,display,control,valueAlarm.
Doing a pvget -r "field(<field-list>)" ca:///double01 with some field lists fetches the wrong fields. Here's a selection: 

1. alarm fetches value, alarm
2. timeStamp fetches value,alarm,timeStamp
3. value,timeStamp fetches value,alarm,timeStamp
4. value,alarm,display fetches value,alarm,display,valueAlarm
5. display fetches value,alarm,display,valueAlarm
6. control fetches value,alarm,display,control,valueAlarm
7. display,control fetches value,alarm,display,control,valueAlarm
8. alarm,control fetches fetches value,alarm,display,control,valueAlarm
9. timeStamp,control fetches fetches value,alarm,display,control,valueAlarm

Basic rules seem to be:

<ol>
<li>value is fetched whatever you ask for.</li>
<li>If you ask for display or control you always get valueAlarm.</li>
<li>If you ask for valueAlarm you always get display.</li>
<li>If you ask for control you always get display.</li>
<li>If you ask for display, control or valueAlarm you never get timeStamp.</li>
<li>If you don't ask for control display or valueAlarm, but ask for alarm you always get timeStamp.</li>
<li>If you get any ask for everything you get everything except timeStamp.</li>
</ol>

msekoranja, 2016-03-29:
The rule is even simpler. CA talks DBR-s. caProvider examines requested fields and tries to find the DBR that fits your request. Eventually it could do the "stripping". Is would it be really used?

anjohnson, 2016-03-29:
The CA dbr_* types do not provide all combinations of fields, which explains David's specific rules (e.g. no dbr_* types provide both timeStamp and the display/control fields, etc.). I don't think we should try to hide the limitations of CA, so I would not be in favour of making multiple CA requests to try and satisfy a single pva request.

I guess the important question is whether the caProvider should throw exceptions (or otherwise return some kind of error) when asked for impossible combinations of fields? That would make the limitations more visible, but might not fit with the philosophy behind our current code.

dhickin, 2016-03-29:
Although CA talks DBRs, V4 has its own conventions and standards (pvAccess, pvRequest and normative types). caProvider shouldn't just return the closest DBR type it can find it should implement the various V4 standards using CA. If it doesn't strip out unwanted fields it's not implementing pvRequest.

Also it isn't just a case of stripping out fields. A request to get timeStamp and display or a request to get everything doesn't get the timeStamp. Presumably we're trying to match the request to the available DBR types and going with DBR_CTRL_DOUBLE whose structure dbr_ctrl_double has no time stamp. In this instance there's no easy fix. Options are: reject requests for fields don't all lie within a single DBR type or try and patch with multiple requests (or perhaps later a group operation).

dhickin, 2016-03-29:
> Eventually it could do the "stripping". Is would it be really used?
I understand the desire not to spend time implementing something that won't be used (although asking for value + timeStamp and getting  value + timeStamp doesn't seem unreasonable).

Unfortunately we have a standard that says we can pick substructures of the main structure. I expect a lot more people to be trying version 4 in the next few months and pvTools is likely to be the first point of contact. When that happens the errant behaviour of caProvider (and pvaSrv) is likely to be very visible.

anjohnson, 2016-03-29:
If the caProvider is currently returning the values of fields that were not requested then I agree that it probably should be changed since that doesn't match what happens elsewhere. Future releases of pvaSrv shouldn't have to use the DBR types or local CA channels within the IOC (the dbAccess API to the IOC is more flexible than the db_access API) so there is a route to a solution there, eventually.

mdavidsaver, 2017-12-19:
@mrkraimer @anjohnson @sveseli Has/will this issue been addressed?

mrkraimer, 2018-08-02:
This topic address compatibility between using client providers ca and pva.
With changes that have been made to ca provider it has better support, but see

https://github.com/epics-base/pvAccessCPP/blob/master/caProvider.md
For issues related to channel access.

The changes to ca provider make it much more compatible with pvaSrv.
But qsrv has now replaced pvaSrv.
Now there are major compatibility issues.
For examplle:
```
mrk> pvget -p ca -r value,alarm,timeStamp DBRdouble 
DBRdouble
structure 
    double value 9
    alarm_t alarm MAJOR RECORD HIHI
    time_t timeStamp 2018-08-02T09:23:46.055 0

mrk> pvget -p pva -r value,alarm,timeStamp DBRdouble 
DBRdouble
epics:nt/NTScalar:1.0 
    double value 9
    alarm_t alarm MAJOR DEVICE HIHI
    time_t timeStamp 2018-08-02T09:23:46.055 0
    display_t display
        double limitLow -10
        double limitHigh 10
        string description 
        string format 
        string units volts
    control_t control
        double limitLow -9
        double limitHigh 9
        double minStep 0
    valueAlarm_t valueAlarm
        boolean active false
        double lowAlarmLimit -8
        double lowWarningLimit -6
        double highWarningLimit 6
        double highAlarmLimit 8
        int lowAlarmSeverity 0
        int lowWarningSeverity 0
        int highWarningSeverity 0
        int highAlarmSeverity 0
        double hysteresis 0

```

closed 2022-02-23 by anjohnson