epics-base/pvAccessCPP #42: caProvider: Request for part of structure field returns whole structure

dhickin, 2016-03-29:
When doing a channel get or monitor with a request for part of a structure field caProvider provides the whole structure:

For example do a pvget, pvget -m or eget and fetch part of the alarm:
```
$ eget -ir "field(value,alarm.severity)"  ca:///double01
double01
epics:nt/NTScalar:1.0 
    double value 0
    alarm_t alarm
        int severity 3
        int status 3
        string message UDF_ALARM
```

Similarly for other fields.

dhickin, 2016-03-29:
In particular this also applies to the value field of an enum pv:
```
eget -ir "field(value.index)"  ca:///enum01
enum01
epics:nt/NTEnum:1.0 
    enum_t value
        int index 0
        string[] choices [zero,one,two,three]
```

msekoranja, 2016-03-29:
Duplicate of #41

dhickin, 2016-03-29:
I opened as a separate issue to #41 as these could easily end up being done as orthogonal fixes. Pruning the subfields of alarm, timeStamp structures can be done independently of a fix to supply the correct top-level subfields. There's no technical issue in fixing these unlike the case with #41 - it's just a pain in this instance.

commit 362f3c02 (2016-05-19): Return requested fields in caProvider

caProvider matches to closest DBR type. Strip out unrequested fields
and only return requested parts of structure fields. Includes enums.
Check fields are non-null rather than throwing. Check labels non-null to
handle requests for choices field without index.

Fixes #42. Covers most cases of #41.
closed 2016-05-19 by msekoranja