epics-base/pvAccessCPP #43: LGPL Code in pvAccessCPP

anjohnson, 2016-04-05:
According to comments in the source code the file pvAccessCPP/src/utils/wildcard.cpp and its associated header are licensed under the LGPLv2+, which is not compatible with the module's LICENSE file. Either this code **must** be removed and a replacement written, or we have to change the license for pvAccessCPP. I have not checked pvAccessJava, I don't know how similar the code-base is but it might already be using a standard library function.

There is a wild-card string matching function provided in Base-3.14 libCom/misc/epicsString.c (`#include <epicsString.h>`)
```
int epicsStrGlobMatch(const char *str, const char *pattern);
```
However this function only supports the wild-card patterns `*` and `?`, and not the character sets features of the LGPL code — does anyone actually use the `[a-z]` or `[!a-z]` patterns?

I would suggest we change the `Wildcard::wildcardfit()` routine to call the above routine immediately (should be a drop-in replacement, just swap the arguments), and decide whether we need to add the other features later. There may need to be a documentation adjustment to the code that is using this utility (just rpcServer.cpp and pipelineServer.cpp AFAICT).

msekoranja, 2016-04-11:
@gregoryraymondwhite Do you use set patterns?

gregoryraymondwhite, 2016-04-11:
@msekoranja If you mean specifically use of [char-char] in pvlist, then no, I personally don't. But having had a lot of experience now with PV searches using our directory service, I'd say that some installations might find that useful. 

I've included some general examples of searches for PVs we've below. See examples file below. However, in the context of pvlist, since 98% of our PVs are still CA, and pvlist doesn't support "-p ca", and since the development of our Directory Service (ds) (on top of channel finder), all our PV searches go though the DS, which knows all the CA and all the PVA PVs.

[memeExamples.txt](https://github.com/epics-base/pvAccessCPP/files/213714/memeExamples.txt)

msekoranja, 2016-04-12:
Agreed on a meeting to take EPICS base version.

closed 2016-04-14 by msekoranja

anjohnson, 2016-04-14:
@msekoranja I just committed the following additional non-functional changes:

1. Replaced the LGPL header comments in src/utils/pv/wildcard.h with our standard header
2. Edited the comments below that header to remove the description of character ranges that `epicsStrGlobMatch()` doesn't support
3. In wildcard.cpp moved `#include <epicsString.h>` *above* `#define epicsExportSharedSymbols`