epics-base/pvAccessCPP #45: testServer crashes on Windows (static build)

dhickin, 2016-04-14:
On Windows testServer stops working immediately after starting with an unknown software exception after printing

WSAIoctl SIO_GET_INTERFACE_LIST failed 100093

Code is latest (6dc9b0f96f56c18729db00e257e67235988c1cd1). Building against latest pvDataCPP and base R3.14.12.3 (static build).

Happens every time.

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	testServer.exe
  Application Version:	0.0.0.0
  Application Timestamp:	570fa164
  Fault Module Name:	testServer.exe
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	570fa164
  Exception Code:	40000015
  Exception Offset:	000000000005b6b5
  OS Version:	6.1.7601.2.1.0.272.7
  Locale ID:	2057
  Additional Information 1:	972b
  Additional Information 2:	972bb8d937273d3d7d4227a5a1999a78
  Additional Information 3:	08f0
  Additional Information 4:	08f01994ce1efcb439f348c4a26adffc

dhickin, 2016-04-14:
100093 is WSANOTINITIALISED. So successful WSAStartup not yet performed or WSAStartup not called more times than WSACleanup:

 "Either the application has not called WSAStartup or WSAStartup failed. The application may be accessing a socket that the current active task does not own (that is, trying to share a socket between tasks), or WSACleanup has been called too many times."

Not seeing this in Linux. Never seen this before in earlier version on Windows.
Could it be a static initialisation order fiasco bug again?

Doing a dynamic build now to see if that's any better.

dhickin, 2016-04-14:
Same issue with pvaSrv, rpcServer and pvDatabase examples (static builds).

Don't see the problem with dynamic builds.

anjohnson, 2016-04-14:
I saw the first error on Windows the other day when trying the self-tests, but I forget which target I was running at the time. I get a different error earlier today from cygwin when running testServer.exe, testServerContext.exe and testChannelConnect.exe:
```
2016-04-14T10:29:14.109 Error binding socket: Cannot assign requested address.
2016-04-14T10:29:14.141 Error binding socket: Cannot assign requested address.
```

None of the other Windows targets report errors from testServer.exe for me today though (I tried windows-x64, windows-x64-static and native-built windows-x64-mingw).

commit 4e264f22 (2016-04-26): win32: moved osiSockAttach() call before loadConfiguration

should fix #45
closed 2016-04-26 by msekoranja