epics-base/pvAccessCPP #49: PVs of unsigned types converted to signed integer or floating point fields

dhickin, 2016-05-18:
PVs of unsigned types (i.e. the DBF type of the record field/underlying value) are converted to signed integer or floating point fields instead of unsigned integer fields by caProvider as follows

UCHAR->byte not ubyte
USHORT->int not ushort
ULONG->double not uint

So for example if we run testTop we get 
```
pvget -p ca -r "value"  ubyte01 ubyteArray01 ushort01 ushortArray01 uint01 uintArray01 
ubyte01
epics:nt/NTScalar:1.0 
    byte value 0

ubyteArray01
epics:nt/NTScalarArray:1.0 
    byte[] value []

ushort01
epics:nt/NTScalar:1.0 
    int value 0

ushortArray01
epics:nt/NTScalarArray:1.0 
    int[] value []

uint01
epics:nt/NTScalar:1.0 
    double value 0

uintArray01
epics:nt/NTScalarArray:1.0 
    double[] value []
```

In contrast pvaSrv works as expected
```
$ pvget  -r "value"  ubyte01 ubyteArray01 ushort01 ushortArray01 uint01 uintArray01 #float01 double01
ubyte01
epics:nt/NTScalar:1.0 
    ubyte value 0

ubyteArray01
epics:nt/NTScalarArray:1.0 
    ubyte[] value []

ushort01
epics:nt/NTScalar:1.0 
    ushort value 0

ushortArray01
epics:nt/NTScalarArray:1.0 
    ushort[] value []

uint01
epics:nt/NTScalar:1.0 
    uint value 0

uintArray01
epics:nt/NTScalarArray:1.0 
    uint[] value []
```

Channel access turns the DBF types into the offending DBR types, so looks like a fundamental channel access issue, and as such may not be easily solvable.

anjohnson, 2016-05-18:
Channel Access has no unsigned types, so there is no way to distinguish between signed and unsigned integers of the same sizes when going through the local CA → rsrv → db_access interface which the current pvaSrv code calls. @mdavidsaver's replacement server interface calls the newer dbAccess interface which uses the same data types as the IOC itself, so this issue will be solved by that code (but only on Base-3.16.x).

It should be possible to modify the pvaSrv code to call the dbAccess API and thus solve the issue for users of Base-3.14.12.x and 3.15.x as well, but that might be quite a bit of work and may not be worth the effort unless a specific site needs it and is willing to do the implementation.

dhickin, 2016-05-18:
@anjohnson: Sorry, not sure I follow you. It's a caProvider issue not a pvaSrv issue - pvaSrv handles things correctly.  I don't see how a change to pvaSrv or a new implementation of it can solves the problem. For one thing the issue exists even when pvaSrv is not running on an IOC.

As I said in the original, I get that it's a channel access limitation and as such I understand may not be solvable. It is, though, a departure from expected behaviour (and leads to different types over pva and ca) so I've logged it - we may of course decide to do nothing.

mdavidsaver, 2016-05-18:
> but only on Base-3.16.x

The QSRV code is split into single and group access components.  Only the group access part needs 3.16.  With some ifdef to skip this, the remainder could build with 3.15.

anjohnson, 2016-05-18:
@dhickin I misread your bug report, but my reasoning is still mostly on-point. Any CA client must use the DBF_* and DBR_* type codes defined in db_access.h (since those values are what is passed over the wire when using the CA protocol) and there are no unsigned codes there. Thus both the IOC's DBF_CHAR and DBF_UCHAR field type codes from dbFldTypes.h are mapped to the same DBF_CHAR value from db_access.h by the CA server (this actually happens in the db_access API). Further-more the IOC's DBF_ULONG field type gets mapped to a Channel Access DBF_DOUBLE type. When the caProvider sees a PV with field type DBF_CHAR it has no way to know whether the corresponding IOC field is really signed or unsigned.

The advantage that pvaSrv has (which I forgot about when writing my previous response) is that it can see the true underlying field type by calling the dbAccess API to the IOC; it only needs the db_access API for monitor update notifications, thus it can always serve the correct pvData type for the field over the pvAccess protocol. The caProvider though doesn't have any way to distinguish between the two possible underlying field types, so from your report it apparently always picks the signed one. I guess there might be a way for a caller that knows to provide it with a hint in the pvRequest object, but I don't know how feasible that would be in practice.

ralphlange, 2016-05-19:
To rephrase @anjohnson 's last comment:

- Channel Access has no unsigned data types, the database has unsigned field types.
- pvaSrv is directly on top of the database, so it knows the true type and can convert accordingly.
- caProvider is on top of the CA client, so it only sees the limited set of CA data types.

Even though the CA client API has a call that claims to "return the native type in the server of the process variable", this actually returns the field type on the server *after* it has been matched to CA types. If you access a record with a `DBF_ULONG` field, the clients sees it as `DBF_DOUBLE`.

Can't the client application trigger conversion in any case by setting the data type of the corresponding element in the pvData structure it requests? Since this is all happening in the client, I see no advantage in using an additional pvRequest that is being interpreted in a different layer on the client side. The data type of the target structure should IMHO suffice.

dhickin, 2016-05-19:
Yes. I understand the points you and Andrew are making. 

<blockquote>Can't the client application trigger conversion in any case by setting the data type of the corresponding element in the pvData structure it requests?</blockquote>

I'm not clear what you mean. Do you mean can the client request the type of the value field through the pvRequest?  If so then no, not through the existing syntax because the type of field.value in the pvRequest is structure, not a scalar. I guess that could be an option as a subfield of field.value, but that's unspecified. In any case it's not clear how a client would no the type. GetField will return the same signed/floating point type as a get or monitor. If you could get that information we could solve the problem, but I know of no way of doing this through channel access.

It may well be that we leave this unsolved. Similarly with the issue with requests for timestamps with property data.

Note that this issue and the issue with requests for timestamps with property data are the only two issues I can't solve in terms of returning structures of the correct type. I can otherwise now make the get/monitor fields return exactly the fields promised by GetField, pvaSrv and caProvider behaviour can be made identical and any requested  structure can be obtained in both cases.

ralphlange, 2016-05-19:
>  I'm not clear what you mean. Do you mean can the client request the type of the value field through the pvRequest?

No, I don't mean by a pvRequest structure.
Can't the client provide a structure for pvAccess to fill in, so that pvCopy is called to copy the matching parts over, converting as necessary? If so, can't the client provide the structure with the type it wants to?

anjohnson, 2017-08-30:
Closing this issue, I don't believe it can be solved (and it's been more than a year...).

closed 2017-08-30 by anjohnson