epics-base/pvAccessCPP #50: client ca provider monitor provider

mrkraimer, 2016-06-07:
When a client tries to create a monitor that uses provider ca, a failure occurs when monitor is started.

mrkraimer, 2016-06-07:
Bad title. 
Should have said
"Monitors do not work when client requests provider ca"

arkilic, 2016-06-07:
Nice to have you back @mrkraimer 👍

mrkraimer, 2016-06-08:
I have added more debug output to
https://github.com/mrkraimer/pvaClientCPP
and
https://github.com/mrkraimer/exampleCPP

To demonstrate the monitor problem build against the above and then

1) Start the example database as part on a V3 IOC
mrk> pwd
/home/epicsv4/master/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd

2) Then run monitorForever.
The following works just fine
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/monitorForever pva DBRdouble00 false.

But the following crashes
mrk> bin/linux-x86_64/monitorForever ca DBRdouble00 false
provider "ca" pvaSrv false caSrv true channelName DBRdouble00 debug false
_____monitorForever starting_______
Segmentation fault (core dumped)

To see debug info run
mrk> valgrind bin/linux-x86_64/monitorForever ca DBRdouble00 true

mrkraimer, 2016-06-08:
I have spent some more time looking at the problem.
It was in pvaClientCPP NOT in pvAccessCPP.

The fix is

diff --git a/src/pvaClientMonitor.cpp b/src/pvaClientMonitor.cpp
index ebfb361..6bdd970 100644
--- a/src/pvaClientMonitor.cpp
+++ b/src/pvaClientMonitor.cpp
@@ -240,7 +240,7 @@ Status PvaClientMonitor::waitConnect()
 {
     if(PvaClient::getDebug()) cout << "PvaClientMonitor::waitConnect\n";
     if(connectState==connected) {
-         if(connectStatus.isOK()) connectState = connectIdle;
+         if(!connectStatus.isOK()) connectState = connectIdle;
          return connectStatus;
     }

I am closing this issue.

Sorry Matej!!!!

closed 2016-06-08 by mrkraimer