epics-base/pvAccessCPP #53: caProvider can crash after CAClientFactory::stop() is called.

mrkraimer, 2016-06-09:
The problem is that 
CAChannel::disconnected() can be called after CAChannel::~CAChannel()

The following fixes the problem.

git diff src/ca/caChannel.cpp
diff --git a/src/ca/caChannel.cpp b/src/ca/caChannel.cpp
index 2e51da1..1fd068a 100644
--- a/src/ca/caChannel.cpp
+++ b/src/ca/caChannel.cpp
@@ -241,6 +241,7 @@ void CAChannel::connected()
 void CAChannel::disconnected()
 {
     EXCEPTION_GUARD(channelRequester->channelStateChange(shared_from_this(), Channel::DISCONNECTED));
+    channelID = 0;
 }
 
 CAChannel::CAChannel(std::string const & _channelName,
@@ -278,6 +279,7 @@ void CAChannel::activate(short priority)
 CAChannel::~CAChannel()
 {
     PVACCESS_REFCOUNT_MONITOR_DESTRUCT(caChannel);
+    if(channelID!=0) ca_clear_channel(channelID);
 }
 
I want Matej to verify that this is a valid fix for the problem.

msekoranja, 2016-06-09:
I see a problem when channel is not destroyed using destroy(). Note that it is not required to be. When it is not a CA callback causes SIGSEG if channel is freed from memory.

Setting channelID to zero is not way to do it (lack of thread synchronization). What I would do - I would simply call  ca_clear_channel(channelID); in CAChannel::~CAChannel(). There is a possible duplicate call of ca_clear_channel, but it results in silent error returned by the call.

mrkraimer, 2016-06-10:
I removed the call set clientID to 0.
Code is now:

diff --git a/src/ca/caChannel.cpp b/src/ca/caChannel.cpp
index 2e51da1..ea18331 100644
--- a/src/ca/caChannel.cpp
+++ b/src/ca/caChannel.cpp
@@ -240,7 +240,9 @@ void CAChannel::connected()
 
 void CAChannel::disconnected()
 {
+std::cout << "CAChannel::disconnected()\n";
     EXCEPTION_GUARD(channelRequester->channelStateChange(shared_from_this(), Channel::DISCONNECTED));
+//    channelID = 0;
 }
 
 CAChannel::CAChannel(std::string const & _channelName,
@@ -278,6 +280,7 @@ void CAChannel::activate(short priority)
 CAChannel::~CAChannel()
 {
     PVACCESS_REFCOUNT_MONITOR_DESTRUCT(caChannel);
+    if(channelID!=0) ca_clear_channel(channelID);
 }

But then the crash again occurs:

_____testRAII done_______
after pvaClient looses context
CAChannel::disconnected()
==20426== 
==20426== Process terminating with default action of signal 6 (SIGABRT)
==20426==    at 0x681CA98: raise (in /usr/lib64/libc-2.22.so)
==20426==    by 0x681E699: abort (in /usr/lib64/libc-2.22.so)
==20426==    by 0x5FDCAEC: __gnu_cxx::__verbose_terminate_handler() (in /usr/lib64/libstdc++.so.6.0.21)
==20426==    by 0x5FDA935: ??? (in /usr/lib64/libstdc++.so.6.0.21)
==20426==    by 0x5FDA980: std::terminate() (in /usr/lib64/libstdc++.so.6.0.21)
==20426==    by 0x5FDAB98: __cxa_throw (in /usr/lib64/libstdc++.so.6.0.21)
==20426==    by 0x5CFC9AA: epicsMutex::lock() [clone .part.3] (epicsMutex.cpp:257)
==20426==    by 0x5D17F19: epicsMutex::lock() (epicsMutex.cpp:248)
==20426==    by 0x5A9D7E8: epicsGuard (epicsGuard.h:71)
==20426==    by 0x5A9D7E8: ca_clear_channel (access.cpp:380)
==20426==    by 0x5225CBE: epics::pvAccess::ca::CAChannel::~CAChannel() (caChannel.cpp:283)
==20426==    by 0x522A264: ~CAChannel (caChannel.cpp:284)
==20426==    by 0x522A264: operator() (shared_ptr.h:285)
==20426==    by 0x522A264: std::tr1::_Sp_counted_base_impl<epics::pvAccess::ca::CAChannel*, std::tr1::_Sp_deleter<epics::pvAccess::ca::CAChannel>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (shared_ptr.h:257)
==20426==    by 0x4E7B977: _M_release (shared_ptr.h:141)
==20426==    by 0x4E7B977: ~__shared_count (shared_ptr.h:341)
==20426==    by 0x4E7B977: ~__shared_ptr (shared_ptr.h:541)
==20426==    by 0x4E7B977: ~shared_ptr (shared_ptr.h:985)
==20426==    by 0x4E7B977: epics::pvaClient::PvaClientChannel::~PvaClientChannel() (pvaClientChannel.cpp:201)

I pushed changes in mrkraimer github for both pvaClientCPP and exampleCPP.
To reproduce the problem 

In one window
mrk> pwd
/home/epicsv4/master/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 

and in another window:
mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/testRAII ca  DBRdouble00 1

It ends with
_____testRAII done_______
after pvaClient looses context
CAChannel::disconnected()
CA.Client.Exception...............................................
    Warning: "Identical process variable names on multiple servers"
    Context: "Channel: "DBRdouble00", Connecting to: 10.0.0.37:5064, Ignored: localhost.localdomain:5064"
    Source File: ../cac.cpp line 1298
    Current Time: Fri Jun 10 2016 08:00:36.366978676
..................................................................
CA.Client.Exception...............................................
    Warning: "Identical process variable names on multiple servers"
    Context: "Channel: "DBRdouble00", Connecting to: localhost.localdomain:5064, Ignored: 192.168.124.1:5064"
    Source File: ../cac.cpp line 1298
    Current Time: Fri Jun 10 2016 08:00:36.367050503
..................................................................
Segmentation fault (core dumped)

To see details
mrk> valgrind bin/linux-x86_64/testRAII ca  DBRdouble00 1 >& temp
Then look at temp

mrkraimer, 2016-06-10:
I now see that setting
channelID = 0;
in 
CAChannel::disconnected()
causes reconnect to fail.

To see this start the exampleDatabase as described earlier
then run
mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/getForever ca DBRdouble00
Then exit the database.
The following appears
CA.Client.Exception...............................................
    Warning: "Virtual circuit disconnect"
    Context: "localhost.localdomain:5064"
    Source File: ../cac.cpp line 1215
    Current Time: Fri Jun 10 2016 09:44:35.542978318
..................................................................
CAChannel::disconnected()
error DBRdouble00 DBRdouble00 connection state DISCONNECTED
CAChannel::disconnected() after calling channelStateChange DBRdouble00
Then restart the example database.
The following appears
Segmentation fault (core dumped)

Before setting 
channelID = 0;
reconnect  worked.

msekoranja, 2016-06-10:
~CAChannel should contain only

ca_clear_channel(channelID);

no if (channelID != 0) condition check.
You can try this out, anyway I will commit a fix later.

mrkraimer, 2016-06-10:
I tried both of the following:
diff --git a/src/ca/caChannel.cpp b/src/ca/caChannel.cpp
index 2e51da1..9d2e2ac 100644
--- a/src/ca/caChannel.cpp
+++ b/src/ca/caChannel.cpp
@@ -278,6 +278,8 @@ void CAChannel::activate(short priority)
 CAChannel::~CAChannel()
 {
     PVACCESS_REFCOUNT_MONITOR_DESTRUCT(caChannel);
+    threadAttach();
+    ca_clear_channel(channelID);
 }
 
 
mrk> git diff caChannel.cpp
diff --git a/src/ca/caChannel.cpp b/src/ca/caChannel.cpp
index 2e51da1..9d2e2ac 100644
--- a/src/ca/caChannel.cpp
+++ b/src/ca/caChannel.cpp
@@ -278,6 +278,8 @@ void CAChannel::activate(short priority)
 CAChannel::~CAChannel()
 {
     PVACCESS_REFCOUNT_MONITOR_DESTRUCT(caChannel);
+    threadAttach();
+    ca_clear_channel(channelID);
 }

With either I get

mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/testRAII ca  DBRdouble00 2
...
_____testRAII done_______
after pvaClient looses context
...
Segmentation fault (core dumped)

mrkraimer, 2016-06-10:
Sorry the previous message showed two identical diffs.
The other one I tries was
diff --git a/src/ca/caChannel.cpp b/src/ca/caChannel.cpp
index 2e51da1..f530931 100644
--- a/src/ca/caChannel.cpp
+++ b/src/ca/caChannel.cpp
@@ -278,6 +278,7 @@ void CAChannel::activate(short priority)
 CAChannel::~CAChannel()
 {
     PVACCESS_REFCOUNT_MONITOR_DESTRUCT(caChannel);
+    ca_clear_channel(channelID);
 }

Thus without threadAttach();.

But same ending with seg fault.

mrkraimer, 2016-06-11:
I spent some more time looking at the problem.
First observation is that calling CAChannel::destroy before CAChannel::~CAChannel() 
causes a crash.
I suspect that calling ca_clear_channel twice for the same clientID may cause the problem.

I then I created a minimal example using pvaClientCPP that causes a crash.

To run it get the latest from
https://github.com/mrkraimer/pvaClientCPP
and
https://github.com/mrkraimer/exampleCPP

I also made the following changes to pvAccessCPP (no commits or pushes)
mrk> git diff src/ca/caChannel.cpp
diff --git a/src/ca/caChannel.cpp b/src/ca/caChannel.cpp
index 2e51da1..56fbc53 100644
--- a/src/ca/caChannel.cpp
+++ b/src/ca/caChannel.cpp
@@ -240,6 +240,7 @@ void CAChannel::connected()
 
 void CAChannel::disconnected()
 {
+ std::cout << "CAChannel::disconnected()\n";
     EXCEPTION_GUARD(channelRequester->channelStateChange(shared_from_this(), Channel::DISCONNECTED));
 }
 
@@ -277,7 +278,9 @@ void CAChannel::activate(short priority)
 
 CAChannel::~CAChannel()
 {
+std::cout << "CAChannel::~CAChannel() " << channelName << std::endl;
     PVACCESS_REFCOUNT_MONITOR_DESTRUCT(caChannel);
+    ca_clear_channel(channelID);
 }
 
 
@@ -454,6 +457,7 @@ void CAChannel::printInfo(std::ostream& out)
 
 void CAChannel::destroy()
 {
+std::cout << "CAChannel::destroys()\n";
     threadAttach();
 
     Lock lock(requestsMutex);

mrk> git diff src/ca/caProvider.cpp
diff --git a/src/ca/caProvider.cpp b/src/ca/caProvider.cpp
index 0ec6d3c..1ee2a94 100644
--- a/src/ca/caProvider.cpp
+++ b/src/ca/caProvider.cpp
@@ -232,6 +232,7 @@ void CAClientFactory::start()
 
 void CAClientFactory::stop()
 {
+std::cout << "CAClientFactory::stop\n";
     Lock guard(mutex);
 
     if (factory.get())
mrk> 

Then in exampleCPP
Start exampleDatabase
mrk> pwd
/home/epicsv4/master/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 

and then

mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/getForever ca DBRdouble00 false
provider "ca" pvaSrv false caSrv true channelName DBRdouble00 debug false
_____getForever starting__  provider ca channelName DBRdouble00
exCAChannel::~CAChannel() DBRdouble00
exception channel DBRdouble00 PvaClientChannel::connect DBRdouble00 not connected
CAClientFactory::stop
mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/getForever ca DBRdouble00 false
provider "ca" pvaSrv false caSrv true channelName DBRdouble00 debug false
_____getForever starting__  provider ca channelName DBRdouble00
value 0
exit
Type exit to stop: 
CAClientFactory::stop
CAChannel::disconnected()
CAChannel::~CAChannel() DBRdouble00
Segmentation fault (core dumped)
mrk> 

But notice that

mrk> bin/linux-x86_64/getForever pva DBRdouble00 false

works just fine.
Thus problem is with provider ca but not with provider pva.

msekoranja, 2016-06-13:
The second case SIGSEG I am getting is due to different order of destruction. Once the process ends, the order of destruction is different as expected. ca context is destroyed first and then CAChannel that calls ca_clear_channel. Looking into that...

mrkraimer, 2016-06-13:
Here is a possible clue

*
 *  ca_clear_channel ()
 *
 * a known defect here is that there will be a
 * crash if they destroy the channel after destroying
 * its context
 */

This is from
epics/base-3.15.3/src/ca/client/access.cpp

Maybe the following is a solution.

In caProvider.h
replace
typedef std::map<void*, Channel::weak_pointer> ChannelList;
by
typedef std::map<void*, CAChannel::weak_pointer> channelList;

Then CAChannelProvider::createChannel needs to add the newly created CAChannel to channelList.

caChannel.h needs a new variable.
bool isDestroyed;

It is initiaized to false.
CAChannel::destroy sets to to true and also allows destroy to be called mutiple times.
CAChannel::~CACChannel has the code
if(!isDestroyed) ca_clear_channel(channelID);

mdavidsaver, 2016-06-13:
> I suspect that calling ca_clear_channel twice for the same clientID may cause the problem.

At the risk of stating the obvious ```chid``` is really a pointer.  So ```ca_clear_channel(chid)``` would have to be exceptionally clever to avoid a double free in this situation.

```c
typedef struct oldChannelNotify *chid;
```

closed 2016-06-14 by msekoranja

mrkraimer, 2016-06-15:
Looking good!!!
I did some quick tests and looks like crashes no longer occur.

commit 3a73cb68 (2016-06-14): ca provider destruction fixed

resolves #53