epics-base/pvAccessCPP #54: ca provider has problems when used in a V3 IOC.

mrkraimer, 2016-06-20:
The following problems occur when CAClientFactory::start() is called during initialization.

1) When the ioc exits pvaClient calls CAClientFactory::stop();
If any client is connected to a channel via provider ca a crash occurs.
This happens even if clients are only connected to channels in other IOCs.

2) If any client is connected to a channel via provider ca and the channel is in the same IOC,
then none of get, put, and monitor work.
Each just returns the original value.
If the channel is in another IOC each does work.

exampleCPP now has tests that show the problems.

To run the examples:

Start the following:

mrk> pwd
/home/epicsv4/master/exampleCPP/exampleLink/iocBoot/doubleArrayExternal
mrk> ../../bin/linux-x86_64/exampleLink st.cmd 

In
mrk> pwd
/home/epicsv4/master/exampleCPP/exampleLink/iocBoot/exampleLink

There are files:
st.ca
st.extca

If 
mrk> ../../bin/linux-x86_64/exampleLink st.ca

is started then both problems 1) and 2) can be seen.
problem 1) occurs when exit is entered.

If 
mrk> ../../bin/linux-x86_64/exampleLink st.extca 
is started then problem 1) occurs when exit is entered.

Once either is started then run the following:

mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/monitorForever pva exampleMonitorLink

Then also issue 

mrk> pvput examplePutLink 5 1 2 3 4 5
and
mrk> pvget -r "record[process=true]field()" exampleGetLink

You will see that problem 2) occurs when st.ca is used but not when st.extca is used.

msekoranja, 2016-06-22:
1) Calling CAClientFactory::stop() will explicitly call ca_context_destroy() and this will cause a crash.
2) (wow, that's interesting...)

mdavidsaver, 2017-12-19:
@mrkraimer @anjohnson @sveseli Has this issue been addressed?

anjohnson, 2018-01-27:
I'm not aware that this has been addressed, but I have a some ideas as to why it might be happening, related to the local CA provider inside the IOC. Over Christmas I managed to cause a crash on shutdown of an IOC with @mdavidsaver 's lnkCa implementation for what may be similar reasons. I'm not promising I will have time to get to this soon though.

anjohnson, 2018-08-01:
If running inside an IOC process, creating a CA context before `iocInit()` calls `dbCaLinkInit()` which calls `dbServiceIOInit()` will prevent local CA channels using this context from connecting up through the short-circuit mechanism (which probably means they will never connect). If you want to create the context in an initHook use initHookAfterCaLinkInit or later. Oh, and even though local channels with then connect instantly and call your connection callback, you must **not** do any I/O through them until the IOC is actually running, because locksets aren't set up yet and doing any I/O too soon will cause the db_access.c code to try to lock a record that has no lockset...

Shutdown was another area which I found problems when using local CA channels. I fixed LP: #743076 back in February (on all Base branches) which was one important issue, but another as-yet unfixed problem is that any test programs using local CA channels that call `testdbCleanup()` in dbUnitTest.c will see segfaults unless the call to `db_cleanup_events()` is removed from that routine, or some other change is added to prevent local channels from using dbEvent.c resources that get deleted during that clean-up.

anjohnson, 2022-02-23:
`CAClientFactory::stop()` now does nothing, cleanup is handled in the destructors.

closed 2022-02-23 by anjohnson