epics-base/pvAccessCPP #55: pvAccess does not support on-line delete

mrkraimer, 2016-06-21:
pvDatabase provides the ability for a PVRecord to be deleted after startPVAServer is called.
However there is no way to notify clients.

exampleCPP can be used to show the problem.

Start:
mrk> pwd
/home/epicsv4/master/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd

Then in another window:
mrk> pwd
/home/epicsv4/master/exampleCPP/exampleClient
mrk> bin/linux-x86_64/getForever pva PVRdouble

press the enter key and you will see a new get request issued.

In another window

pvput -r "argument.recordName" PVRremoveRecord PVRdouble

New clients will no longer be able to connect to PVRdouble.
Existing clients are still connected and still work.
Go back to getForever and press enter and you will see that it still works.

msekoranja, 2016-06-22:
This was implemented for pva2pva gateway and Michael uses it. Your channel implementation just needs to call channelRequester->channelStateChange w/ DISCONNECTED/DESTROYED state.

msekoranja, 2016-06-22:
Closing, please reopen if needed.

closed 2016-06-22 by msekoranja