epics-base/pvAccessCPP #58: Channel::destroy() vs. Channel::~Channel()

anjohnson, 2016-10-13:
Following up from a recent [tech-talk conversation](http://www.aps.anl.gov/epics/tech-talk/2016/msg01780.php), the apparent need to call Channel::destroy() to tear down a channel connection is probably an anachronism to C++ programmers. Just because Java may require the user to call destroy() we shouldn't make our C++ API follow that behavior. Can this be fixed please?

msekoranja, 2016-10-14:
Since we use shared pointers you cannot (must not) delete on Channel. Destroy does disconnect but does not delete the object. This is done by the shared_ptr when reference drops to 0. However, calling the destroy() is not necessary (RTAII principle is used) - when reference drops to 0 everything is cleaned up.

msekoranja, 2016-10-14:
Note: be careful when implementing xxxRequester classes. Do not have a strong shared_ptr to the channel field there. If you have it you will introduce a cycling dependency: Channel -> xxxRequester -> Channel. If you have this, you will need to call destroy.

Use wear_ptr in xxxRequester or do not use it at all - all the callback have caller (e.g. channel) paramater.

closed 2016-10-14 by msekoranja

ralphlange, 2016-10-14:
Maybe "destroy" is a misleading name if the method does not destroy the object, but only disconnects.
What about "release"? (As "disconnect"would also be misleading...)

msekoranja, 2016-10-14:
destroy() actually releases all the resources (all it can).
Yes, release() would be a better name.

mrkraimer, 2016-10-14:
> Note: be careful when implementing xxxRequester classes. Do not have a
> strong shared_ptr to the channel field there. If you have it you will
> introduce a cycling dependency: Channel -> xxxRequester [...]

Starting with release 4.6 pvaClientCPP is RAII compliant.
This if a client uses pvaClientCPP it does not normally have to call any 
destroy methods of pvAccess.
If however it call methods like
epics::pvAccess::Channel::shared_pointer getChannel();
then it must worry about destroy.

mrkraimer, 2016-10-14:
>
> destroy() actually releases all the resources (all it can).
> Yes, release() would be a better name.

Is it really a better name?
Also this change breaks lots of code.

ralphlange, 2016-10-14:
It that demonstrated in the examples? It should be.

ralphlange, 2016-10-14:
How can it break lots of code if it doesn't need to be called?

mrkraimer, 2016-10-14:
>
> It that demonstrated in the examples? It should be.

All the examples assume RAII, i. e. none of the examples call any 
destroy methods.
The documentation does mention RAII compliance.
Release 4.5 of pvaClientCPP was not RAII compliant and had destroy methods.
With release 4.6 the destroy methods are all declared as
void destroy()  EPICS_DEPRECATED {}

dhickin, 2016-10-14:
> How can it break lots of code if the method doesn't need to be called?
Even if it doesn't need to be called it still might be being called and would therefore break code.

I don't think that there's much difference between destroy and release as names. C++ matches Java so a change would bring them out of sync or force a Java change that would break code. Destroy is a common name throughout code. API stability is important so I suggest leaving the name alone.

ralphlange, 2016-10-14:
>> How can it break lots of code if the method doesn't need to be called?
> Even if it doesn't need to be called it still might be being called and would therefore break code.
> [...] API stability is important so I suggest leaving the name alone.

Agreed.
Still: an explanation saying "destroy() does not destroy the object, it only releases all the resources" is a strong hint that "destroy()" is a bad name, and "release()" is a better one.

ralphlange, 2016-10-14:
On Fri, Oct 14, 2016 at 1:04 PM, Marty Kraimer <notifications@github.com>
wrote:

> On 10/14/2016 06:56 AM, Ralph Lange wrote:
>
> All the examples assume RAII, i. e. none of the examples call any
> destroy methods.
> The documentation does mention RAII compliance.
> Release 4.5 of [...]

That's fine, then.

However it seems to contradict @mdavidsaver's statement in the original
tech-talk thread, where he wrote:

> Calling Channel::destroy() is mandatory.

msekoranja, 2016-10-14:
I have already sent an email to @mdavidsaver regarding this - his ChannelRequester class have a string shared_ptr to the Channel making a cycling dependency.

mdavidsaver, 2016-10-14:
On 10/14/2016 07:30 AM, Matej Sekoranja wrote:
> I have already sent an email to @mdavidsaver
> <https://github.com/mdavidsaver> regarding this - his ChannelRequester
> class have a string shared_ptr to the Channel making a cycling dependency.

I wouldn't be surprised it does. but this is beside the point.  The
issue I saw with the PVA client provider, *not* my test code.  This was
one of the reasons I then spent time writing all that test code.

The best hint I can find atm. is to point out that while
InternalClientContextImpl::m_channelsByCID is keeping weak refs,
SimpleChannelSearchManagerImpl::m_channels keeps strong refs.

Also, I mis-spoke.  I'm certainly *not* advocating calling ~Channel
directly.  Rather, I found that the returned shared_ptr<Channel> was not
unique(), or at least not always, so releasing my (caller's) ref. would
not be enough to destroy it.

mrkraimer, 2016-10-14:
>
> > >
> > > It that demonstrated in the examples? It should be.
> > >
> >
> > All the examples assume RAII, i. e. none of the examples call any
> > destroy methods.
> > The documentation does mention RAII compliance.
> > Release 4.5 of pvaClientCPP was not RAII compliant and had destroy 
> methods.
> > With release 4.6 the destroy methods are all declared as
> > void destroy() EPICS_DEPRECATED {}
> >
>
> That's fine, then.
>
> However it seems to contradict @mdavidsaver's statement in the original
>
> > Calling Channel::destroy() is mandatory.

The pvaClientCPP implementation calls this.

mdavidsaver, 2016-10-14:
I'm regretting that I didn't keep better notes about this behavior.  I don't remember exactly when I encountered it, probably fall 2015, around the time that I discovered that server side channel close wasn't implemented.  Well before I wrote the test code @msekoranja mentions.

At the time I found it annoying, but didn't consider it enough of a bug to put in the public issue tracker.  Rather like [1]  Also, it seemed like the kind of behavior which would trigger another java/c++ "compatibility" argument.  I find these incredibly frustrating, so my threshold to risk starting one is high.

I'll try to do better about reporting these kind of issues in future.

[1] https://github.com/mdavidsaver/pva2pva/blob/master/p2pApp/moncache.cpp#L99

mdavidsaver, 2016-10-14:
> With release 4.6 the destroy methods are all declared as void destroy() EPICS_DEPRECATED {}

This method is helpful in some situations, rather like ```shutdown()``` for a tcp socket.  It just shouldn't be mandatory to avoid leaking Channels.