epics-base/pvAccessCPP #59: provider ca fails for string array

mrkraimer, 2016-11-30:
The following demonstrates the problem:
mrk> caput DBRstringArray01 5 aaa bbb ccc ddd eee
Old : DBRstringArray01 10          
New : DBRstringArray01 10 5 aaa bbb ccc ddd eee         
mrk> pvget -p ca DBRstringArray01
DBRstringArray01 0
mrk> pvget DBRstringArray01
DBRstringArray01 1 5 aaa bbb ccc ddd eee
mrk> caget DBRstringArray01
DBRstringArray01 10 5 aaa bbb ccc ddd eee

ralphlange, 2016-11-30:
Sorry, but I don't understand the setup.
What server are your commands working against?

Note that your first caput is *not*  writing an array (it would need the '-a'option to do that). It writes a single string containing "5 aaa bbb ccc ddd eee".

mrkraimer, 2016-11-30:
You are correct.
I am working on something else that uses the ca provider from pvAccess and found problems.
I will close this issue.

Sorry!!

Marty

closed 2016-11-30 by mrkraimer

anjohnson, 2016-11-30:
>  It writes a single string containing "5 aaa bbb ccc ddd eee".

This behavior of caput (combining multiple value arguments when writing scalars) doesn't appear to be documented anywhere — I wasn't aware of it, but I do see that the code implements it. It would be nice to adjust the usage() string in caput.c to explain.

ralphlange, 2016-12-06:
Yes, it's always been that way.
The original requirement (as I was told) was that it should be possible to use `caget` output as `caput` input. That's the reason for the (ignored) size argument of `caput -a` and the ability to use spaces in strings without having to add quotes.

anjohnson, 2016-12-06:
I just documented this in CAref.html and caput.c, and resynchronized all the command-line option tables with the catools program usage strings.