epics-base/pvAccessCPP #62: ChannelPut::put has problem with V3 busy record

mrkraimer, 2017-04-27:
Summary:

ChannelPut::put always does a put with a callback.
channelAccess implements put with and without a callback.
This presents a problem for V3 records like the busy record.

Details:

The problem was discovered using caqtdm to control the areaDetector simDetector.

There are two caqtdm message buttons to start and stop acquisition.
Each button is connected to the same channel, which is a busy record.
When the start button is pressed a 1 is sent to the channel.
When the stop button is pressed a 0 is sent to the channel.
When a 1 is sent to the busy record it becomes active and does complete processing until
some client send a 1 to the record.

If the client that sends 1 via a put callback then the put will not complete until
some client sends a 0.

caqtdm can be configured to use:
1) cadef.h supplied with epics base.
2) pvAccess with provider ca
3) pvAccess with provider pva

If 1) is used the a put without callback is issued.
So the same client can send both 1 and 0.

The problem was detected using 2).
I suspect that the same problem will occur with 3).

What happens is that when the start button is pressed a put callback is used to
send 1 to the record. The put callback will not complete until the record becomes inactive.

If the stop button is pressed a new put callback is issued by the same client.
This request will not get to the record until the first callback completes.

I did get caqtdm to work with 2) by doing the following:
If the value field type is enumerated AND the provider is ca than
I use cadef.h code to issue a put without callback.
But this is NOT a good solution.
It does work with the wait record because to presents itself as a enumerated record.

I think this will be a problem for all of:
pvAccessCPP
pvAccessJava
pvaClientCPP
pvaClientJava
pvaSrv.

But what to do?

mrkraimer, 2017-04-28:
A possible solution is via a record option
record[putcallback=nowait]
The pvAccess server is responsible for implementing the semantics.
This is implemented as follows
1) pvAccessJava and pvAccessCPP
The caProviider will make put instead of putCallback requests.
2) pvDatabaseJava and pvDatabaseCPP
The local provider will implement the semantics
3) pvaSrv
It will implement the semantics.

No other changes are required.

ralphlange, 2017-04-28:
See [this thread on the mailing list from 2014](https://sourceforge.net/p/epics-pvdata/mailman/epics-pvdata-devel/thread/5B7A62384C1DF4489186ABF43720BAD27C4F3838%40EXCHMBX03.fed.cclrc.ac.uk/#msg32930494).

I have been working on this with Dave Hickin in December 2014 until he was happy with the behavior. Commit epics-base/pvaSrv@3f53cad0502b66b09f43a1c12abd3f5f735a0b41 contains the related code changes in pvaSrv.

mrkraimer, 2017-04-28:
I read the thread and see 
record[immediate=true]
So similar to 
record[putcallback=nowait]

So two issues:
1) what record option name and value should we use?
One question is should option only apply to channelPut?
What about channelGet and channelPutGet?
If it applies to more than channelPut a number of issues might arise.
2) Sounds like pvaSrv has an implementation. That leaves the following to implement the semantics:
   a) The caProvider of pvAccessCPP and pvAccessJava
   b) The local provider of pvDatabaseCPP and pvDatabaseJava

ralphlange, 2017-04-28:
There have been a lot more threads and discussions on this back in 2014. I am not going to index or repeat them all.
The current implementation uses `record[block=true]` or `record[block=false]` to indicate a blocking or non-blocking call.

I am not really in favour of changing a well-discussed, agreed-upon API without justification and review.

mrkraimer, 2017-04-28:
I read the complete
     See this thread on the mailing list from 2014.
It never mentioned [block=]
So please excuse my ignorance of this topic.

In any case I looked at the latest code for pvaSrv.
I see that [block=false] is honored for all of  channelProcess, channelGet, and channelPut.
Good!!! Already implemented.
I am not sure why a client would use channelGet with 
record[process=true,block=false]
but does no harm.

I think that  record[block=false]
should be honored for at least  channelProcess and channelPut.
But what about channelGet, channelPutGet, and channelRPC?

I do not see how  the caProvider on pvAccess  can implement it for channelGet.
The caProvide does not implement channelPutGet or channelRPC.

pvDatabase could implement it for channelGet, channelPutGet, and channelRPC.
But I do not see why a client would want to use it for channelGet or channelPutGet.
I can see that a client might want to use it for channelRPC.

What do You think?

anjohnson, 2017-04-28:
`[block=false]` wouldn't seem to be very useful for any operation that involves returning data (except possibly for a `channelGetPut` if there were such a thing). Since CA doesn't implement the more complex operations anyway I would say don't worry about adding support for a block parameter to the other operations at this point. There's an acronym from Extreme Programming/Agile development [YAGNI](https://martinfowler.com/bliki/Yagni.html) which I think applies here.

mrkraimer, 2017-04-28:
So caProvider from pvAccess and the local provider from pvDatabase will implement
record[block=false] 
only for channelProcess and for channelPut.
pvaSrv can stay as is.

Does this sound correct?

ralphlange, 2017-04-28:
Well, it's all a matter of definition.

IIRC, in the EPICS database, PP input links that point to records with asynchronous device support de-facto implement [process=true,block=false] behavior, as they trigger processing but fetch the current timestamp and value that are the result of the previous processing.
That is the default behavior, and only after we added SoftCallback device support it became possible to select [process=true,block=true] style.

It would seem strange to me to deliberately not allow the default behavior of V3 input links, as this would break compatibility: the current default behavior of a cross-IOC input link could not be achieved using pvAccess.

As for possible use cases, read-then-process would sound reasonable when dealing with a FIFO on the other side of a slow connection. Get-value-then-trigger-next-read is the most efficient way to handle that.

anjohnson, 2017-04-28:
@ralphlange I would argue that in reality the PP→asynchronous input link behavior (which applies to DB links only) is broken, and we just don't have any acceptable way to fix it — it certainly breaks the Principle of Least Astonishment. The problem with input links is that their API is synchronous, so they can't block without holding up the database.

We could add the equivalent of a PP flag to a new pvAccess input link type, but its getValue() method is still going to have to return immediately with some value, which should be the last monitor update it got for that channel (any get operation would have to block, which isn't allowed).

ralphlange, 2017-05-01:
Fine with me.
I wasn't arguing about the wrongness or rightness of the database behavior, just that we need a way to get the same behavior with pvAccess links that we have with CA links, else we would break compatibility.

In terms of design, having the synchronous case being the standard and the asynchronous being the special case is clearly the wrong way round, but I guess we all agree on that.

mrkraimer, 2017-05-02:
Sounds like we have agreement.
Here is info I will add to developerGuide.html under the section that describes record options

 block
       This is used for a channel request that results in the provider issuing a process request
       It specifies if a  request should block until processing completes .

 For example:
      pvput -r "record[block=false]field(value)" someScalarChannel 10.0
      pvput -r "record[block=false]field(value)" -p "ca" someV3Channel 10.0

This option is honored by the following:

pvDatabase
For channelProcess and channelPut
pvaSrv
For channelProcess, channelPut, and channelGet
pvAccess: provider ca
Only for channelPut. It is implemented by issuing a put instead of a put callback.

Do You want me to add the support to pvDatbase and pvAccess (both CPP and Java)?
Andrew and Ralph : please respond.

ralphlange, 2017-05-02:
Needs the specification which choice is the default.
I would argue that - as with CA - fire-and-forget (i.e. block=false) should be the default.

mrkraimer, 2017-05-02:
I will argue that block=true should be the default.
This is what pvAccess has supported starting with the javaIOC.

ralphlange, 2017-05-02:
Then lets compare the number of CA / EPICS V3 users that would experience an unexpected change in behavior against the number of pvAccess / JavaIOC users that would see such a change.
Principle of Least Astonishment - see above.

mrkraimer, 2017-05-03:
I think I agree with block=false being the default.

For  pvDatabase  provider local can  send back a status showing if the record is already
being processed. In any case the client does receive a callback.

For the pvAccess ca provider the client will receive a callback after the ca_put is issued but will always return status OK

mrkraimer, 2017-05-10:
After much thought and looking at pvaSrv and pvDatabase here are my conclusions,
1) the record option "record[block=true]" or "record[block=false]" should have the default value false.
It should be supported by
     a)pvaSrv,
     b) the  ca provider of pvAccessCPP, and the ca provider of pvAccessJava.
     c) It could be supported by the local provider of both pvDataCPP and pvDataJava.
2) pvaSrv already supports block and looks good!!
It supports block for channelProcess, channelGet, and channelPut.
3) The ca provider of both pvAccessJava and pvAccessCPP do not support this option.
They should and should implement the same semantics as pvaSrv.
This means to implement block for channelPut.
This is really important since it fixes the busy record problem.
The ca provider does not support channelProcess.
It needs some thought to see if it should support block for  channelGet.
4) pvDatabase (CPP and Java) currently only supports synchronous record processing.
Both PVRecord and the local provider would need changes to also support asynchronous processing.
I am working on a proposed solution that consists of adding some methods to PVRecord that can optionally be implemented by support code that wants to implement asynchronous processing. But this needs some more work before I am ready to present the proposal.

What is important is making the ca provider from both pvAccessCPP and pvAccessJava support block for channelPut.
I think this should be implemented before the first release of EPICS V7.

ralphlange, 2017-05-10:
Thanks a lot, Marty.

One comment ad 4.
Looking at daily use of control systems with their bus systems connecting to "intelligent" controllers in the field and what weird counter-intuitive workarounds the existing EPICS database uses to implement asynchronous on top of synchronous processing, it is IMHO important that the design is focused on getting asynchronous processing done right. It is much easier to implement synchronous processing as a special case of asynchronous operation (i.e. the callback is immediate) than the other way round.

mrkraimer, 2017-05-10:
Ralph,

Does you last comment refer to client or to server?
I will guess client.
Note that I created a new issue in pvDatabaseJava that has a proposal for adding asynchronous support to pvDatabase.

ralphlange, 2017-05-10:
I was commenting on your point labeled "4." i.e. pvDatabase, i.e. server.

mrkraimer, 2017-09-05:
This will be fixed  in release 6.0.
Until then it is fixed on branch master.

mdavidsaver, 2017-09-05:
@mrkraimer Please leave issues open until a release is made.  Add the "fix committed" tag.

anjohnson, 2017-09-05:
@mdavidsaver I'm not objecting to that policy, but I don't remember seeing it written down anywhere or being discussed by the group before, so I don't see how Marty would have known about it.

mdavidsaver, 2017-09-05:
It wasn't discussed, and I would have been shocked if Marty could anticipate it.  This is me being dictatorial.

mdavidsaver, 2017-12-19:
Released in module version 6.0.0, bundle version 7.0.1

closed 2017-12-19 by mdavidsaver