epics-base/pvAccessCPP #66: Problems with the latest EPICS V4 master branch modules.

mrkraimer, 2017-08-09:
I found problems with the latest EPICS V4 master branch modules.
I am using the following modules:
pvDataCPP
pvAccessCPP
normativeTypesCPP
pvaSrv
pvDatabaseCPP
exampleCPP

In addition there is an additional test at
https://github.com/mrkraimer/testClientCPP

I suspect that the problems are in pvAccessCPP and pvaSrv but this may be wrong.

I have found problem using the following base releases.
base-3.14.12.5
base-3.15.5
base-3.16.1

The problems are different for each release so each will be described separately.

*************base-3.14.12.5*********************************

*****Following shows problem*****

The only problem I found is that, when an example IOC is exited, a crash occurs if any pva clients are attached.
This means a client connected via pvaSrv to a DBRecord or connected via remote pva to a PVRecord.

In one window do the following:

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme --provider pva --request "field()" DBRdoubleArray

In another window do the following:

mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 
...
iocRun: All initialization complete
exampleDatabase
epics> exit
2017-08-09T09:10:16.533 Server context still has 1 transport(s) active and closing...
Segmentation fault (core dumped)
mrk> 

This also happens if monitorme attaches to a PVRecord.

In one window

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme --provider pva --request "field()" PVRdouble01Array

Then in another window:

mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 
...
iocRun: All initialization complete
exampleDatabase
epics> exit
Segmentation fault (core dumped)
mrk> 

If I start the database from release EPICS-CPP-4.6.0 the crash does not occur.

mrk> pwd
/home/install/EPICS-CPP-4.6.0/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 
...
iocRun: All initialization complete
exampleDatabase
startPVAServer
...
epics> exit
mrk> 

****Following works with base-3.14.12.5 but fails for 3.15 and 3.16***

In one window run

mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 

In another window run:

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme --provider pva --request "field()" DBRdoubleArray
Use provider: pva
Event DBRdoubleArray [] Changed:{0} overrun:{}
Event DBRdoubleArray [0,1,2,3,4] Changed:{1, 3, 4, 5, 7, 8} overrun:{}
Event DBRdoubleArray [5,6,7,8,9] Changed:{1, 7, 8} overrun:{}

In another window:

mrk> pwd
/home/epicsv4/masterCPP/testClientCPP
mrk> pwd
/home/epicsv4/masterCPP/testClientCPP
mrk> bin/linux-x86_64/putNoBlock pva DBRdoubleArray value false
providerName pva pvaSrv true caSrv false channelName DBRdoubleArray request value debug false
_____putNoBlock starting__
channelPutConnect status Status [type=OK]

Type exit to stop: 
putDone status Status [type=OK]

Type exit to stop: 
putDone status Status [type=OK]
exit
Type exit to stop: 
mrk> 

*************base-3.15.5*********************************

The same failures that were described for base-3.14.12.5 still exists, but a new problem exists.

The following shows a failure with arrays.

In one window:

mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 

In another window:

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme --provider pva --request "field()" DBRdoubleArray
Use provider: pva
Event DBRdoubleArray [] Changed:{0} overrun:{}
Event DBRdoubleArray [0,0,0,0,0] Changed:{1, 3, 4, 5, 7, 8} overrun:{}
Event DBRdoubleArray [0,0,0,0,0] Changed:{1, 7, 8} overrun:{}
Event DBRdoubleArray [0,0,0,0,0] Changed:{1, 7, 8} overrun:{}

NOTE THAT VALUES SHOULD NOT BE 0. SEE base-3.14.12.5 for correct values.

In another window:

mrk> pwd
/home/epicsv4/masterCPP/testClientCPP
mrk> bin/linux-x86_64/putNoBlock pva DBRdoubleArray value false
providerName pva pvaSrv true caSrv false channelName DBRdoubleArray request value debug false
_____putNoBlock starting__
channelPutConnect status Status [type=OK]

Type exit to stop: 
putDone status Status [type=OK]

Type exit to stop: 
putDone status Status [type=OK]

Type exit to stop: 
putDone status Status [type=OK]
exit

mrk> 

But it works correctly with provider ca (must merge pvAccessCPP pull request #65) 

 
mrk> pwd
/home/epicsv4/masterCPP/testClientCPP
mrk> bin/linux-x86_64/putNoBlock ca DBRdoubleArray value false

*************base-3.16.1*********************************

Now the failure shown for base-3.15.5 fails worse.

In one window start:

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme --provider pva --request "field()" DBRdoubleArray

In another window:

mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 
.....
epics>

Now in another window:

mrk> bin/linux-x86_64/putNoBlock pva DBRdoubleArray value false
providerName pva pvaSrv true caSrv false channelName DBRdoubleArray request value debug false
_____putNoBlock starting__
channelPutConnect status Status [type=OK]

Type exit to stop: 
error DBRdoubleArray DBRdoubleArray connection state DISCONNECTED

In the database window You will see:

epics> Segmentation fault (core dumped)
mrk>

mdavidsaver, 2017-08-09:
> Segmentation fault (core dumped)

When reporting a crash, please provide a stack trace.  Either run the test with a debugger, or analyze the core file which is created.

mdavidsaver, 2017-08-09:
> The only problem I found is that, when an example IOC is exited, a crash occurs if any pva clients are attached.
> This means a client connected via pvaSrv to a DBRecord or connected via remote pva to a PVRecord.

The issue with pvaSrv is not new, and not related to pvAccessCPP.  In caContext.cpp the function ca_attach_context() is called after the epicsExit() has run ca_client_exit_handler() which deletes the thread local variable used to hold CA context pointers.

mdavidsaver, 2017-08-09:
Specifically epics-base/pvaSrv#55

mdavidsaver, 2017-08-09:
@mrkraimer please see if the crash of exampleDatabase is from some other cause.

mrkraimer, 2017-08-09:
Here is the crash for base 3.16.
What causes it is:
mrk> pwd
/home/epicsv4/masterCPP/testClientCPP
mrk> bin/linux-x86_64/putNoBlock pva DBRdoubleArray
providerName pva pvaSrv true caSrv false channelName DBRdoubleArray request value debug false
_____putNoBlock starting__
channelPutConnect status Status [type=OK]

Type exit to stop: 

pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> valgrind ../../bin/linux-x86_64/exampleDatabase st.cmd 
==24324== Memcheck, a memory error detector
==24324== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==24324== Using Valgrind-3.12.0 and LibVEX; rerun with -h for copyright info
==24324== Command: ../../bin/linux-x86_64/exampleDatabase st.cmd
==24324== 
< envPaths
epicsEnvSet("IOC","exampleDatabase")
epicsEnvSet("TOP","/home/epicsv4/masterCPP/exampleCPP/database")
epicsEnvSet("EPICS4_DIR","/home/epicsv4/masterCPP")
epicsEnvSet("EPICS_BASE","/home/install/epics/base-3.16.1")
epicsEnvSet("PYTHON_BASE","/usr")
cd /home/epicsv4/masterCPP/exampleCPP/database
## Register all support components
dbLoadDatabase("dbd/exampleDatabase.dbd")
exampleDatabase_registerRecordDeviceDriver(pdbbase)
## Load record instances
dbLoadRecords("db/dbStringArray.db","name=DBRstringArray01")
dbLoadRecords("db/dbEnum.db","name=DBRenum01")
dbLoadRecords("db/dbCounter.db","name=DBRcounter01")
dbLoadRecords("db/dbArray.db","name=DBRdoubleArray,type=DOUBLE")
dbLoadRecords("db/test.db")
cd /home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
iocInit()
Starting iocInit
############################################################################
## EPICS R3.16.1
## EPICS Base built Aug  7 2017
############################################################################
iocRun: All initialization complete
2017-08-09T14:02:11.806 Using dynamically assigned TCP port 39403.
exampleDatabase
epics> ==24324== Thread 31 TCP-rx:
==24324== Invalid read of size 8
==24324==    at 0x61B8E28: dbLinkDoLocked (dbLink.c:422)
==24324==    by 0x5F70037: read_wf (devWfSoft.c:89)
==24324==    by 0x5F6E1CC: readValue (waveformRecord.c:321)
==24324==    by 0x5F6E1CC: process (waveformRecord.c:144)
==24324==    by 0x61A715C: dbProcess (dbAccess.c:595)
==24324==    by 0x61B9568: processNotifyCommon (dbNotify.c:264)
==24324==    by 0x52D9D68: epics::pvaSrv::DbPvPut::put(std::shared_ptr<epics::pvData::PVStructure> const&, std::shared_ptr<epics::pvData::BitSet> const&) (dbPvPut.cpp:137)
==24324==    by 0x73CBBD8: epics::pvAccess::ServerPutHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:1354)
==24324==    by 0x73C38AE: epics::pvAccess::ServerResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:157)
==24324==    by 0x73916C8: processApplicationMessage (codec.h:351)
==24324==    by 0x73916C8: epics::pvAccess::detail::AbstractCodec::processReadNormal() (codec.cpp:214)
==24324==    by 0x7391AE7: epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() (codec.cpp:1086)
==24324==    by 0x5D010A8: epicsThreadCallEntryPoint (epicsThread.cpp:83)
==24324==    by 0x5D06FFF: start_routine (osdThread.c:403)
==24324==  Address 0x40100000000000a0 is not stack'd, malloc'd or (recently) free'd
==24324== 
==24324== 
==24324== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==24324==  General Protection Fault
==24324==    at 0x61B8E28: dbLinkDoLocked (dbLink.c:422)
==24324==    by 0x5F70037: read_wf (devWfSoft.c:89)
==24324==    by 0x5F6E1CC: readValue (waveformRecord.c:321)
==24324==    by 0x5F6E1CC: process (waveformRecord.c:144)
==24324==    by 0x61A715C: dbProcess (dbAccess.c:595)
==24324==    by 0x61B9568: processNotifyCommon (dbNotify.c:264)
==24324==    by 0x52D9D68: epics::pvaSrv::DbPvPut::put(std::shared_ptr<epics::pvData::PVStructure> const&, std::shared_ptr<epics::pvData::BitSet> const&) (dbPvPut.cpp:137)
==24324==    by 0x73CBBD8: epics::pvAccess::ServerPutHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:1354)
==24324==    by 0x73C38AE: epics::pvAccess::ServerResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:157)
==24324==    by 0x73916C8: processApplicationMessage (codec.h:351)
==24324==    by 0x73916C8: epics::pvAccess::detail::AbstractCodec::processReadNormal() (codec.cpp:214)
==24324==    by 0x7391AE7: epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() (codec.cpp:1086)
==24324==    by 0x5D010A8: epicsThreadCallEntryPoint (epicsThread.cpp:83)
==24324==    by 0x5D06FFF: start_routine (osdThread.c:403)
==24324== 
==24324== HEAP SUMMARY:
==24324==     in use at exit: 5,082,541 bytes in 16,968 blocks
==24324==   total heap usage: 56,774 allocs, 2,105 frees, 8,713,868 bytes allocated
==24324== 
==24324== LEAK SUMMARY:
==24324==    definitely lost: 536 bytes in 17 blocks
==24324==    indirectly lost: 15 bytes in 1 blocks
==24324==      possibly lost: 74,830 bytes in 88 blocks
==24324==    still reachable: 4,834,464 bytes in 19,306 blocks
==24324==                       of which reachable via heuristic:
==24324==                         newarray           : 192 bytes in 2 blocks
==24324==         suppressed: 0 bytes in 0 blocks
==24324== Rerun with --leak-check=full to see details of leaked memory
==24324== 
==24324== For counts of detected and suppressed errors, rerun with: -v
==24324== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
mrk>

mdavidsaver, 2017-08-09:
Try issuing "stopPVAServer" before "exit".  Tomorrow I'll add an exit hook to do this automatically.

mrkraimer, 2017-08-09:
I am in the base 3.16 version.
Issuing "stopPVAServer" does fix the crash when database exits:

epics> stopPVAServer
2017-08-09T14:35:07.676 Server context still has 1 transport(s) active and closing...
caContext::stop referenceCount != 0 value 3
epics> exit
mrk> 

BUT note that this has nothing to do with the base 3.15 and 3.16 problems.
See my last message for 3.16 crash

mrkraimer, 2017-08-09:
Can you merge pull request #65 ?
If not it will be harder  for use to keep in sync.
As soon as it is merged I will build pvAccessCPP against epics-base instead of mrkraimer.
Also I will be gone most of the day tomorrow.
I will check my email before I leave.

mrkraimer, 2017-08-10:
I see that You made a change to pvaSrv.
I pulled and rebuild all modules I am using via base 3.16.
I still get the same crash

mrk> valgrind ../../bin/linux-x86_64/exampleDatabase st.cmd 
==1375== Memcheck, a memory error detector
==1375== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==1375== Using Valgrind-3.12.0 and LibVEX; rerun with -h for copyright info
==1375== Command: ../../bin/linux-x86_64/exampleDatabase st.cmd
==1375== 
< envPaths
epicsEnvSet("IOC","exampleDatabase")
epicsEnvSet("TOP","/home/epicsv4/masterCPP/exampleCPP/database")
epicsEnvSet("EPICS4_DIR","/home/epicsv4/masterCPP")
epicsEnvSet("EPICS_BASE","/home/install/epics/base-3.16.1")
epicsEnvSet("PYTHON_BASE","/usr")
cd /home/epicsv4/masterCPP/exampleCPP/database
## Register all support components
dbLoadDatabase("dbd/exampleDatabase.dbd")
exampleDatabase_registerRecordDeviceDriver(pdbbase)
## Load record instances
dbLoadRecords("db/dbStringArray.db","name=DBRstringArray01")
dbLoadRecords("db/dbEnum.db","name=DBRenum01")
dbLoadRecords("db/dbCounter.db","name=DBRcounter01")
dbLoadRecords("db/dbArray.db","name=DBRdoubleArray,type=DOUBLE")
dbLoadRecords("db/test.db")
cd /home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
iocInit()
Starting iocInit
############################################################################
## EPICS R3.16.1
## EPICS Base built Aug  7 2017
############################################################################
iocRun: All initialization complete
exampleDatabase
epics> ==1375== Thread 34 TCP-rx:
==1375== Invalid read of size 8
==1375==    at 0x61B8E28: dbLinkDoLocked (dbLink.c:422)
==1375==    by 0x5F70037: read_wf (devWfSoft.c:89)
==1375==    by 0x5F6E1CC: readValue (waveformRecord.c:321)
==1375==    by 0x5F6E1CC: process (waveformRecord.c:144)
==1375==    by 0x61A715C: dbProcess (dbAccess.c:595)
==1375==    by 0x61B9568: processNotifyCommon (dbNotify.c:264)
==1375==    by 0x52D9C78: epics::pvaSrv::DbPvPut::put(std::shared_ptr<epics::pvData::PVStructure> const&, std::shared_ptr<epics::pvData::BitSet> const&) (dbPvPut.cpp:137)
==1375==    by 0x73CBBD8: epics::pvAccess::ServerPutHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:1354)
==1375==    by 0x73C38AE: epics::pvAccess::ServerResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:157)
==1375==    by 0x73916C8: processApplicationMessage (codec.h:351)
==1375==    by 0x73916C8: epics::pvAccess::detail::AbstractCodec::processReadNormal() (codec.cpp:214)
==1375==    by 0x7391AE7: epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() (codec.cpp:1086)
==1375==    by 0x5D010A8: epicsThreadCallEntryPoint (epicsThread.cpp:83)
==1375==    by 0x5D06FFF: start_routine (osdThread.c:403)
==1375==  Address 0x40100000000000a0 is not stack'd, malloc'd or (recently) free'd
==1375== 
==1375== 
==1375== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==1375==  General Protection Fault
==1375==    at 0x61B8E28: dbLinkDoLocked (dbLink.c:422)
==1375==    by 0x5F70037: read_wf (devWfSoft.c:89)
==1375==    by 0x5F6E1CC: readValue (waveformRecord.c:321)
==1375==    by 0x5F6E1CC: process (waveformRecord.c:144)
==1375==    by 0x61A715C: dbProcess (dbAccess.c:595)
==1375==    by 0x61B9568: processNotifyCommon (dbNotify.c:264)
==1375==    by 0x52D9C78: epics::pvaSrv::DbPvPut::put(std::shared_ptr<epics::pvData::PVStructure> const&, std::shared_ptr<epics::pvData::BitSet> const&) (dbPvPut.cpp:137)
==1375==    by 0x73CBBD8: epics::pvAccess::ServerPutHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:1354)
==1375==    by 0x73C38AE: epics::pvAccess::ServerResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:157)
==1375==    by 0x73916C8: processApplicationMessage (codec.h:351)
==1375==    by 0x73916C8: epics::pvAccess::detail::AbstractCodec::processReadNormal() (codec.cpp:214)
==1375==    by 0x7391AE7: epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() (codec.cpp:1086)
==1375==    by 0x5D010A8: epicsThreadCallEntryPoint (epicsThread.cpp:83)
==1375==    by 0x5D06FFF: start_routine (osdThread.c:403)
==1375== 
==1375== HEAP SUMMARY:
==1375==     in use at exit: 7,891,054 bytes in 17,197 blocks
==1375==   total heap usage: 56,763 allocs, 1,885 frees, 8,707,382 bytes allocated
==1375== 
==1375== LEAK SUMMARY:
==1375==    definitely lost: 400 bytes in 16 blocks
==1375==    indirectly lost: 0 bytes in 0 blocks
==1375==      possibly lost: 75,654 bytes in 90 blocks
==1375==    still reachable: 7,644,928 bytes in 19,535 blocks
==1375==                       of which reachable via heuristic:
==1375==                         newarray           : 192 bytes in 2 blocks
==1375==         suppressed: 0 bytes in 0 blocks
==1375== Rerun with --leak-check=full to see details of leaked memory
==1375== 
==1375== For counts of detected and suppressed errors, rerun with: -v
==1375== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
mrk>

mdavidsaver, 2017-08-10:
> I see that You made a change to pvaSrv.
> I pulled and rebuild all modules I am using via base 3.16.
> I still get the same crash

This change is unrelated cleanup work https://github.com/epics-base/pvaSrv/commit/d2ced0f8b0b77294581c67b6c89bb8c1d648de5b

mrkraimer, 2017-08-11:
I found what causes the crash in base 3.16.1 .
It happens when a ChannelPut is created and the first operation is a put.
If the first operation is a get all works fine.
After the first get multiple puts can be issued.

This is probably also the cause of the error reported for 3.15.

I have not looked carefully but suspect that the problem is in the 3.15 code for pvaSrv.
DbUtil::get does a lot.
But it is never called for a channePut unless ChannelPut::get is called before the first ChannelPut::put.

mrkraimer, 2017-08-15:
The following change to pvaSrv fixes both the base 3.15 and 3.16 problems described above

mrk> pwd
/home/epicsv4/masterCPP/pvaSrv
mrk> git diff src/dbPv/3.15/dbPvPut.cpp
diff --git a/src/dbPv/3.15/dbPvPut.cpp b/src/dbPv/3.15/dbPvPut.cpp
index 02842b4..728996a 100644
--- a/src/dbPv/3.15/dbPvPut.cpp
+++ b/src/dbPv/3.15/dbPvPut.cpp
@@ -97,8 +97,17 @@ bool DbPvPut::init(PVStructure::shared_pointer const &pvRequest)
     }
     int numFields = pvStructure->getNumberFields();
     bitSet.reset(new BitSet(numFields));
+    dbScanLock(dbChannelRecord(dbPv->getDbChannel()));
+    Status status = dbUtil->get(
+        req,
+        propertyMask,
+        dbPv->getDbChannel(),
+        pvStructure,
+         bitSet,
+         0);
+    dbScanUnlock(dbChannelRecord(dbPv->getDbChannel()));
     if(req) req->channelPutConnect(
-       Status::Ok,
+       status,
        getPtrSelf(),
        pvStructure->getStructure());
     return true;
mrk>

mdavidsaver, 2017-08-23:
@mrkraimer Is there a pull request come for this?

mdavidsaver, 2017-08-24:
I'm closing this issue as I think it has been established that the reported problems are bugs in pvaSrv.  Re-open if this isn't the case.

closed 2017-08-24 by mdavidsaver

mrkraimer, 2017-08-24:
I am not preparing a pull request.
I gave a fix above but am not sure that this is correct fix.
I will leave it to You to decide on the correct fix.