epics-base/pvAccessCPP #68: does not properly honor -r for  channelPut

mrkraimer, 2017-10-04:
The database has a record
PVRhelloPutGet
structure 
    structure argument
        string value 
    structure result
        string value 
        time_t timeStamp 2106-02-07T01:28:16.000 0

The following happens:

mrk> pvput -r "argument.value" PVRhelloPutGet Marty
Old : 
structure 
    structure argument
        string value 

Segmentation fault (core dumped)

This did not happen with last official V4 release

rk> /home/epicsv4/release4_6/pvAccessCPP/bin/linux-x86_64/pvput -r "argument.value" PVRhelloPutGet Marty
Old : 
structure 
    structure argument
        string value 

New : 
structure 
    structure argument
        string value Marty

mdavidsaver, 2017-10-04:
Please pull down 41dd55e00f0a5e49fd19049aaa8fd1fcab621c92 and retest.  If you can still cause a crash,  I'll need you to capture a stack trace.

mdavidsaver, 2017-10-04:
Also, to repeat what was said at the meeting.  I've changed the way arguments for pvput are given when writing to non-value fields such that

> pvput -r "argument.value" PVRhelloPutGet Marty

needs to become

> pvput -r "argument.value" PVRhelloPutGet argument.value=Marty

mrkraimer, 2017-10-04:
I tried 
pvput -r "argument.value" PVRhelloPutGet argument.value=Marty

and it works just fine

I pulled latest from pvAccessCPP and now I get

mrk> pvput -r "argument.vaue" PVRhelloPutGet Marty
terminate called after throwing an instance of 'std::runtime_error'
  what():  Can't construct Structure, NULL in fields
Old : Aborted (core dumped)
mrk> 
Is this what You expect?

If so we can close issue.

mdavidsaver, 2017-10-04:
> Is this what You expect?

> Old : Aborted (core dumped)

No!  Core dump== segfault

Can you run in gdb and capture a stack trace?

mrkraimer, 2017-10-04:
This is from valgrind

==15794== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==15794==  Access not within mapped region at address 0x40
==15794==    at 0x40F7CF: getSubField<char const*> (pvData.h:796)
==15794==    by 0x40F7CF: (anonymous namespace)::Putter::putBuild(std::shared_ptr<epics::pvData::Structure const> const&, pvac::ClientChannel::PutCallback::Args&) (pvput.cpp:287)
==15794==    by 0x510E374: (anonymous namespace)::GetPutter::channelPutConnect(epics::pvData::Status const&, std::shared_ptr<epics::pvAccess::ChannelPut> const&, std::shared_ptr<epics::pvData::Structure const> const&) (clientGet.cpp:108)
==15794==    by 0x515C5FF: (anonymous namespace)::ChannelPutImpl::initResponse(std::shared_ptr<epics::pvAccess::Transport> const&, signed char, epics::pvData::ByteBuffer*, signed char, epics::pvData::Status const&) (clientContextImpl.cpp:863)
==15794==    by 0x513B04C: (anonymous namespace)::BaseRequestImpl::response(std::shared_ptr<epics::pvAccess::Transport> const&, signed char, epics::pvData::ByteBuffer*) (clientContextImpl.cpp:270)
==15794==    by 0x5141BDF: (anonymous namespace)::DataResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (clientContextImpl.cpp:2539)
==15794==    by 0x513A30D: (anonymous namespace)::ClientResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (clientContextImpl.cpp:3044)
==15794==    by 0x512E5F8: processApplicationMessage (codec.h:353)
==15794==    by 0x512E5F8: epics::pvAccess::detail::AbstractCodec::processReadNormal() (codec.cpp:215)
==15794==    by 0x512EA17: epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() (codec.cpp:1089)
==15794==    by 0x59E70E8: epicsThreadCallEntryPoint (epicsThread.cpp:83)
==15794==    by 0x59ED07F: start_routine (osdThread.c:403)
==15794==    by 0x6893739: start_thread (in /usr/lib64/libpthread-2.24.so)
==15794==    by 0x65CDE7E: clone (in /usr/lib64/libc-2.24.so)

mdavidsaver, 2017-10-05:
cb9a154934e1711392664713b1e394fa849b437b Fixes a missing NULL check.  Please confirm that you now get an error "Structure has not .value".

ralphlange, 2017-10-05:
Should be "does not have" or "has no"?

mrkraimer, 2017-10-05:
Now it shows

mrk> pvput -r "argument.value" PVRhelloPutGet Marty
Old : 
structure 
    structure argument
        string value Marty

Error: Structure has no .value
New : 
structure 
    structure argument
        string value Marty
Look good!!

I will close the issue

mrkraimer, 2017-10-05:
Following shows that it does not change server

mrk> pvput -r "argument.value" PVRhelloPutGet NewName
Old : 
structure 
    structure argument
        string value Marty

Error: Structure has no .value
New : 
structure 
    structure argument
        string value Marty

closed 2017-10-05 by mrkraimer