epics-base/pvAccessCPP #71: *pvget* is unable to get the time stamp when using the ```-p ca``` argument (channel access)

SkyToGround, 2017-11-15:
When using caget (channel access) to access a pv, a value and timestamp can be obtained:
```bash
jonasnilsson@Muspelheim ~> caget -a LabS-Embla:Chop-CHIC-01:Ref
LabS-Embla:Chop-CHIC-01:Ref    2017-11-15 14:30:06.235021 8.79601e+08
```
However, when using pvget, it appears the I can only get the value. There appears to be no timestamp. Note also how the structure is printed twice.
```
jonasnilsson@Muspelheim ~> pvget -p ca -r "" LabS-Embla:Chop-CHIC-01:Ref
CAChannelGet::lastRequest() LabS-Embla:Chop-CHIC-01:Ref does not do anything
CAChannelGet::lastRequest() LabS-Embla:Chop-CHIC-01:Ref does not do anything
LabS-Embla:Chop-CHIC-01:Ref
epics:nt/NTScalar:1.0
    double value 8.79601e+08
    alarm_t alarm NO_ALARM NO_STATUS NO_ALARM
    display_t display
        double limitLow 0
        double limitHigh 0
        string description
        string format %f
        string units
    control_t control
        double limitLow 0
        double limitHigh 0
        double minStep 0
    valueAlarm_t valueAlarm
        boolean active false
        double lowAlarmLimit nan
        double lowWarningLimit nan
        double highWarningLimit nan
        double highAlarmLimit nan
        int lowAlarmSeverity 0
        int lowWarningSeverity 0
        int highWarningSeverity 0
        int highAlarmSeverity 0
        double hysteresis 0

LabS-Embla:Chop-CHIC-01:Ref
epics:nt/NTScalar:1.0
    double value 8.79601e+08
    alarm_t alarm NO_ALARM NO_STATUS NO_ALARM
    display_t display
        double limitLow 0
        double limitHigh 0
        string description
        string format %f
        string units
    control_t control
        double limitLow 0
        double limitHigh 0
        double minStep 0
    valueAlarm_t valueAlarm
        boolean active false
        double lowAlarmLimit nan
        double lowWarningLimit nan
        double highWarningLimit nan
        double highAlarmLimit nan
        int lowAlarmSeverity 0
        int lowWarningSeverity 0
        int highWarningSeverity 0
        int highAlarmSeverity 0
        double hysteresis 0
```

Furthermore, when requesting the actual timeStamp (using ```-r```), we do get the time stamp:
```bash
jonasnilsson@Muspelheim ~> pvget -p ca -r "timeStamp" LabS-Embla:Chop-CHIC-01:Ref
CAChannelGet::lastRequest() LabS-Embla:Chop-CHIC-01:Ref does not do anything
CAChannelGet::lastRequest() LabS-Embla:Chop-CHIC-01:Ref does not do anything
LabS-Embla:Chop-CHIC-01:Ref
structure
    time_t timeStamp 2017-11-15T14:43:11.819 0

LabS-Embla:Chop-CHIC-01:Ref
structure
    time_t timeStamp 2017-11-15T14:43:11.819 0
```

And by the way, ```pvinfo```does not work:
```bash
jonasnilsson@Muspelheim ~> pvinfo -p ca LabS-Embla:Chop-CHIC-01:Ref
[LabS-Embla:Chop-CHIC-01:Ref] failed to get channel introspection data: [ERROR] field '' not found
CHANNEL  : LabS-Embla:Chop-CHIC-01:Ref
STATE    : NEVER_CONNECTED
(null introspection data)
```

EPICSv4 was built using the versions of the submodules checked out from the [bundleCPP](https://github.com/epics-base/bundleCPP) repository and EPICS base version 3.16.1.

mdavidsaver, 2017-11-15:
Which revision of bundleCPP are you building?  Please give the output of "git log -n1" and "git submodule status".

SkyToGround, 2017-11-16:
I am using the latest available commit of bundleCPP:
```bash
jonasnilsson@Muspelheim /o/e/m/epicsv4> git log -n1
commit 721586497e414c9787db3a9b58816b9ec1fac0cb (HEAD -> master, origin/master, origin/HEAD)
Author: Andrew Johnson <anj@aps.anl.gov>
Date:   Fri Nov 10 13:31:57 2017 -0600

    Module updates
jonasnilsson@Muspelheim /o/e/m/epicsv4> git submodule status
 8e4a96c57b50a6ea480e2fbdfeec11d59f771c65 exampleCPP (4.2.1-86-g8e4a96c)
 fc4b51cb4922310e9bfa70603460a4113ed747fc normativeTypesCPP (5.0.0-38-gfc4b51c)
 3b565d0c561a5cf3f9e861cdaec55460f061fb8c pvAccessCPP (4.1.2-491-g3b565d0)
 7cd95677ffc510340170f220367ff681b3ee952d pvCommonCPP (4.1.0-34-g7cd9567)
 7136098c3c6433b203fd84f77dc4c83c8389806d pvDataCPP (5.0.0-228-g7136098)
 c6339f4f1bffbd906419ac1bb3d22fc382ccd942 pvDatabaseCPP (4.1.1-85-gc6339f4)
 e472e552a080358139c9d6bce47349ac24bd8bb3 pva2pva (e472e55)
 411560ea8dc57ee131315d4e9d6e746146822061 pvaClientCPP (4.1.1-110-g411560e)
 b806e7e1c12116ca130d8b3e3c68bb588921970e pvaPy (0.9.0)
```

mrkraimer, 2017-11-16:
In the latest from master branch I no longer see 
CAChannelGet::lastRequest() LabS-Embla:Chop-CHIC-01:Ref does not do anything

I thought I had fixed the value appearing twice but I see that it still happens.

I will also look into why
pvlist -p ca ...
does not work.

anjohnson, 2017-11-16:
Hi Jonas, if you use the bundleCPP module you should always run
```
git submodule update --remote
```
to pull in the latest versions of all the submodules. If you don't do that you will get the version they were at when I last committed an update for them to the bundleCPP module itself. I usually only commit updates to the parent module just before a new release, and I last updated it a few days ago because we're about to come out with a new release. I just pushed an update which brings everything into line with the versions that will be in EPICS Base-7.0.1-pre1

Marty, the pvAccessCPP version he is using was several commits behind the tip of the master branch.

SkyToGround, 2017-11-16:
I executed the ```git submodule update --remote``` line and called ```make```again. That removed the ```CAChannelGet::lastRequest() LabS-Embla:Chop-CHIC-01:Ref does not do anything``` line though nothing else has appeared to change. Note however that I did not re-build from a clean state, I will do that tomorrow as it is getting late over here.

mrkraimer, 2017-11-17:
I am working on the other issues but not done.
It is not so easy since pvAccessCPP now allows calls to getField, createChannelGet, etc.
before the channel has connected.
This is what is causing the failures for pvinfo and pvget,

One note about
 pvget -p ca -r ""

The way  channel access works ca_get has to specify a DBD type.
No DBD types  allows all of value,alarm,timeStamp, control, and display.
For -r "" the code is using DBR_CTRL_STRING.
I will change this to DBR_TIME_STRING.
Then you will get value,alarm, and timeStamp.

ralphlange, 2017-11-17:
Shouldn't it rather be doing two calls?

mrkraimer, 2017-11-17:
pvinfo -p ca channelName will return complete introspection interface

I have just created a new pull request that fixes all these issues

gregoryraymondwhite, 2017-11-17:
Cool Marty!! Presumably not for RPC channels though?

ralphlange, 2017-11-18:
Greg, this is about the ChannelAccess provider: PVA client library, CA protocol on the wire, connecting to a CA server.
What do you mean by "RPC channels" in this context?

SkyToGround, 2017-11-20:
From reading all the comments, it appears that this issue is related to a similar issue we have had with reading waveforms. I will get back to you on that issue when @mrkraimer's changes has been merged and I have had time to test them.

mdavidsaver, 2017-12-19:
@anjohnson Can this issue be closed?

anjohnson, 2017-12-20:
Thanks Jonas.

closed 2017-12-20 by anjohnson