epics-base/pvAccessCPP #72: PVA client sending truncated CMD_MONITOR init message

mdavidsaver, 2017-11-16:
@mrkraimer says:

I did manage to create a simpler version of examplePvaClientPut.cpp that generates the Not-a-first segmented message
Checkout in the same directory that has the other PVA modules.
https://github.com/mrkraimer/testClientCPP

You will also need pvaClientCPP and pvDatabaseCPP

I have tried it with the database started from

> mrk> pwd
> /home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
> mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd
and

> mrk> pwd
> /home/epicsv4/masterCPP/exampleCPP/database
> mrk> bin/linux-x86_64/exampleDatabaseMain
and

> mrk> pwd
> /home/epicsv4/release4_6/exampleCPP/database
> mrk> bin/linux-x86_64/exampleDatabaseMain

All show the same error messages
THUS it is not a new problem. It existed before.

When I run

> mrk> pwd
> /home/epicsv4/masterCPP/testClientCPP
> mrk> bin/linux-x86_64/testPvaClientPut
On the server I see messages like:

> 2017-11-15T15:31:44.152 Not-a-first segmented message expected from the client at ../../src/remote/codec.cpp:326: 10.0.0.127:54690, disconnecting...
> 2017-11-15T15:31:44.370 Not-a-first segmented message expected from the client at ../../src/remote/codec.cpp:326: 10.0.0.127:54692, disconnecting...

Note also that in the code

>     PvaClientChannelPtr channel = pva->channel(channelName,providerName,2.0);
>     PvaClientPutPtr put = channel->put();
>     PvaClientPutDataPtr putData = put->getData();
>     PvaClientMonitorPtr monitor = pva->channel(channelName,providerName,2.0)->monitor("value");
>     PvaClientMonitorDataPtr monitorData = monitor->getData();
>     putData->putDouble(3.0); put->put();
> channel->get("field(value)")->getData()->showChanged(cout) << endl;

channels and channelGets and channelPuts are cached by pvaClientCPP.
But monitors are not cached.
Thus monitors are created and destroyed each time testDouble is called.

Marty

mdavidsaver, 2017-11-16:
I am able to reproduce this.  It looks like the client side is sending a truncated message when starting a new subscription.

mdavidsaver, 2017-11-16:
The offending message is:

> 0000   ca 01 00 0d 09 00 00 00 01 00 00 00 10 00 00 00
> 0010   fe

Where 0xfe is being interpreted as QOS_INIT, which should be followed by a serialized pvRequest.  It isn't, which leads to an attempt to process the following message as a continuation.  The following message isn't an error, which triggers "Not-a-first segmented message" and drops the connection.

mdavidsaver, 2017-11-16:
The issue manifests in ChannelMonitorImpl::send() where I find pendingRequest==4 (QOS_PROCESS) and m_pendingRequest==-2 (PURE_DESTROY_REQUEST).

Looks like a destroy message is received between the call to getPendingRequest() (which locks and reads m_pendingRequest) and the later reads of m_pendingRequest w/o a lock.

mrkraimer, 2017-11-16:
I checked out the latest pvAccessCPP.
I no longer see the Not-a-first segmented messages

Looks good!!