epics-base/pvAccessCPP #76: eget issues

anjohnson, 2017-11-28:
It looks like @mdavidsaver fixed the pvget problems that @sveseli found yesterday with a commit last night, but there are still similar issues with eget.

Here are some ‘interesting’ `eget` behaviors and messages with non-existent channels:
```
tux% /bin/time -p bin/linux-x86_64/eget anj:exit
[anj:exit] failed to get channel introspection data: [ERROR] channel destroyed
[anj:exit] failed to get channel introspection data
real 3.01
user 0.00
sys 0.00
tux% /bin/time -p bin/linux-x86_64/eget anj:exit anj:exit.DESC
[anj:exit] get timeout
[anj:exit.DESC] get timeout

real 6.01
user 0.00
sys 0.00
```
So adding a second non-existent PV doubles the time taken and changes the error message.
```
tux% /bin/time -p bin/linux-x86_64/eget -p ca anj:exit
[anj:exit] failed to get channel introspection data
real 3.01
user 0.00
sys 0.01
tux% /bin/time -p bin/linux-x86_64/eget -p ca anj:exit anj:exit.DESC
[anj:exit] get timeout
[anj:exit.DESC] get timeout

real 6.01
user 0.01
sys 0.01
```
Using CA affects the error message slightly, but for a single channel it isn't very helpful.

With a mixture of existing and non-existing channels, the eget timeout message is slightly more useful than pvget:
```
tux% bin/linux-x86_64/eget -T no-such-channel anj:exit.DESC anj:exit another-bad-channel
[no-such-channel] get timeout
[another-bad-channel] get timeout
Exit subroutine
0
tux% bin/linux-x86_64/pvget no-such-channel anj:exit.DESC anj:exit another-bad-channel
anj:exit.DESC                  Exit subroutine
anj:exit                       0
Timeout
```
With only channels that do exist, `eget -m` seems to be pretty broken:
```
tux% bin/linux-x86_64/eget anj:exit.DESC
Exit subroutine
tux% bin/linux-x86_64/eget -m anj:exit.DESC
```
It just sits there doing nothing. Changing the value on the server has no effect either, and shutting down the server triggers a seg-fault:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff4c1a700 (LWP 5495)]
0x00007ffff790b9bf in (anonymous namespace)::InternalClientContextImpl::InternalChannelImpl::reportChannelStateChange (this=0x6b3aa0)
    at ../../src/remoteClient/clientContextImpl.cpp:3827
3827	                        EXCEPTION_GUARD(R->getRequester()->channelDisconnect(connectionState==Channel::DESTROYED);)
Missing separate debuginfos, use: debuginfo-install glibc-2.17-196.el7.x86_64 libgcc-4.8.5-16.el7.x86_64 libstdc++-4.8.5-16.el7.x86_64 ncurses-libs-5.9-13.20130511.el7.x86_64 readline-6.2-10.el7.x86_64
(gdb) bt
#0  0x00007ffff790b9bf in (anonymous namespace)::InternalClientContextImpl::InternalChannelImpl::reportChannelStateChange (this=0x6b3aa0)
    at ../../src/remoteClient/clientContextImpl.cpp:3827
#1  0x00007ffff78e48b5 in epics::pvAccess::detail::BlockingClientTCPTransportCodec::closedNotifyClients (this=0x7fffec000a70)
    at ../../src/remote/codec.cpp:1805
#2  0x00007ffff78e26b8 in epics::pvAccess::detail::BlockingTCPTransportCodec::close (this=0x7fffec000a70) at ../../src/remote/codec.cpp:1037
#3  0x00007ffff78e0628 in epics::pvAccess::detail::AbstractCodec::readToBuffer
    (this=this@entry=0x7fffec000a70, requiredBytes=requiredBytes@entry=8, 
    persistent=persistent@entry=false) at ../../src/remote/codec.cpp:381
[... 83 lines cut ...]

Thread 3 (Thread 0x7ffff7fa5700 (LWP 5437)):
#0  0x00007ffff624ecf2 in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib64/libpthread.so.0
#1  0x00007ffff707cc89 in epicsEventWaitWithTimeout (pevent=0x62f9d0, 
    timeout=0.234900372) at ../../src/osi/os/posix/osdEvent.c:130
#2  0x00007ffff7566e23 in epics::pvData::Event::wait (
    this=this@entry=0x62f940, timeOut=<optimized out>)
    at ../../src/misc/event.cpp:60
#3  0x00007ffff756917c in epics::pvData::Timer::run (this=0x62f920)
    at ../../src/misc/timer.cpp:138
#4  0x00007ffff7073fd9 in epicsThreadCallEntryPoint (pPvt=0x62f950)
    at ../../src/osi/epicsThread.cpp:83
#5  0x00007ffff707a3cc in start_routine (arg=0x62fb90)
    at ../../src/osi/os/posix/osdThread.c:403
#6  0x00007ffff624ae25 in start_thread () from /lib64/libpthread.so.0
#7  0x00007ffff655734d in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7ffff7fa7740 (LWP 5429)):
#0  0x00007ffff651e1ad in nanosleep () from /lib64/libc.so.6
#1  0x00007ffff707b3f8 in epicsThreadSleep (seconds=<optimized out>)
    at ../../src/osi/os/posix/osdThread.c:715
#2  0x0000000000409837 in main (argc=<optimized out>, argv=<optimized out>)
    at ../eget.cpp:1877
```

mdavidsaver, 2017-11-28:
Sigh... It's true that I didn't give 'eget' the same attention as pvget/put/info.  I hadn't even notice that it had a Monitor mode.  All of these tools, and eget most especially, are in need if some cleanup and reworking.  eg. I'd like to see the "-m" modes of pvget and eget split out as separate executables.

Unless someone steps up to do the fixing, I'm going to remove "-m" from 'eget' for the 7 release.  I don't have time to dive into this now.

anjohnson, 2017-11-28:
Disabling eget monitors for now is fine by me — I wasn't trying to pile more stuff on you, just to make sure we had a record of the issues, honest!

I agree that all the pvTools need cleaning up and unifying, which might be a nice project for someone coming in who wants to learn the APIs; the capabilities of the catools should provide pointers on what could be done. Just be aware that some sites are probably using these programs in scripts already (IIRC the issue of breaking users' scripts was why we already have both pvget and eget).

mdavidsaver, 2017-11-28:
"eget -m" disabled with 4e743f2d95851e8e372337aa0969763fe0a556b7.  This issue will remain open until a permanent solution is found.

mdavidsaver, 2019-03-18:
Further work will/would be done in the get repo.  cf. epics-base/eget#1

closed 2019-03-18 by mdavidsaver