epics-base/pvAccessCPP #77: caProvider deadlock on monitor shutdown

mdavidsaver, 2017-12-19:
Testing with softIocPVA and the following .db and the 'monitorme' example hits a deadlock:

```sh
cat <<EOF > spam2.db 
record(calc, "X1") {
    field(INPA, "X1.VAL CP")
    field(CALC, "A+1")
}
EOF
softIocPVA -d spam2.db
```

then concurrently:

```sh
./examples/O.linux-x86_64/monitorme -p ca X1
```

Hit Ctrl+c or otherwise SIGINT to trigger.

mdavidsaver, 2017-12-19:
Looks like a lock order problem.

```
(gdb) thr app all bt

Thread 6 (Thread 0x7f3392248700 (LWP 10176)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007f3394ea248b in epicsEventWait (pevent=0x7f338400e4a0) at ../../../src/libCom/osi/os/posix/osdEvent.c:103
#2  0x00007f3394e9afcc in epicsEvent::wait (this=<optimized out>) at ../../../src/libCom/osi/epicsEvent.cpp:66
#3  0x00007f3395768b4e in tcpSendThread::run (this=0x7f3384000a00) at ../tcpiiu.cpp:87
#4  0x00007f3394e99b59 in epicsThreadCallEntryPoint (pPvt=0x7f3384000a08) at ../../../src/libCom/osi/epicsThread.cpp:83
#5  0x00007f3394e9fd30 in start_routine (arg=0x7f338400a210) at ../../../src/libCom/osi/os/posix/osdThread.c:403
#6  0x00007f3394015494 in start_thread (arg=0x7f3392248700) at pthread_create.c:333
#7  0x00007f3394313aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

Thread 5 (Thread 0x7f3392449700 (LWP 10175)):
#0  __lll_lock_wait () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007f3394017c06 in __GI___pthread_mutex_lock (mutex=0x561dd599bee0) at ../nptl/pthread_mutex_lock.c:115
#2  0x00007f3394ea1fca in mutexLock (id=0x561dd599bee0) at ../../../src/libCom/osi/os/posix/osdMutex.c:46
#3  epicsMutexOsdLock (pmutex=0x561dd599bee0) at ../../../src/libCom/osi/os/posix/osdMutex.c:130
#4  0x00007f3394e9a9a9 in epicsMutexLock (pmutexNode=<optimized out>) at ../../../src/libCom/osi/epicsMutex.cpp:150
#5  0x00007f3394e9ac3c in epicsMutex::lock (this=this@entry=0x561dd599be58) at ../../../src/libCom/osi/epicsMutex.cpp:273
#6  0x00007f339547280e in epicsGuardRelease<epicsMutex>::~epicsGuardRelease (this=<synthetic pointer>, 
    __in_chrg=<optimized out>) at /home/mdavidsaver/work/epics/pv/epics-base/include/epicsGuard.h:105
#7  pvac::Monitor::Impl::callEvent (evt=pvac::MonitorEvent::Data, G=<synthetic pointer>..., this=0x561dd599be40)
    at ../../src/client/clientMonitor.cpp:60
#8  pvac::Monitor::Impl::monitorEvent (this=this@entry=0x561dd599be40, 
    monitor=std::shared_ptr (count 2, weak 2) 0x561dd599bfa0) at ../../src/client/clientMonitor.cpp:156
#9  0x00007f33959b3e08 in epics::pvAccess::ca::CAChannelMonitor::subscriptionEvent (this=0x561dd599bfa0, args=...)
    at ../caChannel.cpp:1914
#10 0x00007f33959b3f53 in epics::pvAccess::ca::ca_subscription_handler (args=...) at ../caChannel.cpp:1696
#11 0x00007f3395772e30 in oldSubscription::current (this=<optimized out>, guard=..., type=20, count=1, pData=0x7f338400a468)
    at ../oldSubscription.cpp:68
#12 0x00007f3395748d85 in cac::eventRespAction (this=0x561dd5977b50, iiu=..., hdr=..., pMsgBdy=0x7f338400a468)
---Type <return> to continue, or q <return> to quit---
    at ../cac.cpp:968
#13 0x00007f339576a424 in tcpiiu::processIncoming (this=0x7f33840008c0, currentTime=..., mgr=...) at ../tcpiiu.cpp:1264
#14 0x00007f339576a9fc in tcpRecvThread::run (this=0x7f33840009a8) at ../tcpiiu.cpp:519
#15 0x00007f3394e99b59 in epicsThreadCallEntryPoint (pPvt=0x7f33840009b0) at ../../../src/libCom/osi/epicsThread.cpp:83
#16 0x00007f3394e9fd30 in start_routine (arg=0x7f3384009e70) at ../../../src/libCom/osi/os/posix/osdThread.c:403
#17 0x00007f3394015494 in start_thread (arg=0x7f3392449700) at pthread_create.c:333
#18 0x00007f3394313aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

Thread 4 (Thread 0x7f339254a700 (LWP 10174)):
#0  0x00007f3394314a13 in recvfrom () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f339576297b in udpRecvThread::run (this=0x561dd59986e8) at ../udpiiu.cpp:409
#2  0x00007f3394e99b59 in epicsThreadCallEntryPoint (pPvt=0x561dd5998708) at ../../../src/libCom/osi/epicsThread.cpp:83
#3  0x00007f3394e9fd30 in start_routine (arg=0x561dd5998950) at ../../../src/libCom/osi/os/posix/osdThread.c:403
#4  0x00007f3394015494 in start_thread (arg=0x7f339254a700) at pthread_create.c:333
#5  0x00007f3394313aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

Thread 3 (Thread 0x7f3395d32700 (LWP 10173)):
#0  pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
#1  0x00007f3394ea2609 in epicsEventWaitWithTimeout (pevent=0x561dd5978e70, timeout=0.016195654)
    at ../../../src/libCom/osi/os/posix/osdEvent.c:130
#2  0x00007f3394e9b01c in epicsEvent::wait (this=this@entry=0x561dd5978c40, timeOut=<optimized out>)
    at ../../../src/libCom/osi/epicsEvent.cpp:75
#3  0x00007f3394eac0bf in timerQueueActive::run (this=0x561dd5978b80) at ../../../src/libCom/timer/timerQueueActive.cpp:95
#4  0x00007f3394e99b59 in epicsThreadCallEntryPoint (pPvt=0x561dd5978c50) at ../../../src/libCom/osi/epicsThread.cpp:83
#5  0x00007f3394e9fd30 in start_routine (arg=0x561dd59790a0) at ../../../src/libCom/osi/os/posix/osdThread.c:403
#6  0x00007f3394015494 in start_thread (arg=0x7f3395d32700) at pthread_create.c:333
#7  0x00007f3394313aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97

Thread 2 (Thread 0x7f339274b700 (LWP 10172)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
---Type <return> to continue, or q <return> to quit---
#1  0x00007f3394ea248b in epicsEventWait (pevent=0x561dd5978640) at ../../../src/libCom/osi/os/posix/osdEvent.c:103
#2  0x00007f3394e9afcc in epicsEvent::wait (this=this@entry=0x561dd59784f0) at ../../../src/libCom/osi/epicsEvent.cpp:66
#3  0x00007f3394e979ca in (anonymous namespace)::ipAddrToAsciiGlobal::run (this=0x561dd59780b0)
[... cut ...]
#22 std::_Destroy<std::shared_ptr<(anonymous namespace)::MonTracker>*, std::shared_ptr<(anonymous namespace)::MonTracker> >
    (__last=0x561dd599c070, __first=<optimized out>) at /usr/include/c++/6/bits/stl_construct.h:151
#23 std::vector<std::shared_ptr<(anonymous namespace)::MonTracker>, std::allocator<std::shared_ptr<(anonymous namespace)::MonTracker> > >::~vector (this=0x7ffc48a97090, __in_chrg=<optimized out>) at /usr/include/c++/6/bits/stl_vector.h:426
#24 0x0000561dd431392e in main (argc=<optimized out>, argv=<optimized out>) at ../monitorme.cpp:244
(gdb)
```

mrkraimer, 2017-12-19:
I can not reproduce this problem.
I started the exampl database from exampleCPP/database, i.e.

mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/database/iocBoot/exampleDatabase
mrk> ../../bin/linux-x86_64/exampleDatabase st.cmd 

I then started:

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme -p ca DBRdoubleArray

In another window I executed:
mrk> pvput -p ca DBRdoubleArray 3 1 2 3
Old : DBRdoubleArray 5 3.05948e-308 0 0 0 0
New : DBRdoubleArray 5 1 2 3 0 0
mrk> 

I the monitorme window the complete output is:

mrk> pwd
/home/epicsv4/masterCPP/pvAccessCPP/examples
mrk> O.linux-x86_64/monitorme -p ca DBRdoubleArray
Use provider: ca

^Cmrk> O.linux-x86_64/monitorme -p ca DBRdoubleArray
Use provider: ca
Event DBRdoubleArray [] Changed:{0} overrun:{}
..................................................................
Event DBRdoubleArray [1,2,3] Changed:{0} overrun:{}
^Cmrk> 
 I entered ctl/c to make monitorme exit.

So I do not see any problem.

mdavidsaver, 2017-12-19:
@mrkraimer You need to use the DB I provided which post monitors at a high rate.  This deadlock occurs when the monitorEvent() callback is in progress when the subscription is destroyed.

ralphlange, 2017-12-19:
@mrkraimer Try to follow the instructions for reproducing the issue that Michael put in the original report.
If you're trying something different, not seeing the issue is no surprise.

mrkraimer, 2017-12-20:
I do not see how to create softIocPVA.

But I did reproduce the problem with softIOC

mrk> ll softIOC
lrwxrwxrwx. 1 mrk mrk 65 Dec 20 05:24 softIOC -> /home/epicsv4/masterCPP/epics-base/src/std/O.linux-x86_64/softIoc*
mrk> ./softIOC -d spam2.db 

I am looking at the problem.

mdavidsaver, 2017-12-20:
softIocPVA is build by the pva2pva module.

mrkraimer, 2017-12-20:
The following IS NOT a solution.

I made the change in the epicsAtExit  handler for caProvider
mrk> git diff caProvider.cpp
diff --git a/src/ca/caProvider.cpp b/src/ca/caProvider.cpp
index 3da9c69..613e9b0 100644
--- a/src/ca/caProvider.cpp
+++ b/src/ca/caProvider.cpp
@@ -152,6 +152,7 @@ void ca_factory_cleanup(void*)
     if(DEBUG_LEVEL>0) std::cout << "ca_factory_cleanup\n";
     try {
         ChannelProviderRegistry::clients()->remove("ca");
+epicsThreadSleep(1.0);
         ca_context_destroy(); 
     } catch(std::exception& e) {
         LOG(logLevelWarn, "Error when unregister \"ca\" factory");
mrk> 
Thus the problem appears to be caused if the channels are not deleted before ca_context_destroy is called.

What to do?

In the reference manual for channel access I see 

ca_context_destroy()
...
A program running in an IOC context must delete all of its channels prior to calling ca_context_destroy() to avoid a crash.

So how can I ensure that all channels are destroyed after 
ChannelProviderRegistry::clients()->remove("ca");
is called and before
ca_context_destroy(); 
is called.

mrkraimer, 2017-12-20:
I forgot to mention that adding the epicsThreadSleep did make the monitorme example terminate properly.

mdavidsaver, 2017-12-20:
> So how can I ensure that all channels are destroyed after

The PVA client does this by having the provider/context keep track of open channels.  See ```InternalClientContextImpl::destroyAllChannels()```.

mdavidsaver, 2017-12-20:
Please note that not clearing channels before ca_context_destroy() is a real problem, but separate from the deadlock I reported.

mrkraimer, 2017-12-22:
NOTE: I have not pushed any changes. I still have more to do.

I did implement "destroy all channels".
Now
pvget -m  -p ca X1
terminates without error. Before it had a mutex exception but did terminate.

I have not fixed the deadlock when
O.linux-x86_64/monitorme -p ca X1
terminates.
I am working on fixing this problem.

mrkraimer, 2017-12-23:
I create a pull request #79 it make sure ca_clear_channel is called for all channels when CACProvider exits.
monitorme -p ca X1 still hangs when ctl/c is entered, but with fix to caProvider not always.
I think that there is a problem with monitorme related to signals, e. g.
#ifdef USE_SIGNAL
        signal(SIGINT, sigdone);
        signal(SIGTERM, sigdone);
        signal(SIGQUIT, sigdone);
#endif

I suspect trapping SIGINT creates problems with the channel access threads.

I have created a version of monitorme that is like the original except that it does not trap signals,
It is available at 
https://github.com/mrkraimer/testClientCPP/blob/master/src/monitorme.cpp

It always terminates when ctl/c is entered.

mdavidsaver, 2017-12-24:
Marty, it's a lock ordering bug.  Running

```sh
valgrind --tool=helgrind ./examples/O.linux-x86_64-debug/monitorme -p ca X1
```

lays it out with reasonable clarify.

```
---Thread-Announcement------------------------------------------

Thread #1 is the program's root thread

----------------------------------------------------------------

Thread #1: lock order "0x74C7980 before 0x7511710" violated

Observed (incorrect) order is: acquisition of lock at 0x7511710
   at 0x4C3010C: mutex_lock_WRK (hg_intercepts.c:904)
   by 0x5B945BC: mutexLock (osdMutex.c:46)
   by 0x5B9479D: epicsMutexOsdLock (osdMutex.c:130)
   by 0x5B8BA2C: epicsMutexLock (epicsMutex.cpp:150)
   by 0x5B8BD9F: epicsMutex::lock() (epicsMutex.cpp:273)
   by 0x53F5E1B: epicsGuard<epicsMutex>::epicsGuard(epicsMutex&) (epicsGuard.h:70)
   by 0x540783C: pvac::Monitor::Impl::cancel() (clientMonitor.cpp:87)
   by 0x5406873: pvac::Monitor::cancel() (clientMonitor.cpp:190)
   by 0x111724: (anonymous namespace)::MonTracker::~MonTracker() (monitorme.cpp:134)
   by 0x111795: (anonymous namespace)::MonTracker::~MonTracker() (monitorme.cpp:134)
   by 0x114C6B: std::_Sp_counted_ptr<(anonymous namespace)::MonTracker*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (shared_ptr_base.h:372)
   by 0x115F83: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:150)

 followed by a later acquisition of lock at 0x74C7980
   at 0x4C3010C: mutex_lock_WRK (hg_intercepts.c:904)
   by 0x5B945BC: mutexLock (osdMutex.c:46)
   by 0x5B9479D: epicsMutexOsdLock (osdMutex.c:130)
   by 0x5B8BA2C: epicsMutexLock (epicsMutex.cpp:150)
   by 0x5B8BD9F: epicsMutex::lock() (epicsMutex.cpp:273)
   by 0x50D3A35: epicsGuard<epicsMutex>::epicsGuard(epicsMutex&) (epicsGuard.h:70)
   by 0x50DAF6A: CallbackGuard::CallbackGuard(epicsMutex&) (cacIO.h:159)
   by 0x50F5D53: ca_clear_subscription (ca_client_context.cpp:781)
   by 0x4E753C8: epics::pvAccess::ca::CAChannelMonitor::stop() (caChannel.cpp:1965)
   by 0x54078AE: pvac::Monitor::Impl::cancel() (clientMonitor.cpp:92)
   by 0x5406873: pvac::Monitor::cancel() (clientMonitor.cpp:190)
   by 0x111724: (anonymous namespace)::MonTracker::~MonTracker() (monitorme.cpp:134)

Required order was established by acquisition of lock at 0x74C7980
   at 0x4C3010C: mutex_lock_WRK (hg_intercepts.c:904)
   by 0x5B945BC: mutexLock (osdMutex.c:46)
   by 0x5B9479D: epicsMutexOsdLock (osdMutex.c:130)
   by 0x5B8BA2C: epicsMutexLock (epicsMutex.cpp:150)
   by 0x5B8BD9F: epicsMutex::lock() (epicsMutex.cpp:273)
   by 0x50D3A35: epicsGuard<epicsMutex>::epicsGuard(epicsMutex&) (epicsGuard.h:70)
   by 0x50D394E: callbackManager::callbackManager(cacContextNotify&, epicsMutex&) (cac.h:405)
   by 0x50EA9D9: tcpRecvThread::run() (tcpiiu.cpp:502)
   by 0x5B8A41A: epicsThreadCallEntryPoint (epicsThread.cpp:83)
   by 0x5B92C43: start_routine (osdThread.c:403)
   by 0x4C32D06: mythread_wrapper (hg_intercepts.c:389)
   by 0x6A0D493: start_thread (pthread_create.c:333)

 followed by a later acquisition of lock at 0x7511710
   at 0x4C3010C: mutex_lock_WRK (hg_intercepts.c:904)
   by 0x5B945BC: mutexLock (osdMutex.c:46)
   by 0x5B9479D: epicsMutexOsdLock (osdMutex.c:130)
   by 0x5B8BA2C: epicsMutexLock (epicsMutex.cpp:150)
   by 0x5B8BD9F: epicsMutex::lock() (epicsMutex.cpp:273)
   by 0x53F5E1B: epicsGuard<epicsMutex>::epicsGuard(epicsMutex&) (epicsGuard.h:70)
   by 0x5407AC5: pvac::Monitor::Impl::monitorConnect(epics::pvData::Status const&, std::shared_ptr<epics::pvAccess::Monitor> const&, std::shared_ptr<epics::pvData::Structure const> const&) (clientMonitor.cpp:114)
   by 0x4E7452F: epics::pvAccess::ca::CAChannelMonitor::activate() (caChannel.cpp:1829)
   by 0x4E6B5EE: epics::pvAccess::ca::CAChannel::connected() (caChannel.cpp:309)
   by 0x4E69EEE: epics::pvAccess::ca::ca_connection_handler(connection_handler_args) (caChannel.cpp:52)
[... 36 lines cut ...]
   by 0x54073D4: pvac::Monitor::Impl::Impl(pvac::ClientChannel::MonitorCallback*) (clientMonitor.cpp:45)
   by 0x5408156: std::shared_ptr<pvac::Monitor::Impl> pvac::detail::wrapped_shared_from_this<pvac::Monitor::Impl>::build<pvac::ClientChannel::MonitorCallback*>(pvac::ClientChannel::MonitorCallback*) (clientpvt.h:51)
   by 0x5406C14: pvac::ClientChannel::monitor(pvac::ClientChannel::MonitorCallback*, std::shared_ptr<epics::pvData::PVStructure const>) (clientMonitor.cpp:226)
   by 0x112178: main (monitorme.cpp:258)
 Address 0x7511710 is 0 bytes inside a block of size 48 alloc'd
   at 0x4C2EFE5: calloc (vg_replace_malloc.c:711)
   by 0x5B945E1: epicsMutexOsdCreate (osdMutex.c:75)
   by 0x5B8B6B4: epicsMutexOsiCreate (epicsMutex.cpp:93)
   by 0x5B8BC99: epicsMutex::epicsMutex() (epicsMutex.cpp:251)
   by 0x54073D4: pvac::Monitor::Impl::Impl(pvac::ClientChannel::MonitorCallback*) (clientMonitor.cpp:45)
   by 0x5408156: std::shared_ptr<pvac::Monitor::Impl> pvac::detail::wrapped_shared_from_this<pvac::Monitor::Impl>::build<pvac::ClientChannel::MonitorCallback*>(pvac::ClientChannel::MonitorCallback*) (clientpvt.h:51)
   by 0x5406C14: pvac::ClientChannel::monitor(pvac::ClientChannel::MonitorCallback*, std::shared_ptr<epics::pvData::PVStructure const>) (clientMonitor.cpp:226)
   by 0x112178: main (monitorme.cpp:258)
 Block was alloc'd by thread #1
```

The basic requirement which is being violated is what the ChannelProvider API rules places all the of burden of avoiding lock order issues on the ChannelProvider implementation (not my design).  This requires that the provider can't hold _any_ locks when making callbacks.

Unfortunately, libca doesn't quite play by the same rules.  Specifically, the various ca_clear_*() calls take a lock which is also held during event callbacks.

My pvac wrappers takes advantage of these rules by not unlocking unnecessarily, including when calling Monitor::stop().

Thus the deadlock.

In my mind the "rules" say that caProvider needs to be changed.  Though, given how libca does locking, I can't see how this can be done.

mrkraimer, 2018-01-31:
I think the problem is that monitorme calls signal.
It traps SIGINT, SIGTERM and SIGQUIT.
This causes problems with epicsAtExit.

Look at
https://en.wikipedia.org/wiki/Sigaction

It says:

Replacement of deprecated signal()
The sigaction() function provides an interface for reliable signals in replacement of the unreliable and deprecated signal() function. Signal handlers installed by the signal() interface will be uninstalled immediately prior to execution of the handler. Permanent handlers must therefore be reinstalled by a call to signal() during the handler's execution, causing unreliability in the event a signal of the same type is received during the handler's execution but before the reinstall. Handlers installed by the sigaction() interface can be installed permanently and a custom set of signals can be blocked during the execution of the handler. These signals will be unblocked immediately following the normal termination of the handler (but not in the event of an abnormal termination such as a C++ exception throw.)

mdavidsaver, 2018-01-31:
> I think the problem is that monitorme calls signal.
> It traps SIGINT, SIGTERM and SIGQUIT.
> This causes problems with epicsAtExit.

Please elaborate on these problem(s).  What is the mechanism involved?

mrkraimer, 2018-01-31:
If I make the change
mrk> git diff examples/monitorme.cpp
...
-#define USE_SIGNAL
+//#define USE_SIGNAL  // THIS removes calls to signal
 ...

Then monitorme  terminates when I enter ctl/c

I do not know why trapping SIGINT, SIGTERM and SIGQUIT with signal causes the failure.

mdavidsaver, 2018-01-31:
Marty, the deadlock only happens on normal monitor shutdown.  The purpose of the signal handler is to turn ctrl+c into a normal shutdown.  Otherwise you are just killing the process, and none of the cleanup code runs.

mrkraimer, 2018-02-08:
I have made changes for termination and error handling.
I have issued a pull request for these changes.
I have been using pvAccess/examples/monotorme and exampleCPP/exampleClient/monitor to test.

Before I made changes to ca provider I did get monitor to somtimes  hang at termination.
With the changes I made I have no longer seen it hang.

monitorme still hangs

https://github.com/mrkraimer/testClientCPP

has tests for monitor and monitorme.
The code with the name monitor are like exampleCPP/exampleClient/monitor.

The code with the name monitorme are a simplified version of  pvAccess/examples/monotorme,
But signals are not used and no work queue is used.
This was done so that signals and the work queue can be ruled out as a source of termination problems.

Both monitor and monitorme terminate when the enter key is pressed.

There is also code named monitorate and monitormerate.
These are used to measure performance.

There are also files fast1.db, ..., fast4.db that can be used ti start up to 4 IOCs with a  calc record that execute at a very high rate.

Look at:

https://github.com/mrkraimer/testClientCPP/blob/master/performResult

It shows that

1) monitor handles more events than monitorme,
2) For monitor provider ca handles more events than provider pva
3) For monitorme provider pva handles more events than provider ca

anjohnson, 2018-02-20:
> The basic requirement which is being violated is that the ChannelProvider API rules place all the of burden of avoiding lock order issues on the ChannelProvider implementation (not my design). This requires that the provider can't hold any locks when making callbacks.

@mdavidsaver Can you point out where this "basic requirement" comes from?

The lock held by libCa during callbacks protects the client context (cac.mutex), and I think is taken to ensure that monitor callbacks from 2 different IOCs can't get run simultaneously. It's safe to call back into CA from a callback though because the epicsMutex is recursive.

The following change to your pvac code converts the hang on Ctrl-C into one or more LOG messages: `Unhandled exception following exception in ClientChannel::MonitorCallback::monitorEvent(): tr1::bad_weak_ptr`. I don't completely understand how it triggers this specific exception, but this might form the basis of a solution:

```patch
diff --git a/src/client/clientMonitor.cpp b/src/client/clientMonitor.cpp
index 2a7c321..7f9c642 100644
--- a/src/client/clientMonitor.cpp
+++ b/src/client/clientMonitor.cpp
@@ -88,11 +88,12 @@ struct Monitor::Impl : public pva::MonitorRequester,
 
             last.reset();
 
-            if(started && op) {
-                op->stop();
+            temp.swap(op);
+            if (started && temp) {
                 started = false;
+                UnGuard U(G);
+                temp->stop();
             }
-            temp.swap(op);
 
             callEvent(G, MonitorEvent::Cancel);
         }
```

mdavidsaver, 2018-02-20:
> @mdavidsaver Can you point out where this "basic requirement" comes from?

http://mdavidsaver.github.io/pvAccessCPP/providers.html#provider_roles_requester_locking

```
Operations methods may call requester methods, and vis versa. The following rules must be followed to avoid deadlocks.

*   No locks must be held when Requester methods are called.
*   Locks may be held when Operation methods are called.
```

mdavidsaver, 2018-02-21:
FYI. the first time I came across this situation was with the sequencer https://epics.anl.gov/tech-talk/2015/msg00008.php

Locking of this mutex in ca_clear_channel() was introduced by epics-base/epics-base@5e5ea11658b30feeafb2f5bb075b23e6f8663a4d as part of https://bugs.launchpad.net/epics-base/+bug/1179642 which when into Base 3.14.12.3, 3.15.0.2, and all later releases.

anjohnson, 2018-02-21:
The purpose of the two mutexes in the CAC were succinctly described by Jeff [in 2012](https://epics.anl.gov/tech-talk/2012/msg00581.php) as:

> There are two global synchronization locks in the client library. A general mutual exclusion lock is held only briefly to enforce data structure consistency between threads. A callback control lock is held for much longer while a sequence of response messages are dispatched. It is used to enforce that only one callback at a time runs, to enforce non-preemptive callback mode, and to enforce that a callback isn't in progress at certain critical junctures.

Some of the changes he applied for [lp:1179642](https://bugs.launchpad.net/epics-base/+bug/1179642l) make it easy to distinguish between the two global locks described above (although there are also a shipload of other changes which make reading the patch painful). I have no time to try and understand or change the CA client library locking scheme; CA matters to us and will do for quite a long time, but we're Okay with working around its idiosyncrasies instead of fixing them.

The CA provider only calls `ca_clear_channel()` in its `CAChannel::disconnectChannel()`method which is public but not virtual and will usually only be called from the CAChannel destructor, at which time I think the `ClientChannel::Impl.mutex` should normally be released anyway.

This particular issue is stress-testing the simultaneous actions of a monitor callback being run while its subscription is being cancelled from another thread. I don't see any analogue in the CA Provider channel disconnect code that could cause a deadlock there unless a client application explicitly calls the `disconnectChannel()` method, which it can only do through the polymorphic interface by destroying the channel.

Of course I may be misunderstanding your point and/or the code.

If i work up a fix that includes the change I posted above, catches the `bad_weak_ptr` exception and adds an exception for stopping monitors to your "Locks may be held when Operation methods are called" rule, will you accept it as a fix for this issue?

mdavidsaver, 2018-02-21:
> Of course I may be misunderstanding your point and/or the code.
>
> If i work up a fix that includes the change I posted above, catches the bad_weak_ptr exception and adds an exception for stopping [...]

@anjohnson We should talk about this.  My ownership rules changes last summer resulted in objects being delete'd before they previously would have.  The result being immediately noticeable (nothing happens).  The rules change you are proposing would not be immediately noticeable (a new deadlock in previously safe code).  It would also be the single exception to an otherwise uniform and succinct rule.  I'm reluctant to agree to a change of this nature.

My preference would be that this caProvider specific issue have a caProvider specific resolution (eg. call ca_clear_* from a worker thread).

anjohnson, 2018-02-23:
Since this issue is really a libCa problem I wrote a test program for the CA API that quickly demonstrates the deadlock, which _only_ occurs when the `ca_clear_subscription()` routine is called. The `ca_clear_channel()` routine doesn't trigger any callbacks so it can't deadlock.

I just filed a description and my demo program as [a bug against CA](https://bugs.launchpad.net/epics-base/+bug/1751380), in the hope that Jeff might take a look at it. That would be the best place to fix it, and if his private (forked) version of the client library suffers the same problem he may be willing to work on a fix, or at least tell us to document the limitation. The latter would mean we _have_ to fix this in the client code, but I would rather not add a separate worker thread if we can get it fixed upstream.

anjohnson, 2018-02-27:
`CAChannelMonitor::stop()` calls `ca_clear_subscription()` which takes cac::mutex, the other mutex involved in the deadlock. I might not be understanding your point here, but I see nothing inconsistent with my CA bug report or demo code. No response from Jeff yet, would Bruce be willing to ask him...?

closed 2019-08-07 by mdavidsaver