epics-base/pvAccessCPP #82: pvlist fails noisily when pointed at a host with no servers running

anjohnson, 2018-01-12:
The `pvlist` program shouldn't be spewing a stack-dump when asked to talk to a host that isn't running any PVA servers:
```
tux% bin/linux-x86_64/pvlist localhost
2018-01-12T12:03:23.683 getTransport() fails: Failed to verify TCP connection to '127.0.0.1:5075'.
On line 115 of ../../src/remote/blockingTCPConnector.cpp
/epics/base/7.0/lib/linux-x86_64/libpvAccess.so.6.0.1(+0x45f09) [0x7f07418e4f09]
/epics/base/7.0/lib/linux-x86_64/libpvAccess.so.6.0.1(+0x70020) [0x7f074190f020]
/epics/base/7.0/lib/linux-x86_64/libpvAccess.so.6.0.1(+0x90ee2) [0x7f074192fee2]
/epics/base/7.0/lib/linux-x86_64/libpvAccess.so.6.0.1(+0xa5c87) [0x7f0741944c87]
/epics/base/7.0/lib/linux-x86_64/libpvData.so.7.0.1(_ZN5epics6pvData5Timer3runEv+0x10d) [0x7f07415a770d]
/epics/base/7.0/lib/linux-x86_64/libCom.so.3.17.0(+0x33039) [0x7f07410be039]
/epics/base/7.0/lib/linux-x86_64/libCom.so.3.17.0(+0x3941c) [0x7f07410c441c]
/lib64/libpthread.so.0(+0x7e25) [0x7f0740293e25]
/lib64/libc.so.6(clone+0x6d) [0x7f07405a034d]

Error: connection timeout
```

mdavidsaver, 2018-01-16:
This is a certainly a nuisance.  The trouble is that this message is a generic catch all when anything goes wrong on connection setup.

blockingTCPConnector.cpp needs some work in general as eg. it's making blocking connect() calls, with potentially long timeout.  Also, the whole NamedLockPattern thing seems unnecessary.

mdavidsaver, 2018-02-01:
> Also, the whole NamedLockPattern thing seems unnecessary.

And full of bugs.  When this exception is thrown, it's leaving a mutex locked.

mdavidsaver, 2018-02-02:
The message is now reduced (no stack trace) and only shown when debug logging is enabled (eg. "pvlist -d ...").

I've also replaced use of NamedLockPattern to make this exception safe, and noted what this code is actually doing.

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver