epics-base/pvAccessCPP #87: unknown provider name 'pva'

mdavidsaver, 2018-01-31:
Apparently during static linking an anonymous namespace can optimized out even if it has side-effects (global w/ ctor).  At least by g++

At issue is pvaloaderinstance in clientFactory.cpp

mdavidsaver, 2018-01-31:
Ok, this is a stupid one on my part.  During static linking, ld entirely ignores clientFactory.o as none of it's symbols are used by pvlist and friends.  Global ctors are retained if _any_ symbol from a file is used.  Apparently this is a long standing "feature" of binutils (http://www.sourceware.org/ml/binutils/2003-02/msg00155.html).

mdavidsaver, 2018-01-31:
Should be fixed with abce3faa8675c92b9e8b00a26ad17b81aed84557

mdavidsaver, 2018-01-31:
Well, I can't claim that this was new to me either.  I'd just forgotten.

hhslepicka, 2018-01-31:
Thank you for the fix @mdavidsaver . I can confirm that the fix work since I tested it here at SLAC.

mdavidsaver, 2018-01-31:
Thanks Hugo.

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver