epics-base/pvAccessCPP #89: fair_queue::clear() leads to stack overflow

mdavidsaver, 2018-02-14:
A combination of fair_queue::clear() and fair_queue::pop_front_try() can chain
nodes being removed from the queue into an effective linked list of shared_ptr
through the now unused fair_queue::entry.  This list is only destroyed when
the final entry is removed.  Then destruction is recursive.  If the queue
was long enough, this will overflow the stack (eg. after some 145970 frames).

Triggered from BlockingServerTCPTransportCodec::destroyAllChannels()

mdavidsaver, 2018-02-14:
Should be fixed by c6b1d3b738ea330bac4886adb6fe6c7e7773547e

anjohnson, 2018-02-14:
I will ask Tom to update and give this a try. Would you be able to explain (if asked, I'm not actually asking here) why p4p was triggering this but pvaPy wasn't?

mdavidsaver, 2018-02-14:
> I will ask Tom

Please also ask him to report any "warning" messages.  I've started seeing a new one when running his scale test.  I think this is not caused by, though perhaps exposed by, this fix.

```
2018-02-13T18:01:37.944 unprocessed read buffer from client at ../../src/remote/codec.cpp:300: 10.0.142.1:5075, disconnecting...
```

tfors, 2018-03-25:
@mdavidsaver @anjohnson 
I can confirm that this fixes the crash of the IOC, and I too get the same warning message:

```
2018-03-25T10:53:02.934 unprocessed read buffer from client at ../../src/remote/codec.cpp:300: 172.20.10.2:5075, disconnecting...
```

mdavidsaver, 2018-03-26:
@tfors Thanks.

I'm going to treat the message as a lower priority issue.

> Would you be able to explain (if asked, I'm not actually asking here) why p4p was triggering this but pvaPy wasn't?

No, I can't explain.  Issues like this arise because there are different possible sequences to shutdown.  eg. Drop Channel refs. first, then ChannelProvider or vis. versa.  Explicitly calling ```->destroy()```, while no longer necessary, adds another dimension of possible code paths.  And then there are potential races.  Dropping the last Channel doesn't always wait for worker threads to exit, but dropping the ChannelProvider does.

The short answer is that I'd have a hard time predicting, for given client code, how shutdown will proceed.

mdavidsaver, 2018-03-26:
Actually, this is assuming that pvaPy even destroys the ChannelProvider at all.  P4P does because I want to verify that this process works.

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver