epics-base/pvAccessCPP #90: fairQueue.h and NDEBUG

anjohnson, 2018-02-14:
The definition of the `fair_queue::entry::owner` field appears inside `#ifndef NDEBUG` but both the `fair_queue::push_back()` and `fair_queue::pop_front_try()` methods use `P->owner` without that qualification. This code will not build if someone defines NDEBUG.

mdavidsaver, 2018-02-14:
True, however this seems to be a moot point as NDEBUG is not, and apparently never has been, defined by the Base makefiles.

mdavidsaver, 2018-02-14:
685490fc1abc03e4d704037a1be698168d12bdb3

anjohnson, 2018-02-14:
Thanks. My solution would have been to add conditionals, not remove them, since defining NDEBUG still removes the checks of the owner field which all appear inside assert() statements, but this is your module.

We try to ensure that our code still builds and works with NDEBUG defined, although as you say we don't define it ourselves (Till Strauman sometimes does). There have been tech-talk and core-talk discussions on the subject, [this one](https://epics.anl.gov/core-talk/2008/msg00013.php) for example.

closed 2018-02-14 by anjohnson