epics-base/pvAccessCPP #93: pvput doesn't work for NTEnum

mdavidsaver, 2018-02-27:
@ttkorhonen report

> This may be a totally stupid and out of date question but it puzzles me nevertheless.
> 
> I tried pvput to a bo record on a EPICS 7 setup (using qsrv), built from 7.0.1.1. tarball. The pvput with string fails with a message:
> 
>> pvput timo:step YES
> Old : timo:step                      NO
> Error: parseToPOD: No digits to convert
> New : timo:step                      NO
> 
> Putting a numerical value (0 or 1 in this case) works.
> Have I forgotten something? I thought this was working earlier.
> 
> Timo

mdavidsaver, 2018-02-27:
A regression introduced in 6.0.0.  The issue is that the value.choices array isn't being fetched.

mdavidsaver, 2018-02-27:
74fbd22d40d54a4c63dd63f210628b1ef1b8267d saves the previously fetched value (now always fetched) and extract ```values.choices``` as needed.

mdavidsaver, 2018-02-27:
```sh
$ ./bin/linux-x86_64/pvput binary ONE
Old : binary                         ZERO
New : binary                         ONE
```

msekoranja, 2018-02-28:
Initial idea was: pvput does know nothing about normativeTypes (and NTEnum is a normative types), eget does.

ralphlange, 2018-02-28:
I like that initial idea, as it is simple.
caput handles that case by always sending a string, the server does the conversion.
Did this work with pvput against pvaSrv? Then pvaSrv was probably handling it the same way?!

msekoranja, 2018-02-28:
As a recall NTEnum is defined as "int index; String[] labels". So, if you want to do a put you need to write to an "index" field. Therefore you need to get labels in order to do a conversion on the client side.  There is no server side conversion.

mdavidsaver, 2018-02-28:
Simple ideas are nice, however what is simple for developers isn't always simple for users.  I don't want to have to explain to a user that enumeration strings no longer work at CLI.  The expectation of users (as expressed by @ttkorhonen) is clear.  Handling of enumerations has long been transparent, and this needs to be continued.

I like the idea of having clients handle type conversions, with this exception.  imo. the proper "fix" would be to redefine NTEnum around a Union so that clients can send the string, and the server can handle mapping between string and index.  This mapping is trivially easy (and atomic) on the server side as the choices list is readily available, while on the client side this is not so.

hhslepicka, 2018-03-22:
@mdavidsaver I can confirm that with the version available at master it works.

```bash
slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/ $ cat a.db
record(mbbo, "TEST:ENUM")
{
   field(PINI, "YES")
   field(ZRST, "YES")
   field(ZRVL, "0")
   field(ONST, "NO")
   field(ONVL, "1")
   field(TWST, "MAYBE")
   field(TWVL, "2")
   field(VAL,  "0")
}

#### Start Test IOC
slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs  $ softIocPVA -d a.db

#### TESTS
slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ caget TEST:ENUM
TEST:ENUM                      YES

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ caput TEST:ENUM NO
Old : TEST:ENUM                      YES
New : TEST:ENUM                      NO

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ caput TEST:ENUM YES
Old : TEST:ENUM                      NO
New : TEST:ENUM                      YES

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ caput TEST:ENUM MAYBE
Old : TEST:ENUM                      YES
New : TEST:ENUM                      MAYBE

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ pvget TEST:ENUM
TEST:ENUM                      MAYBE

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ pvput TEST:ENUM NO
Old : TEST:ENUM                      MAYBE
New : TEST:ENUM                      NO

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ pvput TEST:ENUM 0
Old : TEST:ENUM                      YES
New : TEST:ENUM                      YES

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ pvput TEST:ENUM 1
Old : TEST:ENUM                      YES
New : TEST:ENUM                      NO

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ pvput TEST:ENUM 2
Old : TEST:ENUM                      NO
New : TEST:ENUM                      MAYBE

slepicka@dhcp-swlan-public-81-101:~/sandbox/iocs/bpm  $ pvput TEST:ENUM value.index=0
Old : TEST:ENUM                      MAYBE
New : TEST:ENUM                      YES

```

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver