epics-base/pvAccessCPP #94: pvget et al. don't detect multiple servers

mdavidsaver, 2018-03-22:
The PVA client doesn't seem to detect/notify when two servers answer a search request.  I haven't look in depth to find if this is attempted or not.

```sh
$ caget cnt
CAC: error = "Operation not permitted" sending UDP msg to 198.129.223.255:5064
cnt                            9
CA.Client.Exception...............................................
    Warning: "Identical process variable names on multiple servers"
    Context: "Channel: "cnt", Connecting to: 10.0.142.1:41653, Ignored: 10.0.142.1:41993"
    Source File: ../cac.cpp line 1306
    Current Time: Thu Mar 22 2018 10:38:50.918611605
..................................................................
$ ./bin/linux-x86_64/pvget cnt
cnt                            14
```

mdavidsaver, 2019-09-27:
The closest to this check is

https://github.com/epics-base/pvAccessCPP/blob/866b75a36de52a7126204c0f8a45e7f67ff51a01/src/remoteClient/clientContextImpl.cpp#L3633-L3638

Which does seem to detect if same server replies twice, eg. because client configuration sends it both a broadcast and unicast.

kasemir, 2019-09-30:
With base-7.0.2.2, I get a "warning: More than one channel .." from pvget, but not pvmonitor. Would have expected the other way around, because pvmonitor keeps running and should thus see replies from duplicate servers, while pvget might already have executed the 'get' and shut down when duplicate search responses arrive.

In any case, added a check to the Java client, which now logs a warning like this:
```
2019-09-30 11:18:41 WARNING [org.epics.pva] More than one channel with name 'ramp' detected, connected to /10.158.80.148:49998 GUID 0x4715925D00000000A64EA727, ignored /10.158.80.148:5075 GUID 0x4115925D0000000066811418
```