epics-base/pvAccessCPP #99: Compilation failure when using -std=c++11

brunoseivam, 2018-05-01:
The following snippet compiles fine if not using `-std=c++11`, fails otherwise.

pvAccess version:

```
martins@martins-win7:~/e/e/m/pvAccess|HEAD✓                                                          
➤ git status
HEAD detached at 51fdd50
```

Code that triggers it:

```
martins@martins-win7:~/test-clientprovider                                                                                            
➤ cat testApp/src/Makefile 
TOP=../..

include $(TOP)/configure/CONFIG

PROD_IOC = test

test_LIBS += pvAccess pvData
test_SRCS += main.cpp

USR_CXXFLAGS += -std=c++11

include $(TOP)/configure/RULES

martins@martins-win7:~/test-clientprovider                                                                                            
➤ cat testApp/src/main.cpp 
#include <pva/client.h>

int main(int argc, char **argv)
{
    pvac::ClientProvider p("pva");
    return EXIT_SUCCESS;
}

```

g++ complaints:

```
/usr/bin/g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux     -O3 -g   -Wall     -std=c++11  -mtune=generic     -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include -I/home/martins/epics/base/include/compiler/gcc -I/home/martins/epics/base/include/os/Linux -I/home/martins/epics/base/include        -c ../main.cpp
/usr/bin/g++ -o test  -L/home/martins/test-clientprovider/lib/linux-x86_64 -L/home/martins/epics/epics-base/lib/linux-x86_64 -Wl,-rpath,/home/martins/test-clientprovider/lib/linux-x86_64 -Wl,-rpath,/home/martins/epics/epics-base/lib/linux-x86_64           -rdynamic -m64            main.o    -lpvAccess -lpvData   
main.o: In function `main':
/home/martins/test-clientprovider/testApp/src/O.linux-x86_64/../main.cpp:5: undefined reference to `pvac::ClientProvider::ClientProvider(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<epics::pvAccess::Configuration> const&)'
collect2: error: ld returned 1 exit status
```
g++ version: `g++ (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609`

I don't know enough of C++ to know where to start to look for this issue...

mdavidsaver, 2018-05-01:
> USR_CXXFLAGS += -std=c++11

At the moment use of c++11 is all or nothing with the PVA modules due to the condition use of some types based on standard version.  ```std::shared_ptr``` vs. ```std::tr1::shared_ptr```, and ```std::unique_ptr``` vs ```std::auto_ptr```.

Are you able to rebuild everything (or at least pvDataCPP and all of it's dependent modules) with '-std=c++11'?

anjohnson, 2018-05-01:
Regarding @mdavidsaver 's "At the moment": I doubt it's worth trying to make it possible to switch the C++ standard at will when using gcc 5. Toggling that flag causes the compiler to use a [completely different implementation](https://gcc.gnu.org/onlinedocs/gcc-5.2.0/libstdc++/manual/manual/using_dual_abi.html) of several template classes including `std::string` and `std::list`.

mdavidsaver, 2018-05-02:
Ok, yet another example of why maintaining (or even understanding) ABI compatibility with c++ is a lost cause.

closed 2018-05-02 by mdavidsaver