epics-base/pvAccessCPP #101: Thread priorities not set

mdavidsaver, 2018-05-18:
The PVA worker threads, server and client, are all hard-coded to ```epicsThreadPriorityMedium```.  They need to have proper priorities assigned as with the CA workers.

mdavidsaver, 2018-05-18:
I have no immediate plans to do this.

mdavidsaver, 2018-06-30:
I had a quick look at what libca and RSRV do.  I was surprised to see that libca context threads key off the priority of the thread which creates the context (cf. initializingThreadsPriority).  Should the PVA client do this as well?

@anjohnson @ralphlange

anjohnson, 2018-07-02:
> Should the PVA client do this as well?

That would be good, although not a priority. I suspect that a CPU-starved clients are probably less common now than they were when Jeff made the client code multi-threaded, and testing whether the result performs better under those conditions could be challenging, but it is certainly desirable if that will help tune the code so it works better under degraded circumstances.