epics-base/pvAccessCPP #102: Stack overflows observed on RTEMS

mdavidsaver, 2018-06-04:
SLAC reports seeing strong evidence of stack corruption in analysis of crashes.  The TCP-rx and/or TCP-tx threads are suspected.

While not directly comparable with RTEMS/ppc, on Linux/amd64 I observed (with valgrind/DRD) that TCP-rx used up to ~7KB of stack compared to 2-3KB each for the other threads (eg. scanning, errlog, etc.).

till-s, 2018-06-04:
It's not so much the TCP-rx / TCP-tx thread code itself but the fact that c++ exception handling  / stack unwinding (at least under RTEMS/ppc) requires a considerable amount of stack space. Empirical tests indicated that a simple 
try {
   throw something;
} ( catch something &s ) {
}

requires ~6kB; if the exception is rethrown ~8kB were needed.

mdavidsaver, 2018-06-04:
@till-s FYI libca has been throwing/catching for quite some time without (apparently) overflowing the default .  Including at least one re-throw (in dbContext.cpp).

I do notice though that all of the scan and CA threads use epicsThreadStackBig or Medium.  The PVA threads are all defaulting to Small.

mdavidsaver, 2018-06-04:
c8292caf915a9a97bcf10b702701ad2e074925b5 Changes TCP-rx and TCP-tx to ```epicsThreadStackBig```

mdavidsaver, 2018-06-04:
It occurs to me that, like epics-base/pvDataCPP#54, there is an element of hard-float/soft-float here as well.  Could you repeat your analysis on a soft-float target?

till-s, 2018-06-04:
I havent' looked at libca in detail, so I can't comment.

I am, however, certain that stack unwinding caused the overflow we observed here. I analyzed multiple stack traces and found:
 - at the time of the overflow the code called 'explicitly' from the TCP thread used only a small amount (<1k)
 - the stack unwinding code used enough to overflow the small stack allocated to the thread. From looking at the assembly as well as the source code it was obvious that the dwarf2 unwinder uses significantly sized stack frames.

till-s, 2018-06-04:
The connection to hard-/soft-float is not obvious to me - could you please elaborate how this relates to the stack overflow we are discussing here?

mdavidsaver, 2018-06-04:
I thought that with PPC some FP registers pushed to stack during function calls?

mdavidsaver, 2018-06-25:
The increased stack sizes, along with an RTEMS fix, are reported to give stable operation of a (patched) RTEMS 4.10.1.

https://devel.rtems.org/ticket/3449

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver