epics-base/pvAccessCPP #109: Build failure on MinGW gcc-7.1.0 (native build, not cross)

anjohnson, 2018-06-29:
I just set up MinGW builds on my recently upgraded Windows 10 system and installed Strawberry Perl 5.26.2 which comes with gcc-7.1.0. The base modules all build fine (one fix committed to 3.15 for epicsCalcTest.cpp) but the pvAccess build dies in pvtoolsSrc:

```
gmake[4]: Entering directory 'C:/epics/base-7.0/modules/pvAccess/pvtoolsSrc/O.windows-x64-mingw'
g++           -I../../src/utils -I../../src/remote -DNOMINMAX   -D_MINGW     -O3   -Wall      -m64   -DEPICS_BUILD_DLL -DEPICS_CALL_DLL   -I. -I../O.Common -I. -I. -I.. -IC:/epics/base-7.0/include/compiler/gcc -IC:/epics/base-7.0/include/os/WIN32 -IC:/epics/base-7.0/include -IC:/epics/base-7.0/include/compiler/gcc -IC:/epics/base-7.0/include/os/WIN32 -IC:/epics/base-7.0/include        -o pvget.obj -c ../pvget.cpp
In file included from C:/Strawberry/c/lib/gcc/x86_64-w64-mingw32/7.1.0/include/c++/iostream:39:0,
                 from ../pvget.cpp:1:
C:/Strawberry/c/lib/gcc/x86_64-w64-mingw32/7.1.0/include/c++/ostream: In instantiation of 'struct std::__is_convertible_to_basic_ostream<pvutil_ostream&>':
C:/Strawberry/c/lib/gcc/x86_64-w64-mingw32/7.1.0/include/c++/ostream:656:5:   required by substitution of 'template<class _Ostream, class _Tp> typename std::enable_if<std::__and_<std::__not_<std::is_lvalue_reference<_Tp> >, std::__is_convertible_to_basic_ostream<_Ostream>, std::__is_insertable<_Ostream&, const _Tp&, void> >::value, typename std::__is_convertible_to_basic_ostream<_Tp>::ostream_type>::type std::operator<<(_Ostream&&, const _Tp&) [with _Ostream = pvutil_ostream&; _Tp = epics::pvData::format::indent]'
../pvutils.h:144:44:   required from here
C:/Strawberry/c/lib/gcc/x86_64-w64-mingw32/7.1.0/include/c++/ostream:625:23: error: 'std::basic_ostream<char>' is an inaccessible base of 'pvutil_ostream'
       decltype(__check(declval<typename remove_reference<_Tp>::type*>()));
                ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Strawberry/c/lib/gcc/x86_64-w64-mingw32/7.1.0/include/c++/ostream:625:23: error: 'std::basic_ostream<char>' is an inaccessible base of 'pvutil_ostream'
gmake[4]: *** [C:/epics/base-7.0/configure/RULES_BUILD:240: pvget.obj] Error 1
```

It's complaining about this class declaration in pvutils.h:

```C++
#include <ostream>
#include <iostream>

// usage: pvutil_ostream myos(std::cout.rdbuf());

class pvutil_ostream : private std::ostream
{
public:
    pvutil_ostream(std::streambuf* sb)
        : std::ostream(sb)
    {}
```
If I change the `private` derivation into `public` the build succeeds, but of course the result doesn't actually mean quite the same thing. I can't tell whether it works yet because of an apparent network problem with the client code, which I will report separately.

mdavidsaver, 2018-06-29:
Well... pvutil_ostream seems a rather odd way to do custom formating.  Mixing operator overloads, function argument list overloads, and template vs. non-template methods overloads.

mdavidsaver, 2018-06-29:
Maybe fixed with f29ef3121e9e32f4fcf8f97551be9ff21e792fa4 which switches from sub-classing to composition.  This approach has worked for me in the past, through with less overloading.

anjohnson, 2018-07-03:
Yup, thanks.

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver