epics-base/pvAccessCPP #118: ca provider: longin is reported as short

brunoseivam, 2018-07-30:
```
$ cat ~/test.db
record(ai, "TEST:ai") {
    field(VAL,  "1")
    field(PINI, "YES")
}

record(longin, "TEST:li") {
    field(VAL,  "32768")
    field(PINI, "YES")
}
$ ./bin/linux-x86_64-debug/softIocPVA -d ~/test.db
Starting iocInit
############################################################################
## EPICS R7.0.1.2-DEV
## EPICS Base built Jul 27 2018
############################################################################
iocRun: All initialization complete
2018-07-30T11:14:14.041 Using dynamically assigned TCP port 60191.
epics> dbl
TEST:li
TEST:ai
epics>
```
 On another terminal

```
$ ./bin/linux-x86_64-debug/caget TEST:li 2> /dev/null
TEST:li                        32768
$ ./bin/linux-x86_64-debug/pvget TEST:li 2> /dev/null
TEST:li                        32768
$ ./bin/linux-x86_64-debug/pvget -p ca TEST:li 2> /dev/null
TEST:li                        -32768
```

mrkraimer, 2018-07-30:
Thanks for finding this. It now works in

https://github.com/mrkraimer/pvAccessCPP

```
mrk> pvget -p ca -i DBRint00
DBRint00                       32768

```

anjohnson, 2018-07-30:
Hi Marty, thanks for looking at this.

By committing the fix to your master branch though, where you already have the channelConnect fixes, you are mixing up two different issues which could get us into a similar mess that we had before with a single branch containing several different sets of caProvider changes. As it is right now you can't create Github pull requests from your master branch that fix either this issue or issue #117 .

I would also like to see some tests added which check for these kinds of data conversion faults in the caProvider code, and it would make sense to do that and provide this fix on a completely separate branch. Branches in git are _really_ easy to create and switch between (you don't have to create a new working directory, just do a `git checkout <branch>` to switch, which is almost instantaneous, and then rebuild) — could I get you to move this work to a new branch?

You would first checkout the official pvAccess master branch as a new local branch, write and commit some tests checking for data type conversion faults (some of which should fail for the `DBR_LONG` type because of this bug), then separately commit your fixes, which should repair the test failures. The result would be a branch that is self-contained and self-testing, makes the caProvider more robust against regressions in the future, and can be merged into master independent of any other work.

I can help you with more details in private email if you agree.

mrkraimer, 2018-07-31:
Is this correct?
```
git clone https://github.com/epics-base/pvAccessCPP.git pvAccessCPPissue117
cd pvAccessCPPissue117
git branch issue117
git checkout issue117
git branch
* issue117
  master

```

mrkraimer, 2018-07-31:
On further thought.
From gihub itself create a new branch named issue118

Then
```
git clone https://github.com/epics-base/pvAccessCPP.git 
cd pvAccessCPP
git checkout issue118
```

Is this a better way than the last message?

ralphlange, 2018-07-31:
Equivalent.
Creating a branch remotely and pulling it is the same as creating it locally and pushing it.

mdavidsaver, 2019-01-08:
Released with Base 7.0.2

closed 2019-01-08 by mdavidsaver