epics-base/pvAccessCPP #119: ca provider: pvinfo doesn't work

brunoseivam, 2018-07-30:
```
$ cat ~/test.db
record(ai, "TEST:ai") {
    field(VAL,  "1")
    field(PINI, "YES")
}

record(longin, "TEST:li") {
    field(VAL,  "32768")
    field(PINI, "YES")
}
$ ./bin/linux-x86_64-debug/softIocPVA -d ~/test.db
Starting iocInit
############################################################################
## EPICS R7.0.1.2-DEV
## EPICS Base built Jul 27 2018
############################################################################
iocRun: All initialization complete
2018-07-30T11:14:14.041 Using dynamically assigned TCP port 60191.
epics> dbl
TEST:li
TEST:ai
epics>
```

On another terminal:

```
$ ./bin/linux-x86_64-debug/pvinfo -p ca TEST:li
[TEST:li] failed to get channel introspection data
$ ./bin/linux-x86_64-debug/pvinfo TEST:li
CHANNEL  : TEST:li
STATE    : CONNECTED
ADDRESS  : 35.9.126.36:60191
epics:nt/NTScalar:1.0
    int value
    alarm_t alarm
        int severity
        int status
        string message
    time_t timeStamp
        long secondsPastEpoch
        int nanoseconds
        int userTag
    display_t display
        double limitLow
        double limitHigh
        string description
        string format
        string units
    control_t control
        double limitLow
        double limitHigh
        double minStep
    valueAlarm_t valueAlarm
        boolean active
        int lowAlarmLimit
        int lowWarningLimit
        int highWarningLimit
        int highAlarmLimit
        int lowAlarmSeverity
        int lowWarningSeverity
        int highWarningSeverity
        int highAlarmSeverity
        int hysteresis
```

mrkraimer, 2018-07-30:
This looks like a problem with pvinfo.
The following works:
```
mrk> pwd
/home/epicsv4/masterCPP/exampleCPP/exampleClient
mrk> bin/linux-x86_64/getField -p ca DBRint00
provider ca channelName PVRdouble debug false
_____getField starting__
Type exit, or a fieldname, or just enter return
value
getDone
int
Type exit, or a fieldname, or just enter return

getDone
structure
    int value
    alarm_t alarm
        int severity
        int status
        string message
    display_t display
        double limitLow
        double limitHigh
        string description
        string format
        string units
    control_t control
        double limitLow
        double limitHigh
        double minStep
    valueAlarm_t valueAlarm
        boolean active
        int lowAlarmLimit
        int lowWarningLimit
        int highWarningLimit
        int highAlarmLimit
        int lowAlarmSeverity
        int lowWarningSeverity
        int highWarningSeverity
        int highAlarmSeverity
        int hysteresis

```

mrkraimer, 2018-07-30:
I take it back (sort of).
When the changes were made t allow createChannelGet, etc to be called before a channel connects, getField was not changed properly in ca provider.

I will fix this.

mdavidsaver, 2018-07-30:
getField is an oddball as it has a Requester class, but no corresponding (public) operation class.   ```Channel::getField()``` returns void so, unlike the other PVA operations, there is no way for user code too keep the operation "alive" (or cancel it).   To deal with this I changed the PVA provider to keep the getField op. alive (cf. ```InternalClientContextImpl::InternalChannelImpl::m_getfield```).  The basic idea is that each Channel can have at most one getField operation in progress.  This operation gets automatically canceled when the channel becomes disconnected, or when a second getField is started.

This is a little tricky, and a realize that I didn't document this part.  I'll put this on my TODO list.

commit 28002fa6 (2018-08-29): Merge pull request #121 from epics-base/issue119

From Marty, fixes #119
closed 2018-08-29 by anjohnson