epics-base/pvAccessCPP #141: "Abort trap: 6" on OSX

mdavidsaver, 2019-04-15:
I've had reports of this error appearing in several contexts with P4P and softIocPVA.  This is a general error, so there may be multiple issues involved.

@georgweiss Provides a stack trace of one case.

```
* thread #2, stop reason = signal SIGABRT
  * frame #0: 0x00007fff63690b66 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff6385b080 libsystem_pthread.dylib`pthread_kill + 333
    frame #2: 0x00007fff635ec1ae libsystem_c.dylib`abort + 127
    frame #3: 0x0000000100c42494 libgcc_s.1.dylib`uw_init_context_1.cold.8 + 5
    frame #4: 0x0000000100c40027 libgcc_s.1.dylib`_Unwind_Resume + 55
    frame #5: 0x00000001002a05e7 libpvAccess.6.1.0.dylib`epics::pvAccess::detail::AbstractCodec::processSender(std::shared_ptr<epics::pvAccess::TransportSender> const&) at codec.cpp:883
    frame #6: 0x00000001002319b8 libpvAccess.6.1.0.dylib`epics::pvAccess::detail::BlockingTCPTransportCodec::sendThread(this=0x0000000101801000) at codec.cpp:850
    frame #7: 0x00000001008e2fb9 libCom.3.17.3.dylib`::epicsThreadCallEntryPoint(pPvt=0x0000000101801180) at epicsThread.cpp:83
    frame #8: 0x00000001008ea87b libCom.3.17.3.dylib`start_routine(arg=0x0000000100f50bd0) at osdThread.c:403
    frame #9: 0x00007fff63858661 libsystem_pthread.dylib`_pthread_body + 340
    frame #10: 0x00007fff6385850d libsystem_pthread.dylib`_pthread_start + 377
    frame #11: 0x00007fff63857bf9 libsystem_pthread.dylib`thread_start + 13
```

mdavidsaver, 2019-04-15:
codec.cpp:883 in this case is the final line of:

```c++
void AbstractCodec::processSender(
    TransportSender::shared_pointer const & sender)
{

    ScopedLock lock(sender);
```

So in the context of unwinding, this is unlocking the `sender` mutex.

So from the [full stack trace ](https://github.com/epics-base/pvAccessCPP/files/3081067/thread_dump.txt), this appears to be during TCP connection shutdown as the recv() thread is not present.

mdavidsaver, 2019-04-15:
Ha.  The joke's on me.  Obviously `sender` isn't a mutex. `TransportSender` is `Lockable`, but its lock/unlock methods are/were no-ops (cf. b05e7699b51a02e895fc90802daf1e41b535515b).  So `TransportSender::unlock()` is a no-op, however it is still a virtual call.

mdavidsaver, 2019-04-15:
`valgrind --tool=helgrind` reports a possible race at codec.cpp:883.  I usually dismiss cases involving ~shared_ptr as false positives.  By my understanding, as long as all callers hold a shared_ptr, then no race is actually possible.  valgrind doesn't know enough about shared_ptr to understand this sequencing.

In this case, the call to `processSender()` comes from `processSendQueue()` (which is apparently inline'd into `senderThread()`).  `processSendQueue()` holds a local `TransportSender::shared_pointer` ref.  And `sendThread()` holds a self ref.  Which appears to me to be correct.

However, from the circumstances I may be wrong.

```
Possible data race during write of size 8 at 0x86E5980 by thread #42
Locks held: 1, at address 0xB7B7990
   at 0x59FC39D: epics::pvAccess::ServerChannelGetRequesterImpl::~ServerChannelGetRequesterImpl() (responseHandlers.h:301)
   by 0x59FC4ED: epics::pvAccess::ServerChannelGetRequesterImpl::~ServerChannelGetRequesterImpl() (responseHandlers.h:301)
   by 0x5A0A1FD: std::_Sp_counted_ptr<epics::pvAccess::ServerChannelGetRequesterImpl*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (shared_ptr_base.h:372)
   by 0x4F1DA59: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:150)
   by 0x4F1BEAE: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() (shared_ptr_base.h:662)
   by 0x59FE94B: std::__shared_ptr<epics::pvAccess::BaseChannelRequester, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() (shared_ptr_base.h:928)
   by 0x59FE967: std::shared_ptr<epics::pvAccess::BaseChannelRequester>::~shared_ptr() (shared_ptr.h:93)
   by 0x5A16A7F: std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> >::~pair() (stl_pair.h:194)
   by 0x5A16A9F: void __gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > > >::destroy<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > >(std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> >*) (new_allocator.h:124)
   by 0x5A1698D: void std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > > > >::destroy<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > >(std::allocator<std::_Rb_tree_node<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > > >&, std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> >*) (alloc_traits.h:487)
   by 0x5A164C0: std::_Rb_tree<int, std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> >, std::_Select1st<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > >, std::less<int>, std::allocator<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > > >::_M_destroy_node(std::_Rb_tree_node<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > >*) (stl_tree.h:567)
   by 0x5A15BE0: std::_Rb_tree<int, std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> >, std::_Select1st<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > >, std::less<int>, std::allocator<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > > >::_M_drop_node(std::_Rb_tree_node<std::pair<int const, std::shared_ptr<epics::pvAccess::BaseChannelRequester> > >*) (stl_tree.h:575)

This conflicts with a previous read of size 8 by thread #43
Locks held: none
   at 0x599DCE0: epics::pvAccess::ScopedLock::unlock() (pvAccess.h:126)
   by 0x599DCAB: epics::pvAccess::ScopedLock::~ScopedLock() (pvAccess.h:114)
   by 0x5998574: epics::pvAccess::detail::AbstractCodec::processSender(std::shared_ptr<epics::pvAccess::TransportSender> const&) (codec.cpp:883)
   by 0x5998245: epics::pvAccess::detail::AbstractCodec::processSendQueue() (codec.cpp:850)
   by 0x5997E49: epics::pvAccess::detail::AbstractCodec::processWrite() (codec.cpp:760)
   by 0x59992AE: epics::pvAccess::detail::BlockingTCPTransportCodec::sendThread() (codec.cpp:1161)
   by 0x59A70E7: epics::pvData::detail::MethRunner<epics::pvAccess::detail::BlockingTCPTransportCodec>::run() (thread.h:56)
   by 0x69D8A4B: epicsThreadCallEntryPoint (epicsThread.cpp:95)
 Address 0x86e5980 is 0 bytes inside a block of size 320 alloc'd
   at 0x4C2D63F: operator new(unsigned long) (vg_replace_malloc.c:334)
   by 0x59ECCBC: epics::pvAccess::ServerChannelGetRequesterImpl::create(std::shared_ptr<epics::pvAccess::ServerContextImpl> const&, std::shared_ptr<epics::pvAccess::ServerChannel> const&, int, std::shared_ptr<epics::pvAccess::Transport> const&, std::shared_ptr<epics::pvData::PVStructure> const&) (responseHandlers.cpp:1116)
   by 0x59EC651: epics::pvAccess::ServerGetHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:1047)
   by 0x59E70A6: epics::pvAccess::ServerResponseHandler::handleResponse(osiSockAddr*, std::shared_ptr<epics::pvAccess::Transport> const&, signed char, signed char, unsigned long, epics::pvData::ByteBuffer*) (responseHandlers.cpp:171)
   by 0x599E16F: epics::pvAccess::detail::BlockingTCPTransportCodec::processApplicationMessage() (codec.h:351)
   by 0x5996094: epics::pvAccess::detail::AbstractCodec::processReadNormal() (codec.cpp:223)
   by 0x5995A8F: epics::pvAccess::detail::AbstractCodec::processRead() (codec.cpp:133)
   by 0x59990DE: epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() (codec.cpp:1133)
   by 0x59A70E7: epics::pvData::detail::MethRunner<epics::pvAccess::detail::BlockingTCPTransportCodec>::run() (thread.h:56)
   by 0x69D8A4B: epicsThreadCallEntryPoint (epicsThread.cpp:95)
   by 0x69E1FC2: start_routine (osdThread.c:411)
   by 0x4C32D06: mythread_wrapper (hg_intercepts.c:389)
 Block was alloc'd by thread #42
```

georgweiss, 2019-04-17:
Procedure to reproduce (verified on Mac OS 10.13.6):
1) Install Homebrew gcc, latest version 8.3.0 or 7.4.0.
2) Build EPICS base as instructed here: https://github.com/mdavidsaver/p4p-dev/blob/gw/gateway.md. The build must use the Homebrew compiler, i.e. edit <epics-base>/configure/os/CONFIG_SITE.Common.darwin-x86 to read:
CMPLR_CLASS = gcc
CC  = /usr/local/Cellar/gcc/8.3.0/bin/gcc-8
CCC = /usr/local/Cellar/gcc/8.3.0/bin/g++-8
GNU = YES
3)Build and launch a simple IOC against the compiled EPICS base.
4)Query the IOC for a PV using pvget -> IOC crashes.

**Note**: The IOC does NOT crash when the pvAccess module is build from the 7.0 branch.
**Note**: When using the standard Mac OS compiler (clang) to build EPICS base (with patched pvAccess module), the IOC does NOT crash.

anjohnson, 2019-04-17:
> The IOC does NOT crash when the pvAccess module is built from the 7.0 branch.

So this bug report shouldn't really be reported against [epics-base/pvAccessCPP](https://github.com/epics-base/pvAccessCPP) but should be against the `md/gwdev` branch of [mdavidsaver/pvAccessCPP](https://github.com/mdavidsaver/pvAccessCPP) since that's what those instructions tell you to build.

mdavidsaver, 2019-04-17:
I did not understand this.  I must have mis-read an earlier private email exchange.

georgweiss, 2019-04-24:
It turns out that this is indeed a problem on the 7.0 branch, contrary to my previous comment. However, when repeating the test procedure on Mac OS 10.14.4 (Mojave) with Homebrew gcc/g++ 8.3.0, I can NOT reproduce the observed IOC crash.

Prerequisites to reproduce:

- Mac OS 10.13.6 (High Sierra). 
- Homebrew gcc/g++ 8.3.0 (or 7.4.0).
- Pull from 7.0 branch
- On Mac OS 10.13.6 I was not able to trick the OS to run the gdb debugger as expected (gdb launches, but the OS blocks gdb to run the IOC binary). Instead the native lldb debugger can be used. I could not make the Valgrind debugger work, it is blocked in the same manner as gdb.

To build with the Homebrew compiler, edit configure/os/CONFIG_SITE.Common.darwin-x86 to specify the correct paths, see my earlier comment in this thread. Disable -O3 optimization in configure/CONFIG.gnuCommon for more precise debugger behavior.

Build and run a simple IOC against the EPICS build. Then issue a pvget; the result is delivered back to the client, but the IOC crashes as a consequence of the closed connection. In modules/pvAccess/src/remote/codec.cpp the dropped connection throws an connection_closed_exception which is caught and re-thrown on line 901. A single step in the debugger from line 901 crashes the IOC. The expected behavior would be to catch the exception on line 858, where it is re-thrown to be handled on line 1168, in a block that does not re-throw. But breakpoints on 858 and 1168 are never hit in the debug session.

The expected handling of the connection_closed_exception can be observed if EPICS base is compiled with the native Apple clang compiler. In this case the IOC does not crash.

mark0n, 2019-05-21:
@georgweiss could you please try running a binary compiled with clang and `-g -fsanitize=address` as well as another one compiled with `-g -fsanitize=thread` (please recompile Base with these parameters as well)?

georgweiss, 2019-05-22:
@mark0n, the observed issue is when building with gcc. A clang build does not trigger a crash.

mark0n, 2019-05-22:
@georgweiss I understand but "doesn't crash" doesn't mean correct. I was hoping that Thread Sanitizer might detect a data race. @mdavidsaver found something using Helgrind but we are having a hard time understanding it. Maybe Thread Sanitizer can give us some additional information?

georgweiss, 2019-05-22:
OK, I will give it a go, but it might take a week or so before I can find the time.

mdavidsaver, 2019-06-12:
This might actually be a gcc bug, possibly with some OSX involvement (mixing libstdc++?) given that the same gcc version works with Mac OS 10.14.4, but not 10.13.6.  I can't be sure because "Abort trap: 6" is so generic, but this sounds similar to https://github.com/Homebrew/homebrew-core/issues/7837

@georgweiss  I can't think of many practical suggestions.  If you want to pursue this further, and if it is a code generation bug.  The outcome might be effected by moving the definition of `connection_closed_exception` out of line, and adding a virtual destructor (which really should be there anyway).

In codec.h, replace:

```c++
class connection_closed_exception: public std::runtime_error {
public:
    explicit connection_closed_exception(const std::string &s): std::runtime_error(s) {}
};
```

with

```c++
class connection_closed_exception: public std::runtime_error {
public:
    explicit connection_closed_exception(const std::string &s);
    virtual ~connection_closed_exception();
};
```

and in codec.cpp add:

```c++
connection_closed_exception::connection_closed_exception(const std::string &s): std::runtime_error(s) {}
connection_closed_exception::~connection_closed_exception() {}
```

georgweiss, 2019-06-13:
I will try rebuilding with the suggested changes. While I still have Mac OS 10.13.6...

georgweiss, 2019-06-25:
Modified codec.h and codec.cpp as suggested by @mdavidsaver, but IOC crashes in the same manner on Mac OS 10.13.6. EPICS base and IOC built with gcc 8.3.0_02.

georgweiss, 2019-06-25:
As for building with clang using -fsanitizer=address and -fsanitizer=thread as suggested by @mark0n:
With -fsanitizer=address I see no difference in behavior, i.e. IOC does not crash.
However, with -fsanitizer=thread I get the following when querying for a PV using pvget:

WARNING: ThreadSanitizer: data race (pid=58593)
  Write of size 4 at 0x7b5c00000130 by main thread (mutexes: write M40):
    #0 epics::pvAccess::detail::AtomicValue<int>::getAndSet(int) epicsAtomicDefault.h:199 (libpvAccess.6.1.2.dylib:x86_64+0x151043)
    #1 epics::pvAccess::detail::AtomicValue<bool>::getAndSet(bool) codec.h:103 (libpvAccess.6.1.2.dylib:x86_64+0x1390ed)
    #2 epics::pvAccess::detail::BlockingTCPTransportCodec::close() codec.cpp:1032 (libpvAccess.6.1.2.dylib:x86_64+0x138f5e)
    #3 epics::pvAccess::detail::BlockingClientTCPTransportCodec::release(int) codec.cpp:1835 (libpvAccess.6.1.2.dylib:x86_64+0x14b597)
    #4 non-virtual thunk to epics::pvAccess::detail::BlockingClientTCPTransportCodec::release(int) codec.cpp (libpvAccess.6.1.2.dylib:x86_64+0x14b698)
    #5 (anonymous namespace)::InternalClientContextImpl::InternalChannelImpl::disconnect(bool, bool) clientContextImpl.cpp:3548 (libpvAccess.6.1.2.dylib:x86_64+0x225a76)
    #6 (anonymous namespace)::InternalClientContextImpl::InternalChannelImpl::destroy() clientContextImpl.cpp:3318 (libpvAccess.6.1.2.dylib:x86_64+0x1abbd5)
    #7 non-virtual thunk to (anonymous namespace)::InternalClientContextImpl::InternalChannelImpl::destroy() clientContextImpl.cpp (libpvAccess.6.1.2.dylib:x86_64+0x1ae87c)
    #8 (anonymous namespace)::InternalClientContextImpl::destroyAllChannels() clientContextImpl.cpp:4222 (libpvAccess.6.1.2.dylib:x86_64+0x236515)
    #9 (anonymous namespace)::InternalClientContextImpl::destroy() clientContextImpl.cpp:4098 (libpvAccess.6.1.2.dylib:x86_64+0x17a5cc)
    #10 non-virtual thunk to (anonymous namespace)::InternalClientContextImpl::destroy() clientContextImpl.cpp (libpvAccess.6.1.2.dylib:x86_64+0x17c3ec)
    #11 epics::pvAccess::Destroyable::cleaner::operator()(epics::pvAccess::Destroyable*) destroyable.h:54 (libpvAccess.6.1.2.dylib:x86_64+0x1bf4fb)
    #12 std::__1::__shared_ptr_pointer<(anonymous namespace)::InternalClientContextImpl*, epics::pvAccess::Destroyable::cleaner, std::__1::allocator<(anonymous namespace)::InternalClientContextImpl> >::__on_zero_shared() memory:3586 (libpvAccess.6.1.2.dylib:x86_64+0x2394b8)
    #13 std::__1::shared_ptr<epics::pvData::Field const>::~shared_ptr() memory:3490 (libpvAccess.6.1.2.dylib:x86_64+0x1d1fe)
    #14 std::__1::shared_ptr<epics::pvData::Field const>::~shared_ptr() memory:4466 (libpvAccess.6.1.2.dylib:x86_64+0x18ce8)
    #15 pvac::ClientProvider::Impl::~Impl() client.cpp:239 (libpvAccess.6.1.2.dylib:x86_64+0x92b08)
    #16 pvac::ClientProvider::Impl::~Impl() client.cpp:239 (libpvAccess.6.1.2.dylib:x86_64+0x92a18)
    #17 std::__1::__shared_ptr_pointer<pvac::ClientProvider::Impl*, std::__1::default_delete<pvac::ClientProvider::Impl>, std::__1::allocator<pvac::ClientProvider::Impl> >::__on_zero_shared() memory:2285 (libpvAccess.6.1.2.dylib:x86_64+0x9277a)
    #18 std::__1::shared_ptr<epics::pvData::Field const>::~shared_ptr() memory:3490 (libpvAccess.6.1.2.dylib:x86_64+0x1d1fe)
    #19 std::__1::shared_ptr<epics::pvData::Field const>::~shared_ptr() memory:4466 (libpvAccess.6.1.2.dylib:x86_64+0x18ce8)
    #20 pvac::ClientProvider::~ClientProvider() client.cpp:285 (libpvAccess.6.1.2.dylib:x86_64+0x842b8)
    #21 pvac::ClientProvider::~ClientProvider() client.cpp:285 (libpvAccess.6.1.2.dylib:x86_64+0x84318)
    #22 main pvget.cpp:430 (pvget:x86_64+0x100004567)

  Previous read of size 4 at 0x7b5c00000130 by thread T6:
    #0 epics::pvAccess::detail::AtomicValue<int>::get() epicsAtomicDefault.h:166 (libpvAccess.6.1.2.dylib:x86_64+0x1514ac)
    #1 epics::pvAccess::detail::AtomicValue<bool>::get() codec.h:106 (libpvAccess.6.1.2.dylib:x86_64+0x139358)
    #2 epics::pvAccess::detail::BlockingTCPTransportCodec::isOpen() codec.cpp:1108 (libpvAccess.6.1.2.dylib:x86_64+0x139c3f)
    #3 epics::pvAccess::detail::AbstractCodec::processReadNormal() codec.cpp:226 (libpvAccess.6.1.2.dylib:x86_64+0x12ffa4)
    #4 epics::pvAccess::detail::AbstractCodec::processRead() codec.cpp:128 (libpvAccess.6.1.2.dylib:x86_64+0x12f71e)
    #5 epics::pvAccess::detail::BlockingTCPTransportCodec::receiveThread() codec.cpp:1138 (libpvAccess.6.1.2.dylib:x86_64+0x139f6e)
    #6 epics::pvData::detail::MethRunner<epics::pvAccess::detail::BlockingTCPTransportCodec>::run() thread.h:56 (libpvAccess.6.1.2.dylib:x86_64+0x154af2)
    #7 epicsThreadCallEntryPoint epicsThread.cpp:83 (libCom.3.17.4.dylib:x86_64+0x4bc08)
    #8 start_routine osdThread.c:403 (libCom.3.17.4.dylib:x86_64+0x5c8a4)

  Location is heap block of size 808 at 0x7b5c00000000 allocated by thread T2:
    #0 operator new(unsigned long) <null> (libclang_rt.tsan_osx_dynamic.dylib:x86_64h+0x6d723)
    #1 epics::pvAccess::detail::BlockingClientTCPTransportCodec::create(std::__1::shared_ptr<epics::pvAccess::Context> const&, int, std::__1::shared_ptr<epics::pvAccess::ResponseHandler> const&, int, int, std::__1::shared_ptr<epics::pvAccess::ClientChannelImpl> const&, signed char, float, short) codec.h:641 (libpvAccess.6.1.2.dylib:x86_64+0x102e62)
    #2 epics::pvAccess::BlockingTCPConnector::connect(std::__1::shared_ptr<epics::pvAccess::ClientChannelImpl> const&, std::__1::shared_ptr<epics::pvAccess::ResponseHandler> const&, osiSockAddr&, signed char, short) blockingTCPConnector.cpp:14 [... cut ...]
    #12 pvac::ClientProvider::ClientProvider(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::shared_ptr<epics::pvAccess::Configuration> const&) client.cpp:253 (libpvAccess.6.1.2.dylib:x86_64+0x83b38)
    #13 main pvget.cpp:387 (pvget:x86_64+0x100002aec)

SUMMARY: ThreadSanitizer: data race epicsAtomicDefault.h:199 in epics::pvAccess::detail::AtomicValue<int>::getAndSet(int)
==================
ThreadSanitizer: reported 1 warnings
Abort trap: 6

mdavidsaver, 2019-06-26:
@georgweiss Thanks for trying.  It was a long shot.  I don't see anything more we can do to troubleshoot this.  PVA is just not going to work with the specific combination of Mac OS 10.13.6 and gcc 8.3.0_02.

I suspect this is an error in generation of code and/or unwinding info by gcc.  If this is the case, it is still an open question whether this an actual gcc bug, or a case of garbage in our part (an error in PVA source).

We'll have to wait for this crash turns up in some other environment...

anjohnson, 2022-02-23:
MacOS 10.13 (High Sierra) went out of support in December 2020. I think we can close this.

closed 2022-02-23 by anjohnson