epics-base/pvAccessCPP #142: Echo doesn't, well, echo

kasemir, 2019-05-16:
#139 mentions that the C++ client may never actually send echo requests.
Working on a Java client that sends echo requests I noticed that the C++ server does reply to echo requests, but it always sends an empty reply.
According to the protocol, the echo should contain the payload of the echo request sent by the client, https://github.com/epics-base/pvAccessCPP/wiki/protocol#echo-0x02

I assume the issue is in https://github.com/epics-base/pvAccessCPP/blob/master/src/server/responseHandlers.cpp#L228 which ignores the received payload and always enqueues an empty echo reply:
```
void ServerEchoHandler::handleResponse(osiSockAddr* responseFrom,
                                       Transport::shared_pointer const & transport, int8 version, int8 command,
                                       size_t payloadSize, ByteBuffer* payloadBuffer)
{
    AbstractServerResponseHandler::handleResponse(responseFrom,
            transport, version, command, payloadSize, payloadBuffer);

    // send back
    TransportSender::shared_pointer echoReply(new EchoTransportSender(responseFrom));
    transport->enqueueSendRequest(echoReply);
}
```

Since the echo is requested and received via TCP, there is probably no need to check "is this the reply to _my_ echo request?". Still, a paranoid client might want to include a counter in the echo requests to check if the replies are matched.

mdavidsaver, 2019-05-16:
I agree that this is an omission which should be corrected.  Do we know if the Java server behaves similarly?

kasemir, 2019-05-16:
Yes, it's fully compatible.
Only difference is that the source code contains a commented-out section that tries to include the received payload.

https://github.com/epics-base/epicsCoreJava/blob/master/pvAccessJava/src/org/epics/pvaccess/server/impl/remote/handlers/EchoHandler.java

mdavidsaver, 2019-05-16:
Ah, I see.  Presumably because `payloadBuffer` is shared, and has been reused by the time the send() callback runs.

kasemir, 2019-05-16:
Assume the server is "fixed" and a client now depends on actually receiving the echoed messages. Such a client will fail with older servers, so now we need a new protocol version, ...
Based on the last core phone meeting, I don't see a way to overcome this.

mdavidsaver, 2019-05-16:
> so now we need a new protocol version, ...

Yes.

> I don't see a way to overcome this.

Increment the protocol version and see if anything breaks?  So far I've not come across any code which actually tests the protocol (minor) version.

kasemir, 2019-05-16:
As for the commented section, I think using `payloadBuffer.remaining()` was wrong. Need to use the `payloadSize` that's received based on the size declared in the currently handled message, and not whatever else might follow in the buffer.

mdavidsaver, 2019-05-17:
TODO content...

https://github.com/epics-base/pvAccessCPP/blob/bd7ae1214363e9cd6ae37437568c3f9ed7d12f97/src/server/pv/responseHandlers.h#L102-L106

mdavidsaver, 2019-08-07:
Released in module version 7.0.0, included in EPICS 7.0.3

closed 2019-08-07 by mdavidsaver