epics-base/pvAccessCPP #147: GET sub-commands

kasemir, 2019-06-04:
Protocol for Channel get (0x0A) describes the use of the  subcommand as 0x08 for INIT and 0x40 for GET, with additional 0x10 mask for DESTROY.

When sending 0x40 for the actual GET that "works", but the EPICS 7 `pvget` command i.e. the C++ library, seems to send 0x00 for GET, just like PUT uses 0x00 for the actual operation.

Suggest to simply update the documentation.

mdavidsaver, 2019-06-04:
This tracks.

On the server side the logic accepts QOS_INIT and treats anything else as execute.

https://github.com/epics-base/pvAccessCPP/blob/e66893bb38ed47ec6255cf6dd8911f4549b2d5ef/src/server/responseHandlers.cpp#L1035-L1037

For GET, the client appears to send the combinations: `QOS_INIT`, `QOS_DEFAULT` (aka 0), or `QOS_GET|QOS_DESTROY`.  The last two seem inconsistent to me...

https://github.com/epics-base/pvAccessCPP/blob/e66893bb38ed47ec6255cf6dd8911f4549b2d5ef/src/remoteClient/clientContextImpl.cpp#L384-L385

https://github.com/epics-base/pvAccessCPP/blob/e66893bb38ed47ec6255cf6dd8911f4549b2d5ef/src/remoteClient/clientContextImpl.cpp#L687

kasemir, 2019-06-04:
Updated the protocol doc to mention that 0x00 is used.

closed 2019-06-04 by kasemir