epics-base/pvAccessCPP #148: PVA hexdump changed

kasemir, 2019-06-26:
When setting `EPICS_PVA_DEBUG=100`, the hexdump used to include ASCII detail, somewhat like this:
```
Message [0x3, v0x2] received from 127.0.0.1:58962 on 0.0.0.0:5076
Hexdump [Search request] size = 41
00 00 00 01  00 00 00 00  00 00 00 00  00 00 00 00  .... .... .... .... 
00 00 FF FF  7F 00 00 01  E6 52 01 03  74 63 70 00  .... .... .R.. tcp. 
01 00 00 00  01 03 73 75  6D                        .... ..su m
```

When I use Base 7.0.2.2, the output only lists hex numbers, no ASCII representation:
```
Message [0x3, v0x2] received from 127.0.0.1:56533 on 0.0.0.0:5076 : Search request
0x00 00000001 00000000 00000000 00000000
0x10 0000ffff 7f000001 dcd50103 74637000
0x20 01000000 01037375 6d
```

For what it's worth, I found the previous hexdump format more useful, both the space between bytes (instead of showing 4-byte numbers) and the ASCII representation.

mdavidsaver, 2019-06-26:
I agree that this was useful feature, though no enough to preserve it with ecee708d3a217277fcc9579f1679b3ff1b30459a

This could be added back.

kasemir, 2019-06-26:
I haven't looked into the detailed changes, but agree that logging down to the hexdump level is expensive. For the Java code, when full logging is enabled, the profiler shows that all you're doing it formatting text. But then Java logging works such that the formatting isn't invoked when the log level doesn't apply, so no CPU cycles wasted unless you actually enable the detailed logging.
If I enable logging, I'm not looking for performance but basic protocol details.

mdavidsaver, 2019-06-27:
There is expensive, and there is "expensive".  Multiple memory allocations per character is a different level.

Anyway fd95f90acc7764b879bad1d56c0be57103d07523

mdavidsaver, 2019-08-07:
Released in module version 7.0.0, included in EPICS 7.0.3

closed 2019-08-07 by mdavidsaver