epics-base/pvAccessCPP #160: CMD_DESTROY_CHANNEL (0x08) order of SID, CID

kasemir, 2020-04-07:
The CMD_DESTROY_CHANNEL (0x08) body should contain the SID and CID of the channel to destroy, in that order.

Most of the pvAccessCPP, old and new Java lib and new pvxs C++ lib use it that way.

In the old CPP client's DestroyChannelHandler, however, this piece of code reads them in the opposite order:
https://github.com/epics-base/pvAccessCPP/blob/master/src/remoteClient/clientContextImpl.cpp#L2932
```
pvAccessID cid = payloadBuffer->getInt();
/*pvAccessID sid =*/ payloadBuffer->getInt();
```

So when the PVA gateway indicates that a channel has been dropped and sends a CMD_DESTROY_CHANNEL to the client, the old 'pvmonitor' won't recognize the channel since CID and SID are swapped, or it might report the wrong channel as disconnected.

Fox would be to swap those two lines.

mdavidsaver, 2020-04-07:
Looks like you found a gap in testing.  The swap isn't noticeable unless `CID!=SID`.  Both are being started at `0` and will not differ in one client + one server configurations.

This effects mainly clients behind a gateway, where this would appear as connected clients seeing PVs "freeze" (no updates) when an server behind the gateway restarts.

mdavidsaver, 2020-04-07:
Current state

https://github.com/epics-base/pvAccessCPP/blob/4ef9c1102b422c0217cbb07c6d1a574f5baa4d83/src/remoteClient/clientContextImpl.cpp#L2932-L2933

The fix seems easy enough.

kasemir, 2020-04-07:
Another related nice thing to have: Start either the client or server IDs at some higher number, so simple tests aren't all comparing 1==1

mdavidsaver, 2020-04-07:
Where the server builds CMD_DESTROY_CHANNEL

https://github.com/epics-base/pvAccessCPP/blob/4ef9c1102b422c0217cbb07c6d1a574f5baa4d83/src/server/pv/responseHandlers.h#L259-L261

mdavidsaver, 2020-04-07:
> so simple tests aren't all comparing 1==1

With 19d868eb7dcb0e08b332a9cbe262366eb018c153 this is no longer the case.  I verified that the gateway tests break with just this change.  And then are fixed by ef4bf9120e2d624dab8d25af49d627380c8c98d5

kasemir, 2020-04-07:
Excellent, thanks!

mdavidsaver, 2020-06-03:
Released in module version 7.1.1 with Base 7.0.4

closed 2020-06-03 by mdavidsaver