epics-base/pvAccessCPP #168: pvput and CALC expressions

ttkorhonen, 2020-09-24:
Trying to use pvput to set CALC field of a calcout record fails with a syntax that should be correct:
$ pvput timo:driveSeq.CALC "B<=A"
Old : <undefined>              B<A?VAL+1:0 INVALID DRIVER UDF 
**B< : Warning: no such field
Error: invalid bit-set length**
New : <undefined>              B<A?VAL+1:0 INVALID DRIVER UDF  

The same syntax works with caput though:

$ caput timo:driveSeq.CALC "B<=A"
Old : timo:driveSeq.CALC             B<A?VAL+1:0
New : timo:driveSeq.CALC             B<=A

A bit of a fringe case but I thought to report this nevertheless. I try to use pvtools for all my command line work.

ralphlange, 2020-09-24:
Transferring issues is cool.

ttkorhonen, 2020-09-24:
Thanks. This is why I was not able to transfer it, even if I tried hard...

ralphlange, 2020-09-24:
So sorry!

mdavidsaver, 2020-09-24:
Well, this is weird.

```
$ pvput cnt.CALC 'A>=2'
Old : 2020-09-24 07:38:41.319  A+2 
A> : Warning: no such field
New : 2020-09-24 07:38:41.319  A+2
$ pvput cnt.CALC 'A+1'
Old : 2020-09-24 07:38:50.319  A+2 
New : 2020-09-24 07:38:50.319  A+1
```

mdavidsaver, 2020-09-24:
Ah, but of course.  It's a feature...

```
$ pvput cnt.CALC 'value=A<=2'
Old : 2020-09-24 07:50:27.319  A+2 
New : 2020-09-24 07:50:27.319  A<=2
```

mdavidsaver, 2020-09-24:
To be explicit, this issue is an artifact of how I'm interpreting command line arguments.  Specifically, I'm doing so in two stages.  The first splits on `'='` before the remote type is known, when it isn't possible to distinguish eg. 'A' from a valid field name.  I can't think of a reason why this couldn't be changed.

anjohnson, 2020-09-24:
CALC expressions are long strings and frequently need to be more than 40 chars. AFAIK pvget/put/monitor don't currently have an equivalent of the catools `-S` option or any other way to handle long string data. Different issue, I know...

PR #174 (2021-03-09): Handle inputs with "=" to pvput
This PR resolves #168 

The argument handling code of the `pvput` tool was reworked in order to handle bare value with `=` characters, which could be used for example to write to `DESC` or `CALC` fields. 

When trying to tread an input argument as a `field=value` pair, the code verifies if the `field` exists in the remote PV structure. If it doesn't exist, then it checks if the `.value` field is of type `string` and if it is, then the whole input value is treated as a bare value; otherwise the input is ignored (due to the `filed` being invalid).
closed 2021-03-09 by mdavidsaver