epics-base/pvAccessCPP #177: polluting ostream flags

dirk-zimoch, 2021-11-19:
Some functions/methods/operators manipulate the flags of their `std::ostream` argument  for examples using `std::hex` and/or `std::setw` without saving the initial `flags()` and restoing it when done and thus pollution the state of the ostream.

Examples are:
- HexDump <<
- ResponseHandler::handleResponse
- ClientResponseHandler::handleResponse
- ServerResponseHandler::handleResponse

Also affected but in pvData:
- shared_ptr_base::show_stack

Be aware that `<<std::hex` not only affect the next value but all future values.

The solution is to save and restore the flags:
```
std::ios::fmtflags initialflags = output.flags();
output << std::hex() << ...
output.flags(initialflags);
```

dirk-zimoch, 2021-11-19:
Correction: `setw()` resets after the next argument. But not `hex` or `setfill()`.

anjohnson, 2022-02-23:
I'm going to assume the merge of #178 fixed this issue, please reopen if there were parts not fixed by that.

closed 2022-02-23 by anjohnson