epics-base/pvAccessCPP #188: How to improve PVA put concurrency performance

zhouyunbin, 2023-04-26:
I need to put lots of PV at the same time,  which function I used is  chan.put.set<double>().exec()， but no matter how I use std::jthread or std::thread to achieve, it looks like always sequential execution at the bottom level. Is there a way to put the value without for waiting timeout?

mdavidsaver, 2023-04-26:
Based on `chan.put.set().exec()` I think this is a question about https://github.com/mdavidsaver/pvxs .  I will try to transfer the issue.

sveseli, 2023-04-26:
The question is probably for pvxs client, but you could also try MultiChannel class from the pvaClientCPP module. Here is what one can expect when connecting to and retrieving data from 1000 CA channels running in an IOC on the same machine:

<pre>
>>> from pvaccess import *
>>> cList = ['X%s' % i for i in range (1,1001)]
>>> import time
>>> from pvaccess import *
>>> cList = ['X%s' % i for i in range (1,1001)]
>>> mc = MultiChannel(cList, CA)
>>> t0 = time.time(); pv = mc.get(); t1 = time.time(); print(t1-t0)
0.27422356605529785
>>> t0 = time.time(); pv = mc.get(); t1 = time.time(); print(t1-t0)
0.05835700035095215
>>> t0 = time.time(); pv = mc.get(); t1 = time.time(); print(t1-t0)
0.055959224700927734
</pre>

The initial request is a bit slower because it takes some time to connect to the PVs. Here is what happens with put:

<pre>
>>> vList = [float(i) for i in range (1,1001)]
>>> t0 = time.time(); mc.putAsDoubleArray(vList); t1 = time.time(); print(t1-t0)0.13571906089782715
>>> t0 = time.time(); mc.putAsDoubleArray(vList); t1 = time.time(); print(t1-t0)
0.13674354553222656
</pre>

closed 2023-04-26 by mdavidsaver