epics-base/pvAccessCPP #190: epicsThreadShowAll lists non-existent threads when too many monitors are created

simon-ess, 2023-05-23:
If you create an IOC with a single record, and then create and kill a large number of monitors on that record, then epicsThreadShowAll reveals a number of threads that no longer exist.

I am not getting 100% consistent numbers, but somewhere between 50--100 monitors on a given record will yield these "ghost" threads. Specifically, I run a number of `pvmonitor x`, wait 30 seconds, and then kill them all, and then run `epicsThreadShowAll` in the IOC and I see something like
```
            NAME       EPICS ID   LWP ID   OSIPRI  OSSPRI  STATE
          _main_       0x60b450    23615      0       0       OK
          errlog       0x6c6280    23618     10       0       OK
          taskwd       0x6c6810    23619     10       0       OK
      timerQueue       0x6c7200    23620     70       0       OK
           cbLow       0x6c7650    23621     59       0       OK
        cbMedium       0x6c7840    23622     64       0       OK
          cbHigh       0x6c7c80    23623     71       0       OK
        dbCaLink       0x6d4600    23624     50       0       OK
            PVAL       0x6d4fc0    23625     50       0       OK
       PDB-event       0x6dbad0    23626     19       0       OK
pvAccess-client        0x6dcaa0    23627     35       0       OK
UDP-rx 0.0.0.0:0       0x6de400    23628     50       0       OK
UDP-rx 10.0.2.15       0x73f030    23629     50       0       OK
UDP-rx 224.0.0.1       0x75fdf0    23631     50       0       OK
UDP-rx 10.0.2.25       0x73f490    23630     50       0       OK
        scanOnce       0x766810    23632     67       0       OK
         scan-10       0x766b80    23633     60       0       OK
          scan-5       0x766ef0    23634     61       0       OK
          scan-2       0x767260    23635     62       0       OK
          scan-1       0x7675d0    23636     63       0       OK
        scan-0.5       0x767940    23637     64       0       OK
        scan-0.2       0x767cb0    23638     65       0       OK
        scan-0.1       0x768020    23639     66       0       OK
         CAS-TCP       0x7709e0    23640     18       0       OK
         CAS-UDP       0x770d90    23641     16       0       OK
      CAS-beacon       0x771100    23642     17       0       OK
     PVAS timers       0x771b70    23643     25       0       OK
    TCP-acceptor       0x772a20    23644     50       0       OK
UDP-rx 0.0.0.0:0       0x793390    23645     50       0       OK
UDP-rx 10.0.2.15       0x7d3f70    23646     50       0       OK
UDP-rx 10.0.2.25       0x7d43d0    23647     50       0       OK
UDP-rx 224.0.0.1       0x7f4ce0    23648     50       0       OK
          TCP-rx 0x7fffcc162000    24941     20       0       OK
          TCP-tx 0x7fffcc1624e0    24942     20       0       OK
          TCP-rx 0x7fffcc17c800    24955     20       0       OK
          TCP-tx 0x7fffcc17cce0    24956     20       0       OK
          TCP-tx 0x7fffcc1974f0    25002     20       0       OK
          TCP-rx 0x7fffcc197010    25001     20       0       OK
          TCP-rx 0x7fffcc1b1820    25007     20       0       OK
          TCP-tx 0x7fffcc1b1d00    25008     20       0       OK
          TCP-rx 0x7fffcc083150    25022     20       0       OK
          TCP-tx 0x7fffcc029f70    25049     20       0       OK
          TCP-rx 0x7fffcc0ef820    25164     20       0       OK
          TCP-tx 0x7fffcc0b16b0    25165     20       0       OK
          TCP-rx 0x7fffcc00f8d0    25170     20       0       OK
          TCP-tx 0x7fffcc0b1570    25171     20       0       OK
          TCP-rx 0x7fffcc083cc0    25172     20       0       OK
          TCP-tx 0x7fffcc029e90    25173     20       0       OK
          TCP-rx 0x7fffcc014cf0    25174     20       0       OK
          TCP-tx 0x7fffcc049820    25175     20       0       OK
          TCP-rx 0x7fffcc01f530    25176     20       0       OK
          TCP-tx 0x7fffcc0051b0    25177     20       0       OK
          TCP-rx 0x7fffcc123900    25178     20       0       OK
          TCP-tx 0x7fffcc128d20    25179     20       0       OK
          TCP-rx 0x7fffcc109270    25184     20       0       OK
          TCP-tx 0x7fffcc1190c0    25185     20       0       OK
          TCP-rx 0x7fffcc1475d0    25186     20       0       OK
          TCP-tx 0x7fffcc0ab650    25187     20       0       OK
          TCP-rx 0x7fffcc077d30    25230     20       0       OK
          TCP-tx 0x7fffcc0eedd0    25231     20       0       OK
```
The top threads are obviously expected, but the bottom ones are not. Moreover, they do not seem to exist according to the host OS: if you run `ps -T -C softIocPVA` then it does not  list any of these threads once the monitors have been killed.

If you repeat the process then the list simply gets longer each time.

This is EPICS base 7.0.7 on a CentOS7 VM.

mdavidsaver, 2023-05-23:
Ok...  My being tired of untangling a seemingly unending stream of reference loops like this one played a major role in motivating me to write PVXS.  So I'm going to prioritize work on PVXS over this issue.  So someone else is welcomed to have a go.  Looking for `DEBUG_SHARED_PTR` in `pv/sharedPtr.h` might be a place to start.  This is one of the tools I developed to hopefully make troubleshooting this sort of situation more tractable.

I will also look at making `epicsThreadShowAll()` flag "zombie" threads so that it is more obvious when this sort of situation comes up again.

mdavidsaver, 2023-08-25:
Looking again at this issue and https://github.com/epics-base/epics-base/pull/390, I notice that some of the `pvmonitor` seemingly deadlock on exit.

```
Thread 1 (Thread 0x7f68dce22740 (LWP 231493) "pvmonitor"):
#0  __futex_lock_pi64 (futex_word=futex_word@entry=0x55ccbbec8b80, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ./nptl/futex-internal.c:180
#1  0x00007f68dca8af86 in __pthread_mutex_lock_full (mutex=0x55ccbbec8b80) at ./nptl/pthread_mutex_lock.c:441
#2  0x00007f68dca8b285 in ___pthread_mutex_lock (mutex=mutex@entry=0x55ccbbec8b80) at ./nptl/pthread_mutex_lock.c:86
#3  0x00007f68dce8809a in mutexLock (id=0x55ccbbec8b80) at ../osi/os/posix/osdMutex.c:116
#4  epicsMutexOsdLock (pmutex=0x55ccbbec8b80) at ../osi/os/posix/osdMutex.c:163
#5  0x00007f68dce8093c in epicsMutex::lock (this=this@entry=0x55ccbbec8aa0) at ../osi/epicsMutex.cpp:276
#6  0x00007f68dd0c15ae in epics::pvData::Lock::Lock (m=..., this=<synthetic pointer>) at /home/dev/epics/pv/pvData/include/pv/lock.h:45
#7  (anonymous namespace)::InternalClientContextImpl::InternalChannelImpl::destroy (this=0x55ccbbec8920) at ../../src/remoteClient/clientContextImpl.cpp:3302
#8  0x00007f68dd0b1479 in epics::pvAccess::Destroyable::cleaner::operator() (this=<optimized out>) at ../../src/utils/pv/destroyable.h:54
...
```

mdavidsaver, 2023-08-25:
> ... So I'm going to prioritize work on PVXS over this issue. ...

QSRV2 in PVXS does not have this issue as it does not create dedicated thread(s) for each client.  In general with PVXS I have tried to be stricter about resource ownership transfers through the user API.  So ownership should be rooted in external objects, and situations like a worker thread holding the last reference to itself should not happen, even momentarily.  Also, `pvxmonitor` does not deadlock on exit.