epics-base/pvAccessCPP #194: pvget and pvmonitor need more output options

drm-slac, 2024-03-20:
Output options for pvget and pvmonitor aren't as mature as caget/camonitor, including numeric formatting and more.
This comes from a Tech-talk message on 2024-Mar-06.

chrisalguerrero, 2024-06-10:
@anjohnson @drm-slac I am interested in looking into this issue. Are we looking for changes from the output of the client API? Does anyone have an idea of a good starting point to make the output more numerically formatted? Or if we can come up with a list of tasks that would make it easier to address the issue. I think the classes in the pvac::ClientChannel would be a good start point and we can add a check for each which identifies which format we should be using.

chrisalguerrero, 2024-06-10:
I just read the epics tech-talk thread, and was able to find the pvtools source code here: https://github.com/epics-base/pvAccessCPP/tree/master/pvtoolsSrc

On line 333 of pvget.cpp, what does "T-types format mode" mean?

ttkorhonen, 2024-06-10:
If you look at the `usage()` (line 50), it says that this option, and a few others are deprecated, so it probably is not worth spending time on those options...unless somebody else here disagrees.